cezanne/usbip-win #97: usbip: error: read_devbuf: failed to read: err: 0x6 when trying to attach

athei, 2020-01-31:
My server is a raspbian buster with a bluetooth usb dongle I exposed via usbip:
```
C:\usbip>usbip.exe list -r 192.168.222.31
Exportable USB devices
======================
 - 192.168.222.31
      1-1.5: Cambridge Silicon Radio, Ltd : Bluetooth Dongle (HCI mode) (0a12:0001)
           : /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5
           : Wireless / Radio Frequency / Bluetooth (e0/01/01)
```

Now when I try to attach the device on my Windows10 x64 machine using an admin cmd window:
```
C:\usbip>usbip.exe attach -r 192.168.222.31 -b 1-1.5
usbip: error: open vhci driver
usbip: error: read_devbuf: failed to read: err: 0x6
```

The driver might be the culprit:
[image]

koudis, 2020-01-31:
Hello,

it seems like DESCRIPTOR problem.

Can you try branch for #90? There are some bugfixes around GET_*_DESCRIPTOR commands.

koudis, 2020-01-31:
Edit: Not a descriptor problem. Sorry for misunderstanding.

Are you able to compile my branch from #90? It may helps. (I had simillar problem too)

athei, 2020-01-31:
I removed the currently installed vhci driver and replaced it with the one freshly compiled with from `jan_kubalek/vhci_interfaces`. Unfortunately no dice. The error stays exactly the same. I am sure that I am using the new driver. I checked the driver release date in device manager.
```
$ ./usbip attach -r 192.168.222.31 -b 1-1.5
usbip: error: open vhci driver
usbip: error: read_devbuf: failed to read: err: 0x6
```

Also the error message inside the device manager stays the same:
`The driver for this device might be corrupted or your system may be running low on memory or other resources. (Code 3)`

It changes to this message after a reboot. Before it is just complaining about the invalid signature (because of test signing). It does not work with neither message.

Is there any further diagnostic I can provide?

koudis, 2020-01-31:
Sorry but i am not sure that I can naswer your quistion correctly.
Mayby @cezuni helps?

mjoe, 2020-02-18:
@athei are you in Test Mode `bcdedit.exe /set TESTSIGNING ON` ? The USB/IP VHCI Driver should not show a warning within the Device Manager.

athei, 2020-02-18:
Yeah I have the text on my desktop about test signing mode being enabled. When it is not enabled there is a different warning.

mjoe, 2020-03-02:
The master branch is currently working here on my machine, it's a little bit shaky, but it works. I do not use the precompiled versions. I had similiar issues when

- i was not in this test mode
- the usb port was already in use by someone else
- i didn't start an administrator powershell

hth, mjoe

cezanne, 2020-03-30:
@athei : I want to add another checkpoint on @mjoe recommendations. Please check that a usbip.pfx certificate is installed correctly as in README.

cezanne, 2020-04-02:
@athei : issue closed. Reopen if you want..

closed 2020-04-02 by cezanne