cezanne/usbip-win #99: usbip_vhci: process_irp_urb_req: function: Unknown urb function code: 2a

versaloon, 2020-02-09:
here is part of the log:

00001399	24.07401085	usbip_vhci: store_urbr: urbr: [seq:4]	
00001400	24.07401657	usbip_vhci: store_urbr_submit: urbr: [seq:4]	
00001401	24.07402039	usbip_vhci: store_urbr_submit: urbr: [seq:4], func:URB_FUNCTION_SELECT_CONFIGURATION	
00001402	24.07402802	usbip_vhci: submit_urbr: urb requested: status:STATUS_PENDING	
00001403	24.07403183	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00001404	24.07414436	usbip_vhci: vhci_read: Enter	
00001405	24.07415009	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00001406	24.12319565	usbip_vhci: vhci_write: Enter	
00001407	24.12320709	usbip_vhci: process_urb_res: urbr:[seq:4], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
00001408	24.12321854	usbip_vhci: ep setup[0]: addr:81 intv:16 typ:3 mps:0 mts:8 flags:ffffffff	
00001409	24.12322807	usbip_vhci: ep setup[0]: addr:82 intv:0 typ:2 mps:0 mts:512 flags:ffffffff	
00001410	24.12323380	usbip_vhci: ep setup[1]: addr:02 intv:0 typ:2 mps:0 mts:512 flags:ffffffff	
00001411	24.12324715	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00001412	24.12346268	usbip_vhci: vhci_internal_ioctl: Enter	
00001413	24.12346649	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00001414	24.12346840	usbip_vhci: process_irp_urb_req: function: Unknown urb function code: 2a	
00001415	24.12347031	usbip_vhci:(WW) process_irp_urb_req: unhandled function: Unknown urb function code: 2a: len: 136	
00001416	24.12347603	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	
00001417	24.12347794	usbip_vhci: vhci_internal_ioctl: Enter	
00001418	24.12347984	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00001419	24.12348175	usbip_vhci: process_irp_urb_req: function: Unknown urb function code: 2a	
00001420	24.12348557	usbip_vhci:(WW) process_irp_urb_req: unhandled function: Unknown urb function code: 2a: len: 136	
00001421	24.12348557	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	

After seq4, which is a set_configuration requst, vhci will get unknown urb function code. And tcp traffic is stalled.

I'm running usbip on Windows10 64-bit. usbip server is another program running on the same pc.
tcp traffic from the server:
usbip_server: client connected
usbip_server: backend receiving
usbip_server: backend receive done 4 bytes
usbip_server: rx 4 bytes
00000000: 01 11 80 05                                      | ....
usbip_server: recv USBIP_REQ_DEVLIST
usbip_server: backend receiving
usbip_server: backend receive done 4 bytes
usbip_server: rx 4 bytes
00000000: 00 00 00 00                                      | ....
usbip_server: backend receiving
usbip_server: tx 12 bytes
00000000: 01 11 00 05 00 00 00 00 00 00 00 01              | ............
usbip_server: tx 256 bytes
00000000: 2F 76 73 66 2F 75 73 62 64 2F 31 2D 30 00 00 00  | /vsf/usbd/1-0...
00000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
000000A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
000000B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
000000C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
000000D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
000000E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
000000F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
usbip_server: tx 32 bytes
00000000: 31 2D 30 00 00 00 00 00 00 00 00 00 00 00 00 00  | 1-0.............
00000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
usbip_server: tx 32 bytes
00000000: 00 00 00 00 00 00 00 00 03 00 00 00 A7 A8 23 48  | ..............#H
00000010: 20 00 EF 02 01 01 01 02 02 02 01 00 0A 00 00 00  |  ...............
usbip_server: backend closed
usbip_server: client disconnected
usbip_server: client connected
usbip_server: backend receiving
[... 79 lines cut ...]
usbip_server: backend receiving
usbip_server: tx 123 bytes
00000000: 00 00 00 03 00 00 00 03 00 00 00 00 00 00 00 01  | ................
00000010: 00 00 00 00 00 00 00 00 00 00 00 4B 00 00 00 00  | ...........K....
00000020: 00 00 00 00 00 00 00 00 80 06 00 02 00 00 4B 00  | ..............K.
00000000: 09 02 4B 00 02 01 00 A0 32 08 0B 00 02 02 02 01  | ..K.....2.......
00000010: 04 09 04 00 00 01 02 02 01 04 05 24 00 10 01 05  | ...........$....
00000020: 24 01 00 01 04 24 02 02 05 24 06 00 01 07 05 81  | $....$...$......
00000030: 03 08 00 10 09 04 01 00 02 0A 00 00 04 07 05 82  | ................
00000040: 02 00 02 00 07 05 02 02 00 02 00                 | ...........
usbip_server: backend receive done 4 bytes
usbip_server: rx 4 bytes
00000000: 00 00 00 01                                      | ....
usbip_server: recv USBIP_CMD_SUBMIT
usbip_server: backend receiving
usbip_server: backend receive done 44 bytes
usbip_server: rx 44 bytes
00000000: 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000010: 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000020: 00 00 00 00 00 09 01 00 00 00 00 00              | ............
usbip_server: backend receiving
usbip_server: tx 48 bytes
00000000: 00 00 00 03 00 00 00 04 00 00 00 00 00 00 00 00  | ................
00000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
00000020: 00 00 00 00 00 00 00 00 00 09 01 00 00 00 00 00  | ................

versaloon, 2020-02-09:
I find URB_FUNCTION_GET_MS_FEATURE_DESCRIPTOR in usb.h, which value is 0x002a.

versaloon, 2020-02-09:
BTW, the usb device is a IAD device, with only one IAD descriptor.

versaloon, 2020-02-09:
I add a log output for usbip and get the reason for the failed, it's not usbip's problem.
Please close the issue.

cezanne, 2020-02-10:
@versaloon : 
> I add a log output for usbip and get the reason for the failed, it's not usbip's problem.

What's the reason for the problem? Function driver problem ? Or H/W?

> Please close the issue.

You can close your own issue. 😄 I'll close it.

closed 2020-02-10 by cezanne