cezanne/usbip-win #102: usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136

versaloon, 2020-02-10:
I have fixed previous error and come to new one. I'm using a CDC device with IAD descriptor, and the serial device appears on windows running usbip client. And after I open the serial port with putty, it will fail. Below is the log after open the serial port in putty:

00000001	12.60778236	usbip_vhci: vhci_internal_ioctl: Enter	
00000002	12.60778809	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000003	12.60779095	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER	
00000004	12.60779572	usbip_vhci: store_urbr: urbr: [seq:11]	
00000005	12.60779858	usbip_vhci: store_urbr_submit: urbr: [seq:11]	
00000006	12.60780144	usbip_vhci: store_urbr_submit: urbr: [seq:11], func:URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER	
00000007	12.60780716	usbip_vhci: submit_urbr: urb requested: status:STATUS_PENDING	
00000008	12.60781002	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00000009	12.60782146	usbip_vhci: vhci_internal_ioctl: Enter	
00000010	12.60782433	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000011	12.60782623	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER	
00000012	12.60782909	usbip_vhci: submit_urbr: urb pending	
00000013	12.60783005	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00000014	12.60787964	usbip_vhci: vhci_internal_ioctl: Enter	
00000015	12.60788059	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000016	12.60788345	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CONTROL_TRANSFER	
00000017	12.60788536	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136	
00000018	12.60788918	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	
00000019	12.60790253	usbip_vhci: vhci_internal_ioctl: Enter	
00000020	12.60790443	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000021	12.60790634	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CONTROL_TRANSFER	
00000022	12.60790920	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136	
00000023	12.60791206	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	
00000024	12.60794926	usbip_vhci: vhci_internal_ioctl: Enter	
00000025	12.60795116	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000026	12.60795403	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CONTROL_TRANSFER	
00000027	12.60795498	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136	
00000028	12.60795879	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	
00000029	12.60796833	usbip_vhci: vhci_internal_ioctl: Enter	
00000030	12.60796928	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000031	12.60797215	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CONTROL_TRANSFER	
00000032	12.60797405	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136	
00000033	12.60797787	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	
00000034	12.60799694	usbip_vhci: vhci_internal_ioctl: Enter	
00000035	12.60799885	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000036	12.60800076	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CONTROL_TRANSFER	
00000037	12.60800362	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136	
00000038	12.60800648	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	
00000039	12.60801125	usbip_vhci: vhci_internal_ioctl: Enter	
00000040	12.60801315	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000041	12.60801411	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CONTROL_TRANSFER	
00000042	12.60801697	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136	
00000043	12.60801888	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	
00000044	12.60802078	usbip_vhci: vhci_read: Enter	
00000045	12.60802937	usbip_vhci: store_urbr: urbr: [seq:12]	
00000046	12.60803413	usbip_vhci: store_urbr_submit: urbr: [seq:12]	
00000047	12.60803986	usbip_vhci: store_urbr_submit: urbr: [seq:12], func:URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER	
00000048	12.60804462	usbip_vhci: vhci_read: Leave: STATUS_SUCCESS	
00000049	12.60809803	usbip_vhci: vhci_read: Enter	
00000050	12.60810184	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00000051	26.72192955	usbip_vhci: irp will be cancelled: FFFFC30619437760	
00000052	26.72193527	usbip_vhci: store_urbr: urbr: [seq:13]	
00000053	26.72193909	usbip_vhci: submit_urbr: urb requested: status:STATUS_PENDING	
00000054	26.72194099	usbip_vhci: cancelled urb destroyed: [seq:11]	
00000055	26.72195244	usbip_vhci: irp will be cancelled: FFFFC30614C5D270	
00000056	26.72195435	usbip_vhci: submit_urbr: urb pending	
00000057	26.72195435	usbip_vhci: cancelled urb destroyed: [seq:12]	
00000058	26.72198296	usbip_vhci: vhci_internal_ioctl: Enter	
00000059	26.72198486	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000060	26.72198677	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CONTROL_TRANSFER	
00000061	26.72199059	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136	
00000062	26.72199440	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	
00000063	26.72209930	usbip_vhci: vhci_read: Enter	
00000064	26.72210312	usbip_vhci: store_urbr: urbr: [seq:14]	
00000065	26.72210693	usbip_vhci: vhci_read: Leave: STATUS_SUCCESS	
00000066	26.72215271	usbip_vhci: vhci_read: Enter	
00000067	26.72215271	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00000068	26.73599434	usbip_vhci: vhci_write: Enter	
00000069	26.73599625	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00000070	26.75753403	usbip_vhci: vhci_write: Enter	
00000071	26.75753784	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	

here is the log from usbip server:
usbip_server: recv USBIP_CMD_SUBMIT
usbip_server: submit urb11 IN2 4096 bytes
usbip_server: recv USBIP_CMD_SUBMIT
usbip_server: submit urb12 IN1 10 bytes
usbip_server: recv USBIP_CMD_UNLINK
usbip_server: unlink urb11 IN2
usbip_server: recv USBIP_CMD_UNLINK
usbip_server: unlink urb12 IN1

IN2 is cdc data intput endpoint and IN1 is cdc event input endpoint.
There is some URB_FUNCTION_CONTROL_TRANSFER is not handled.
Maybe they are requests like SET_LINKE_CODING, GET_LINE_CODING, SET_CONTROL_LINE_STATE, GET_CONTROL_LINE_STATE of CDCACM.

versaloon, 2020-02-10:
I found a similar issue here:
https://sourceforge.net/p/usbip/mailman/message/29168927/

versaloon, 2020-02-11:
I also tried MSC BOT, works perfectly. Although MSC BOT has class-specific requests, but seems not affected by this(or maybe not use URB_FUNCTION_CONTROL_TRANSFER).

versaloon, 2020-02-11:
I can provide a test environment of this BUG, and it doesn't need a usb cdc hardware.
I implemented a usbip_dcd(device controller driver) to use my usb device stack and application, so it can implment a cdc device with pure software.

versaloon, 2020-02-15:
I have release my test environment:
https://github.com/vsfteam/vsf
Use visual studio to open "source/example/template/project/vc/vsf_template.sln".
Select x86 platform(x64 is not supported yet).
In userapp/config/vsf_usr_cfg.h, comment USBD_DEMO_USE_MSC and uncomment USBD_DEMO_USE_CDC.
Compile and run, enter command: usbd.
Open power shell and run:
./usbip list -r 127.0.0.1
./usbip attach -r 127.0.0.1 -b 0

versaloon, 2020-02-15:
2 cdc devices will be available, but if try to open it with putty, the error will occur.

cezanne, 2020-03-30:
@versaloon : 
> https://github.com/vsfteam/vsf

I tried to build but failed. I was a little confused. Could you share the binary?
And I also tried to test a vhci driver via an [usbip-vstub](https://github.com/cezanne/usbip-vstub) similar to yours, which is derived from USBIP-Virtual-USB-Device project commented in #52.

versaloon, 2020-04-21:
[vsf_template.zip](https://github.com/cezanne/usbip-win/files/4509304/vsf_template.zip)
Sorry for the late, I uploaded the executable and a readme.txt

versaloon, 2020-04-21:
1. start vsf_template_win.exe
2. run one of commands: usbd_msc/usbd_cdc/usbd_uvc
3. run in windows: usbip list -r 127.0.0.1
4. run in windows: usbip attach -r 127.0.0.1 -b 0

usbd_msc works OK. usbd_cdc will fail while open the emulated device. usbd_uvc is not tested currently.

versaloon, 2020-04-21:
[image]
I tried the latest usbip, the error information is a bit different.
After I connect the CDC port using putty, the service side receive UNLINK commands.

cezanne, 2020-04-21:
@versaloon : I had the same result with the master branch (4825a7fb8d82d8d76f9e74a1c5c7663d77efa960). Because URB_FUNCTION_CONTROL_TRANSFER had not been implemented. Here is  
[patch](https://github.com/cezanne/usbip-win/files/4510201/control_transfer.diff.zip) for URB_FUNCTION_CONTROL_TRANSFER. The diff was generated against the master branch.

For an vsf emulated cdc, I got 2 COM ports after the patch had been applied. Is it a working case?

versaloon, 2020-04-21:
I apply the patch and re-compile usbip.exe.
use the usbip_vhci.sys and run usbip install.
then do the test.
the result is same:
unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136
vhci_internal_ioctl: Leave: STATUS_INVAlD_PARAMETER

I can get 2 CDC ports even without the patch, but fail when I tried to open the CDC port with putty.

cezanne, 2020-04-21:
@versaloon : You should check that a newer vhci driver is installed properly. One easy way to install is that "Disable first, copy a driver and re-enable". I attached the screenshot. It was tested by the uploaded vhci driver binary. _An error case at 663 line_ of my dbgview.exe  should be more investigated.

[image]

versaloon, 2020-04-22:
I remove the vhci device and re-install. It works with the 2-port CDC demo. I will do further test today.

cezanne, 2020-04-25:
@versaloon : Please check a new [control_transfer branch](https://github.com/cezanne/usbip-win/tree/control_transfer). This branch fixes a bug of a previously uploaded patch. The improvement is inspired by @svirot comment in #136.

versaloon, 2020-04-25:
OK, I will test it next Monday, BTW, can you provide the compiled vhci driver cause I have no DDK installed.

cezanne, 2020-04-25:
@versaloon :  Here it is. [usbip_vhci.zip](https://github.com/cezanne/usbip-win/files/4534054/usbip_vhci.zip)

versaloon, 2020-04-28:
@cezanne Thanks. I tested the driver, and it works OK.

cezanne, 2020-04-28:
@versaloon : Could you check the released [binaries(v0.1.0)](https://github.com/cezanne/usbip-win/releases/tag/v0.1.0) ?

versaloon, 2020-04-30:
@cezanne Tested, works OK.

cezanne, 2020-04-30:
@versaloon : Thanks... I'm going to close this issue.

closed 2020-04-30 by cezanne