cezanne/usbip-win #104: raspberry to win 10: DeviceIoControl failed: err: 0x57 on attach

ThereWereProblemsCreatingMyAccount, 2020-02-13:
Hi
Im trying to serve a device from raspberry pi 3b+ to win 10. I get  following log:
```
usbip: debug: C:\some path\usbip.c:147:[run_command] running command: `attach'

usbip: error: walker_devpath: invalid hw id: root\vbdenum

usbip: debug: C:\some path\usbip_vhci.c:60:[usbip_vhci_driver_open] device path: \\?\root#unknown#0000#{d35f7840-6a0c-11d2-b841-00c04fad5171}
usbip: debug: C:\some path\usbip_attach.c:58:[import_device] got free port 1
usbip: error: usbip_vhci_attach_device: DeviceIoControl failed: err: 0x57
usbip: error: import device
usbip: error: read_devbuf: failed to read: err: 0x6
```
below you'll find contents  of _ioctl_usbip_vhci_plugin struct:
[image]

koudis, 2020-02-22:
Hello,

I use RsPi 4 and it work well. Try to download kernel 4.19 and rebuild usbip modules with usbip/usbipd tools.

cezanne, 2020-03-30:
@ThereWereProblemsCreatingMyAccount : 
Win 10 kernel log is required to resolve your issue.

cezanne, 2020-04-21:
@ThereWereProblemsCreatingMyAccount : Please reopen if you want.

closed 2020-04-21 by cezanne