cezanne/usbip-win #105: about unhandled URB_FUNCTION_GET_STATUS_FROM_DEVICE and winusb

versaloon, 2020-02-13:
I'm testing to use winusb on the usb device generated by usbip client. I can installed the winusb driver, and can found the device. But fail when I try to read the device descriptor. Here is the log:

00000796	13.16030407	usbip_vhci: vhci_internal_ioctl: Enter	
00000797	13.16030693	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000798	13.16031075	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE	
00000799	13.16031361	usbip_vhci: store_urbr: urbr: [seq:1]	
00000800	13.16031647	usbip_vhci: store_urbr_submit: urbr: [seq:1]	
00000801	13.16031837	usbip_vhci: store_urbr_submit: urbr: [seq:1], func:URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE	
00000802	13.16032314	usbip_vhci: submit_urbr: urb requested: status:STATUS_PENDING	
00000803	13.16032600	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00000804	13.16040993	usbip_vhci: vhci_read: Enter	
00000805	13.16041183	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00000806	13.16452694	usbip_vhci: vhci_write: Enter	
00000807	13.16453075	usbip_vhci: process_urb_res: urbr:[seq:1], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
00000808	13.16453648	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00000809	13.16455460	usbip_vhci: vhci_internal_ioctl: Enter	
00000810	13.16456127	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000811	13.16456604	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE	
00000812	13.16457176	usbip_vhci: store_urbr: urbr: [seq:2]	
00000813	13.16457558	usbip_vhci: store_urbr_submit: urbr: [seq:2]	
00000814	13.16458130	usbip_vhci: store_urbr_submit: urbr: [seq:2], func:URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE	
00000815	13.16458988	usbip_vhci: submit_urbr: urb requested: status:STATUS_PENDING	
00000816	13.16459370	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00000817	13.16462421	usbip_vhci: vhci_read: Enter	
00000818	13.16462517	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00000819	13.18437576	usbip_vhci: vhci_write: Enter	
00000820	13.18437958	usbip_vhci: process_urb_res: urbr:[seq:2], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
00000821	13.18438435	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00000822	13.18439960	usbip_vhci: vhci_internal_ioctl: Enter	
00000823	13.18440151	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000824	13.18440437	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE	
00000825	13.18440819	usbip_vhci: store_urbr: urbr: [seq:3]	
00000826	13.18441010	usbip_vhci: store_urbr_submit: urbr: [seq:3]	
00000827	13.18441296	usbip_vhci: store_urbr_submit: urbr: [seq:3], func:URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE	
00000828	13.18441772	usbip_vhci: submit_urbr: urb requested: status:STATUS_PENDING	
00000829	13.18441963	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00000830	13.18445396	usbip_vhci: vhci_read: Enter	
00000831	13.18445492	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00000836	13.19409084	usbip_vhci: vhci_write: Enter	
00000837	13.19409561	usbip_vhci: process_urb_res: urbr:[seq:3], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
00000838	13.19410133	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00000839	13.19411087	usbip_vhci: vhci_internal_ioctl: Enter	
00000840	13.19411469	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00000841	13.19411755	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_GET_STATUS_FROM_DEVICE	
00000842	13.19412041	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_GET_STATUS_FROM_DEVICE: len: 136	
00000843	13.19412518	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER	

the first 3urb are good. But the 4th URB_FUNCTION_GET_STATUS_FROM_DEVICE is not supported, and STATUS_INVALID_PARAMETER is returned. So WinUsb_ControlTransfer returns false, and GetLastError says 0x57 which means INVALID_PARAMETER.

koudis, 2020-02-13:
URB_FUNCTION_GET_STATUS_FROM_DEVICE is currently not implemented :(.
I have problem with my modem which needs this call after it resets/starts iface. So I will try to implement this URB soon.

versaloon, 2020-02-13:
Many thanks!!!

cezanne, 2020-03-31:
@versaloon : Please check [get_status branch](https://github.com/cezanne/usbip-win/tree/get_status). 
 URB_FUNCTION_GET_STATUS_FROM_XXX's are added to simply return STATUS_SUCCESS.

cezanne, 2020-04-07:
> @versaloon : Please check get_status branch.
> URB_FUNCTION_GET_STATUS_FROM_XXX's are added to simply return STATUS_SUCCESS.

The get_status branch is merged into the  master branch.

cezanne, 2020-04-30:
@versaloon : Could you check this issue with v0.1.0 ? If this is resolved, please close.

closed 2020-05-06 by versaloon