cezanne/usbip-win #108: Cannot export Yubi USB Key on Windows server to Ubunt 16.04 client

kaplan2539, 2020-02-28:
Trying to attach Yubi USB Key connected to Windows 10 machine to a Ubuntu 16.04 client, I get the errors listed below.

Ubuntu 16.04 client:
```
$ /usr/lib/linux-tools-4.4.0-174/usbip list -r h1474y2                                                                                                               
Exportable USB devices
======================
 - h1474y2
      1-202: unknown vendor : unknown product (0000:0000)
           : \\?\usb#vid_1050&pid_0407&mi_00#7&370641c8&0&0000#{fb265267-c609-41e6-8eca-a20d92a833e6}
           : (Defined at Interface level) (00/00/00)

      1-240: unknown vendor : unknown product (0000:0000)
           : \\?\usb#vid_1050&pid_0407&mi_01#7&370641c8&0&0001#{fb265267-c609-41e6-8eca-a20d92a833e6}
           : (Defined at Interface level) (00/00/00)

       1-22: unknown vendor : unknown product (0000:0000)
           : \\?\usb#vid_1050&pid_0407&mi_02#7&370641c8&0&0002#{fb265267-c609-41e6-8eca-a20d92a833e6}
           : (Defined at Interface level) (00/00/00)

$ /usr/lib/linux-tools-4.4.0-174/usbip attach -r h1474y2 -b 1-22
usbip: error: recv op_common
usbip: error: query
```

Windows 10:
```
PS H:\usbip_win\dist> ./usbip bind -b 1-22
usbip: info: bind_device: bind device on busid 1-22: complete
PS H:\usbip_win\dist> .\usbipd -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: D:\a\usbip-win\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 172.21.21.159:39740
usbip: info: recv_pdu: received request: 0x8005 - list devices
usbip: error: get_devinfo: cannot open device: \\?\usb#vid_1050&pid_0407&mi_00#7&370641c8&0&0000#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 88. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_devinfo: cannot open device: \\?\usb#vid_1050&pid_0407&mi_01#7&370641c8&0&0001#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 155. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_devinfo: cannot open device: \\?\usb#vid_1050&pid_0407&mi_02#7&370641c8&0&0002#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 156. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 253. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 65. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 215. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: info: exportable devices: 3
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:90:[dump_usb_device] path                 = \\?\usb#vid_1050&pid_0407&mi_00#7&370641c8&0&0000#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:91:[dump_usb_device] busid                = 1-202
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:97:[dump_usb_device] Device(C/SC/P)       = (Defined at Interface level) (00/00/00)
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:99:[dump_usb_device] bcdDevice            = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:104:[dump_usb_device] Vendor/Product       = unknown vendor : unknown product (0000:0000)
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:106:[dump_usb_device] bNumConfigurations   = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:107:[dump_usb_device] bNumInterfaces       = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:109:[dump_usb_device] speed                = Unknown Speed
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:111:[dump_usb_device] busnum               = 1
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:112:[dump_usb_device] devnum               = ca
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:90:[dump_usb_device] path                 = \\?\usb#vid_1050&pid_0407&mi_01#7&370641c8&0&0001#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:91:[dump_usb_device] busid                = 1-240
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:97:[dump_usb_device] Device(C/SC/P)       = (Defined at Interface level) (00/00/00)
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:99:[dump_usb_device] bcdDevice            = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:104:[dump_usb_device] Vendor/Product       = unknown vendor : unknown product (0000:0000)
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:106:[dump_usb_device] bNumConfigurations   = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:107:[dump_usb_device] bNumInterfaces       = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:109:[dump_usb_device] speed                = Unknown Speed
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:111:[dump_usb_device] busnum               = 1
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:112:[dump_usb_device] devnum               = f0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:90:[dump_usb_device] path                 = \\?\usb#vid_1050&pid_0407&mi_02#7&370641c8&0&0002#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:91:[dump_usb_device] busid                = 1-22
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:97:[dump_usb_device] Device(C/SC/P)       = (Defined at Interface level) (00/00/00)
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:99:[dump_usb_device] bcdDevice            = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:104:[dump_usb_device] Vendor/Product       = unknown vendor : unknown product (0000:0000)
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:106:[dump_usb_device] bNumConfigurations   = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:107:[dump_usb_device] bNumInterfaces       = 0
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:109:[dump_usb_device] speed                = Unknown Speed
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:111:[dump_usb_device] busnum               = 1
usbip: debug: D:\a\usbip-win\usbip-win\userspace\lib\usbip_common.c:112:[dump_usb_device] devnum               = 16
usbip: info: recv_pdu: request 0x8005: complete
usbip: info: connection from 172.21.21.159:39742
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: error: open_stub_dev: cannot open device: ▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌▌
usbip: error: export_device: cannot open devno: 22
usbip: error: failed to export device: 1-22, err:-1
usbip: info: recv_pdu: request 0x8003: failed
```

kaplan2539, 2020-03-02:
I investigated a little further, and in open_stub_dev(),userspace/src/usbipd/usbipd_stub.c, CreateFile() returns an invalid handle, LastError() returns 0x32 ERROR_NOT_SUPPORTED.

kaplan2539, 2020-03-02:
Could this be a permissions problem?

koudis, 2020-03-02:
Hello,

Ubuntu 16.04 is really old and is not supported anymore. Could you install Ubuntu 18.04 LTS and try it again?

The host driver is not stable and there are many errors. I suggest to use Linux as server.

kaplan2539, 2020-03-02:
Thanks for your reply. I'll try with an Ubuntu 18.04 LTS client next. However, for my use case I need the server to be on Windows 10.

kaplan2539, 2020-03-03:
I have tried with an Ubuntu 18.04 LTS client - I get the same error. The problem seems to be on the host side. Why would one get a  0x32 ERROR_NOT_SUPPORTED when calling win32 CreateFile() on the Windows server side?

cezanne, 2020-03-05:
@kaplan2539 : A stub driver in master works but not nicely as @koudis pointed out. :sweat_smile:
I had tested it with a simple USB mouse. Can you check the drivers after you bind Yubi ? I attach the screenshot of my machine. 

[image]

kaplan2539, 2020-03-16:
@cezanne thank you for your reply! The driver seems to be loaded. I've just tested the new v0.8 version. The `C:\Windows\System32\drivers\usbip_stub.sys` on my system had a different md5sum than the one from the v0.8 release. So I overwrote it with the latest version.
How does this work? Does `usbip.exe` copy the `usbip_stub.sys` file to `C:\Windows\system32\drivers` if it doesn't exist?
The Yubi key is a composite device, showing up as three different devices in `usbip list -l` on Windows. How are composite devices handled?

cezanne, 2020-03-16:
@kaplan2539 : 
> How does this work? Does usbip.exe copy the usbip_stub.sys file to C:\Windows\system32\drivers if it doesn't exist?

Yes, usbip.exe installs usbip_stub.sys via setupapi.dll if stub driver does not exist. So you should confirm that there's no stub driver if you want to install a new driver.

> The Yubi key is a composite device, showing up as three different devices in usbip list -l on Windows.

So I said that a stub driver doesn't work nicely.  😢 I'll check..

cezanne, 2020-03-18:
@kaplan2539 : "usbip list -l" problem was caused by wrong device detection method. Please, check [usb_dev_detection](https://github.com/cezanne/usbip-win/tree/usb_dev_detection). I alose attached binaries [usbip.zip](https://github.com/cezanne/usbip-win/files/4346542/usbip.zip). 
However, handling a composite by stub seems to raise BSOD. :cry:

cezanne, 2020-03-19:
usb_dev_detection branch merged into the master.

cezanne, 2020-04-07:
@kaplan2539 : Please reopen if you want.

closed 2020-04-07 by cezanne