cezanne/usbip-win #110: [Windows server]error: failed to unbind device

DaoYB033, 2020-03-06:
F:\Server>usbip list -l
 - busid 1-233 (04b3:3107)
   IBM Corp. : ThinkPad 800dpi Optical Travel Mouse (04b3:3107)

F:\Server>usbip bind -b 1-233
usbip: info: bind_device: bind device on busid 1-233: complete

F:\Server>usbip unbind -b 1-233
usbip: error: failed to unbind device
==========================================
This device is ok in bind, attach and detach, but error in unbind. Why?

cezanne, 2020-03-14:
@DaoYB033 : I slightly modified the format of your comment for better readability.

Unfortunately, usbip.exe does not provide friendly error messages. I added some code patch for usbip.exe. Please, try this [unbind.zip](https://github.com/cezanne/usbip-win/files/4332389/unbind.zip) !! The zip has diff(against master branch(358cabe)) and usbip.exe.

closed 2020-03-15 by DaoYB033

cezanne, 2020-03-15:
@DaoYB033 : Is your problem resolved? Please share your experience.