cezanne/usbip-win #114: Query unknown interface？

mankitty, 2020-03-14:
One of my devices seems to be wrong during the initialization of vhci_pnp. The previous initialization seems to be okay, but when it is about to end, an error occurs，
00000402	150.21456909	usbip_vhci: vhci_pnp: Enter	
00000403	150.21456909	usbip_vhci: vpdo: minor: IRP_MN_QUERY_INTERFACE, IRP: 0xFFFFFA800EF9D8A0	
00000404	150.21458435	usbip_vhci: Query unknown interface GUID: 20B34747CE62CF11A5D628DB04C10000	
00000405	150.21458435	usbip_vhci: vhci_pnp: Leave: STATUS_NOT_SUPPORTED	
00000406	150.21568298	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	

The complete log file is in the attachment.

Can anyone help me?thank you

and I want to ask, in the following code
if (!IsEqualGUID(interfaceType, (PVOID)&USB_BUS_INTERFACE_USBDI_GUID)){
		DBGI(DBG_GENERAL, "Query unknown interface GUID: %s\n", dbg_GUID(interfaceType));
		return Irp->IoStatus.Status;
}
// {B1A96A13-3DE0-4574-9B01-C08FEAB318D6}
DEFINE_GUID(USB_BUS_INTERFACE_USBDI_GUID, 
0xb1a96a13, 0x3de0, 0x4574, 0x9b, 0x1, 0xc0, 0x8f, 0xea, 0xb3, 0x18, 0xd6);

DEFINE_GUID Is this fixed? Why is this necessary?
[init.LOG](https://github.com/cezanne/usbip-win/files/4332493/init.LOG)

thank you very much !

cezanne, 2020-03-14:
@mankitty : 
> DEFINE_GUID Is this fixed? Why is this necessary?

vhci driver should support USB_BUS_INTERFACE_USBDI_GUID. Because it's a bus for virtual devices. 

Anyway, your problem seems to be related with that there's no proper function driver. Can you compare with the case your device is directly connected without usbip-win?

mankitty, 2020-03-14:
> @mankitty :
>
>
> vhci driver should support USB_BUS_INTERFACE_USBDI_GUID. Because it's a bus for virtual devices.
>
> Anyway, your problem seems to be related with that there's no proper function driver. [...]

Yes, I tested this device separately on a physical machine (window operating system and Ubuntu operating system) (not through the usbip-win project), it is fine, and I also tested the device through two Linux operating systems The USB redirection function is all possible, but it seems that I have the problem described above through the usbip-win project. I wonder if there is a problem with me?thank you for your confusion

cezanne, 2020-03-15:
@mankitty : More detailed information is required to resolve your issue. Please share USB device properties when a device are connected with and without usbip-win. For clarity, I'll give a screenshot when a USB mice is connected via usbip-win.  

[image]

mankitty, 2020-03-16:
> @mankitty : More detailed information is required to resolve your issue. Please share USB device properties when a device are connected with and without usbip-win. For clarity, I'll give a screenshot when a USB mice is connected via usbip-win.
> 
> [image]

Sorry, because I could n’t get the device yesterday, I got your message late，the following is the detailed information of my pass usbip-win and not usbip-win, please check

not pass usbip-win
[image]
and pass usbip-win
[image]

mankitty, 2020-03-16:
> Sorry, because I could n’t get the device yesterday, I got your message late，the following is the detailed information of my pass usbip-win and not usbip-win, please check
>
> not pass usbip-win [...]

seems no different

cezanne, 2020-03-16:
@mankitty : How are they displayed for an upper device(maybe pdo)? You can easily find the device in "device by connection" of device manager.
 
[image]

mankitty, 2020-03-18:
> @mankitty : How are they displayed for an upper device(maybe pdo)? You can easily find the device in "device by connection" of device manager.
> 
> [image]

I ca n’t use the device now, so I wo n’t make it for the time being, thank you for your help

cezanne, 2020-03-31:
@mankitty : This issue will be closed. Please reopen if you need.

closed 2020-03-31 by cezanne