cezanne/usbip-win #116: 【Windows server】blue screen on death

DaoYB033, 2020-03-15:
When testing with U-disk, the Windows server must have a blue screen. Does this project support U-disk?
ps: the Linux server is OK.

DaoYB033, 2020-03-15:
after attach complete

cezanne, 2020-03-16:
@DaoYB033 : USB disks are supported.

Jesuslcp, 2020-03-17:
Hello!! Great job for all of you doing this, thanks for your work.

I tried the 0.0.8 source code compiled with VS 2019 , working on W7 x64 as server and W10 x64 as client. ( Two different compiles for each OS).

Works initially fine capturing a usb joystick with no problems over LAN, but when you tried to disconnect the client and server the problem starts, client side I pressed CTRL+C nothing happens, then in server do the same an then, blue screen and death with a message related to the Stub driver (W7).

I will provide further info when I register the fail, but looks like the detachment of the stub driver after a coms fail or disconnect need to be depured, present code slap on my face before I realise nothing else to say by now.

As recommendation, I would say that to understand the project by now is a bit complicate to understand how works for an average user/developer, so I would say to split the project in Client side and Server side could be a good idea to quickly catch up the problems and don't mix them.

Regards!

koudis, 2020-03-17:
Hello,
the BSOD error message is "Multipre IRQ complete request"? If yes the problem with BSOD can be solved by #120.

cezanne, 2020-03-20:
@Jesuslcp :

> As recommendation, I would say that to understand the project by now is a bit complicate to understand how works for an average user/developer, so I would say to split the project in Client side and Server side could be a good idea to quickly catch up the problems and don't mix them.

Great idea!! But a single project approach would be better.
usbip-win has tried to organize the codes. Doesn't it?
And  both drivers and tool share much codes like as linux distro.

Jesuslcp, 2020-03-21:
That is what I wanted to mean, one single project but split server and client... Easy to say but let see the work it takes...

Jesuslcp, 2020-03-21:
I have been testing the past day the release branch Stup_fix... Everything works this time after attach, detach, CTRL+C, bind unbind, net interruption with no BSOD at all by know.

cezanne, 2020-03-31:
@Jesuslcp : Thanks for your report.

cezanne, 2020-03-31:
@DaoYB033 : Please retry with [v0.0.9](https://github.com/cezanne/usbip-win/releases/tag/v0.0.9) in the release page.

cezanne, 2020-04-07:
@DaoYB033 : Please reopen if you need.

closed 2020-04-07 by cezanne