cezanne/usbip-win #122: Windows client crashes with BSOD when detaching a Xbox one controller.

deckerd007, 2020-03-17:
An Xbox One controller works well by "ubsip attach" command.
(Server: Windows 10, Client: Windows 10)

When using the "usbip detach" command to detach the Xbox one controller, the client always crashes with BSOD.

Used source commit : 141e09a4447a86ef2a5a7f2f8fe8468104dfba55 [141e09a]
(RB_FUNCTION_CONTROL_TRANSFER_EX Implemented)

The output kernel logs were as follows:

=============================================
usbip_vhci: vhci_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_USBIP_VHCI_UNPLUG_HARDWARE
usbip_vhci: Plugging out single device: port: 1
usbip_vhci: found device: port: 1
usbip_vhci: Plugging out: port: 1
usbip_vhci: finish pending irp
usbip_vhci: irp will be cancelled: FFFFCE0220A42180
usbip_vhci:(WW) no matching urbr
usbip_vhci: vhci_internal_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_GET_PORT_STATUS
usbip_vhci:(WW) device is not connected
usbip_vhci: vhci_pnp: Enter
usbip_vhci: vhub: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP:0xFFFFCE022414EC70
usbip_vhci: 	QueryDeviceRelation Type: BusRelations
usbip_vhci: sleep 50ms, let pnp manager send irpusbip_vhci: vhci_read: Enter
usbip_vhci: vhci_read: Leave: STATUS_INVALID_DEVICE_REQUEST
usbip_vhci: vhci_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_USBIP_VHCI_UNPLUG_HARDWARE
usbip_vhci: Device 1 plug out finished
usbip_vhci: # of vpdo's: present: 1, reported: 0
usbip_vhci: vhci_ioctl: Leave: STATUS_SUCCESS
usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
usbip_vhci: Plugging out single device: port: 1
usbip_vhci: found device: port: 1
usbip_vhci: Plugging out: port: 1
usbip_vhci: finish pending irp
usbip_vhci: Device 1 plug out finished
usbip_vhci: vhci_ioctl: Leave: STATUS_SUCCESS
usbip_vhci: vhci_cleanup: Enter
usbip_vhci: vhci_cleanup: Leave
usbip_vhci: vhci_cleanup: Enter
usbip_vhci: vhci_cleanup: Leave
usbip_vhci: vhci_pnp: Enter
usbip_vhci: vpdo: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFCE0227A60CB0
usbip_vhci: QueryDeviceRelation Type: RemovalRelations
usbip_vhci: vhci_pnp: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: vhci_pnp: Enter
usbip_vhci: vpdo: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFCE0227A60CB0
usbip_vhci: QueryDeviceRelation Type: EjectionRelations
usbip_vhci: vhci_pnp: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: vhci_internal_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB
usbip_vhci:(WW) device is not connected
usbip_vhci: vhci_internal_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB
usbip_vhci:(WW) device is not connected
usbip_vhci: vhci_pnp: Enter
usbip_vhci: vpdo: minor: IRP_MN_SURPRISE_REMOVAL, IRP: 0xFFFFCE0227A60CB0
usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
usbip_vhci: vhci_pnp: Enter
usbip_vhci: vpdo: minor: IRP_MN_REMOVE_DEVICE, IRP: 0xFFFFCE0227A60CB0
usbip_vhci: Deleting vpdo: 0xFFFFCE022569EE10
usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
KDTARGET: Refreshing KD connection

*** Fatal System Error: 0x00000044
                       (0xFFFFCE0220A42180,0x0000000000001232,0x0000000000000000,0x0000000000000000)

Break instruction exception - code 80000003 (first chance)

A fatal system error has occurred.
Debugger entered on first try; Bugcheck callbacks have not been invoked.

A fatal system error has occurred.

For analysis of this file, run !analyze -v
nt!DbgBreakPointWithStatus:
fffff803`295cd370 cc              int     3
============================================

koudis, 2020-03-17:
Hello, the #120 can solve this problem (it cleanly cancel all read IRP)

deckerd007, 2020-03-18:
Hi, koudis
Thank you for advice.

I applied #119 and # 120, but the problem was not solved
The same error code came out. ; *** Fatal System Error: 0x00000044

The output kernel logs were pretty much the same as before.

koudis, 2020-03-18:
@deckerd007 the BSOD is MULTIPLE_IRQ_COMPLETE_REQUEST?

deckerd007, 2020-03-18:
@koudis , Yes. you are right, the BSOD is MULTIPLE_IRQ_COMPLETE_REQUEST.

bartlomiejcieszkowski, 2020-03-18:
i've created patch that will add some tracking info about irp completion - #125 ,
can you try getting logs with this one applied?

bartlomiejcieszkowski, 2020-03-18:
i've tried reproducing it locally

1st i tried surprise removal with a simple logitech mouse, it was a success
then i tried surprise removal with xbox one controller and at the master i've hit KERNEL_SECURITY_CHECK_FAILURE :(

deckerd007, 2020-03-19:
Hi, @bartlomiejcieszkowski, 

After applying # 125, the following logs were obtained.

====================================
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci.c@94IoCompleteRequest FFFFD501A8AEA9A0 boost? 0usbip_vhci: vhci_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_USBIP_VHCI_UNPLUG_HARDWARE
usbip_vhci: Plugging out single device: port: 1
usbip_vhci: found device: port: 1
usbip_vhci: Plugging out: port: 1
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_pnp.c@630IoCompleteRequest FFFFD501A64EC6A0 boost? 0usbip_vhci: finish pending irp
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_pnp.c@683IoCompleteRequest FFFFD501A2AFA050 boost? 0usbip_vhci: vhci_pnp: Enter
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_pnp.c@683IoCompleteRequest FFFFD501A2B17CD0 boost? 0usbip_vhci: vhub: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP:0xFFFFD501A60AE660
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_pnp.c@683IoCompleteRequest FFFFD501A2D26070 boost? 0usbip_vhci: 	QueryDeviceRelation Type: BusRelations
usbip_vhci: irp will be cancelled: FFFFD501A80FA6F0
usbip_vhci:(WW) no matching urbr
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\usbreq.c@144IoCompleteRequest FFFFD501A80FA6F0 boost? 0usbip_vhci: sleep 50ms, let pnp manager send irpusbip_vhci: vhci_internal_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_GET_PORT_STATUS
usbip_vhci:(WW) device is not connected
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_ioctl.c@166IoCompleteRequest FFFFD501A80164A0 boost? 0usbip_vhci: vhci_read: Enter
usbip_vhci: vhci_read: Leave: STATUS_INVALID_DEVICE_REQUEST
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_read.c@870IoCompleteRequest FFFFD501A606D660 boost? 0usbip_vhci: vhci_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_USBIP_VHCI_UNPLUG_HARDWARE
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_pnp.c@683IoCompleteRequest FFFFD501A80FA6F0 boost? 0usbip_vhci: Device 1 plug out finished
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_ioctl.c@276IoCompleteRequest FFFFD501A8AEA9A0 boost? 0usbip_vhci: # of vpdo's: present: 1, reported: 0
usbip_vhci: vhci_ioctl: Leave: STATUS_SUCCESS
usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
usbip_vhci: Plugging out single device: port: 1
usbip_vhci: found device: port: 1
usbip_vhci: Plugging out: port: 1
usbip_vhci: finish pending irp
usbip_vhci: Device 1 plug out finished
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_ioctl.c@276IoCompleteRequest FFFFD501A606D660 boost? 0usbip_vhci: vhci_ioctl: Leave: STATUS_SUCCESS
usbip_vhci: vhci_cleanup: Enter
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci.c@144IoCompleteRequest FFFFD501A9039760 boost? 0usbip_vhci: vhci_cleanup: Leave
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci.c@184IoCompleteRequest FFFFD501A9039760 boost? 0usbip_vhci: vhci_cleanup: Enter
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci.c@144IoCompleteRequest FFFFD501A9039760 boost? 0usbip_vhci: vhci_cleanup: Leave
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci.c@184IoCompleteRequest FFFFD501A9039760 boost? 0usbip_vhci: vhci_pnp: Enter
usbip_vhci: vpdo: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFD501A9039760
usbip_vhci: QueryDeviceRelation Type: RemovalRelations
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_vpdo.c@807IoCompleteRequest FFFFD501A9039760 boost? 0usbip_vhci: vhci_pnp: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: vhci_pnp: Enter
usbip_vhci: vpdo: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFD501A9039760
usbip_vhci: QueryDeviceRelation Type: EjectionRelations
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_vpdo.c@807IoCompleteRequest FFFFD501A9039760 boost? 0usbip_vhci: vhci_pnp: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: vhci_internal_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB
usbip_vhci:(WW) device is not connected
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_ioctl.c@166IoCompleteRequest FFFFD501A80FA820 boost? 0usbip_vhci: vhci_internal_ioctl: Enter
usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB
usbip_vhci:(WW) device is not connected
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_ioctl.c@166IoCompleteRequest FFFFD501A80FA820 boost? 0usbip_vhci: vhci_pnp: Enter
usbip_vhci: vpdo: minor: IRP_MN_SURPRISE_REMOVAL, IRP: 0xFFFFD501A9039760
usbip_vhci: D:\GameGrid\usbip\Windows\srcs\usbip-win-c\driver\vhci\vhci_vpdo.c@807IoCompleteRequest FFFFD501A9039760 boost? 0usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
KDTARGET: Refreshing KD connection

*** Fatal System Error: 0x00000044
                       (0xFFFFD501A80FA6F0,0x000000000000257C,0x0000000000000000,0x0000000000000000)

Break instruction exception - code 80000003 (first chance)

A fatal system error has occurred.
Debugger entered on first try; Bugcheck callbacks have not been invoked.

A fatal system error has occurred.

nt!DbgBreakPointWithStatus:
fffff801`71670210 cc              int     3
===========================================

deckerd007, 2020-03-19:
@bartlomiejcieszkowski,

When I pulled out the Xbox One controller from the server, there was no problem.
I just got the following printed message in the client "usbip attach" console.
```
C:\usbip-win>usbip attach -r 192.168.0.61 -b 1-12
usbip: error: read_devbuf: failed to read: err: 0x40
usbip: error: read_devbuf: could the client have dropped the connection?

C:\usbip-win>
```
However, the "usbip detach" command on the client failed and crashed with a BSOD. (MULTIPLE_IRQ_COMPLETE_REQUEST)

bartlomiejcieszkowski, 2020-03-29:
@deckerd007 thanks for the logs

i've did on my development branch some cleanup in the Locks around the IRPs, so they wont be completed multiple times, and it looks good - tested locally with xbox one controller and did the detach path - I will be pushing next week a merge request once ill extract what is only needed :)

closed 2020-06-06 by cezanne