cezanne/usbip-win #123: Windows server - Windows client

bartek012, 2020-03-18:
Hello,
Is it even possible to use usbip in this configuration (Windows10 server - Windows10 client)? 

I have problem with installing vhci driver. I've tried to install it manually by Add Hardware option (usbip_vhci.inf). It says "The folder you specified doesn't contain a compatible software driver for your device. If the folder contains a driver, make sure it is designed to work with Windows for x64-based system". Am I choosing wrong file? 

I have used usbip-win 0.0.8 release and ran 
**server side**
usbip bind -b 1-2
usbipd -4- d 

**client side**
I can see binded device using "list -r <ip>" command
but when I try to attach it, there is an error : open vhci driver, failed to read err: 0x6

output from server console : 

usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 10.8.14.18:49959
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_import.c:18:[forwarder_stub] stub forwarding started
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_import.c:108:[recv_request_import] import request busid 1-2: complete
usbip: info: recv_pdu: request 0x8003: complete
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_import.c:28:[forwarder_stub] stub forwarding stopped

Thanks

Jesuslcp, 2020-03-18:
Yes, no problem to run W10 x64 and W10x64, I did with this version, please be sure you launch both sides with administrator privileges.

bartek012, 2020-03-19:
Yes, but what if none of the following doesn't work? You could install VHCI driver without any problem?

[image]

[image]

After choosing path C:\USBIP\driver\vhci

[image]

bartek012, 2020-03-20:
@Jesuslcp  which option worked in your case command line install or manual ? 

Is there any other option to install vhci driver?

Jesuslcp, 2020-03-20:
I installed both manually in W10 and W7, with no issues, except for W7 that I had to compile for this OS as the W10 driver compiled wasn't working properly.

I have to say that previous to compile, I ran an VS self analysis and depuration of the source code, detecting some issues  changed.

If you need it, I can provide my SC and compiled files to check.

bartek012, 2020-03-20:
@Jesuslcp I would be grateful for that files, Thanks. 
My email: bartk970802@gmail.com

Jesuslcp, 2020-03-20:
Here they are!!

[USBIP Source Code 0.0.8 MOD Jesuslcp.zip](https://github.com/cezanne/usbip-win/files/4359914/USBIP.Source.Code.0.0.8.MOD.Jesuslcp.zip)
[USBIP 0.0.8 MOD Jesuslcp Windows x64.zip](https://github.com/cezanne/usbip-win/files/4359915/USBIP.0.0.8.MOD.Jesuslcp.Windows.x64.zip)

Jesuslcp, 2020-03-20:
You are very welcome! Let us know if this compiled works also for you!

bartek012, 2020-03-20:
It works !!
your version of usbip_vhci.inf file passed through manual install.
Problem solved thank you again. 

Differences between files  (left side is working)
[image]

Jesuslcp, 2020-03-20:
Great! Will have a look to that when I found the moment.

Jesuslcp, 2020-03-21:
Well, after have a look to the problem with the driver, there's no problem. Simply you have taken the driver directly from the source code, but usbip_vhci.inf is modified to the correct architecture during the built process.

cezanne, 2020-04-07:
@bartek012 @Jesuslcp : This issue seems to be resolved. 😃 I'm going to close.

closed 2020-04-07 by cezanne