cezanne/usbip-win #127: Long usbip_vhci build time

bartlomiejcieszkowski, 2020-03-18:
Build itself is not so long, but the inf2cat takes forever -_-"

Initial investigation shows that this is because the inf2cat parses whole folder, and it parses without any point obj and tmh files.

Switching to driver package (which is a fancy project that only does inf2cat / signing) runs only on the usbip_vhci.sys and usbip_vhci.inf - and completes in mere seconds :)

PR wip

cezanne, 2020-03-31:
@bartlomiejcieszkowski : Indeed, too slow vhci build drops quality of life. I have been used to remove a usbip_vhci intermediate directory before compiling vhci driver. #121 is good to work but it requires an additional project. So I devised another approach. Please check [fast_vhci_build branch](https://github.com/cezanne/usbip-win/tree/fast_vhci_build). A batch script at build post event will resolve the issue.

bartlomiejcieszkowski, 2020-03-31:
@cezanne thats simpler, i dont like either an extra project
 i took a look at the branch, and this is ok :) 
just move:
`signtool sign usbip_vhci.cat`
 before
`copy /y usbip_vhci.cat ..`

cezanne, 2020-03-31:
@bartlomiejcieszkowski : Oops, my fault. fast_vhci_build branch was fixed as you pointed out. 👍

cezanne, 2020-04-02:
4c99013a3be15ea5fcb0511aef6929d3dad69c55 resolves this issue. Closed.

closed 2020-04-02 by cezanne