cezanne/usbip-win #128: windows server - linux client, error

nebakanezzer, 2020-03-29:
windows server 2016 (server) 192.168.1.21
debian buster (client) 192.168.1.41

windows logs:

C:\Users\Administrator.Violet\Desktop\usbip-win-master>usbip.exe list -l
usbip: error: failed to open usb id database
usbip: error: add_usbdev: invalid hw id: USB\DEVICE_DESCRIPTOR_FAILURE
 - busid 1-81 (1c4f:0002)
   unknown vendor : unknown product (1c4f:0002)
 - busid 1-189 (03f0:1027)
   unknown vendor : unknown product (03f0:1027)
 - busid 1-151 (03f0:1027)
   unknown vendor : unknown product (03f0:1027)
 - busid 1-119 (1c4f:0002)
   unknown vendor : unknown product (1c4f:0002)
 - busid 1-85 (0658:0200)
   unknown vendor : unknown product (0658:0200)

C:\Users\Administrator.Violet\Desktop\usbip-win-master>usbip.exe bind -b 1-85
usbip: info: bind_device: bind device on busid 1-85: complete

C:\Users\Administrator.Violet\Desktop\usbip-win-master>usbipd.exe -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 192.168.1.41:60236
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: open_stub_dev: invalid devno: 85
usbip: error: export_device: cannot open devno: 85
usbip: error: failed to export device: 1-85, err:-1
usbip: info: recv_pdu: request 0x8003: failed

linux client error:

root@Violet-Vm-HA-01:/home/warmachine# usbip attach -r 192.168.1.21 -b 1-85
usbip: error: Attach Request for 1-85 failed - Request Failed

cezanne, 2020-03-30:
@nebakanezzer : Which version did you test ? Binaries in a release page would suffer from your case. But master branch would not.

nebakanezzer, 2020-03-30:
cloned from the main page here. https://github.com/cezanne/usbip-win.git

on the linux client I did a 'sudo apt-get install usbip' so not sure what version, I'd have to check

cezanne, 2020-03-31:
@nebakanezzer : 
> cloned from the main page here. https://github.com/cezanne/usbip-win.git

Is your master branch ahead of 5e83ddd5224ee343c25eac5c94af571c7981b348 ?
5e83ddd5224ee343c25eac5c94af571c7981b348 commit would resolve your case.

nebakanezzer, 2020-03-31:
583ddd gives me error as well:

C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbip.exe list -l
 - busid 1-105 (03f0:1027)
   Hewlett-Packard : Virtual keyboard and mouse (03f0:1027)
 - busid 1-85 (0658:0200)
   Sigma Designs, Inc. : unknown product (0658:0200)
 - busid 1-202 (0951:1689)
   Kingston Technology : DataTraveler SE9 (0951:1689)

C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>
C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>
C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbip.exe bind -b 1-85
usbip: info: bind_device: bind device on busid 1-85: complete

C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbipd.exe -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 192.168.1.41:37618
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: open_stub_dev: invalid devno: 85
usbip: error: export_device: cannot open devno: 85
usbip: error: failed to export device: 1-85, err:-1
usbip: info: recv_pdu: request 0x8003: failed

I've tried the folders found where the exe's reside and had similar issues (this appears to be version 0.0.9):

C:\Windows\System32>cd C:\Users\Administrator.Violet\Desktop\usbip-win-0.0.9

C:\Users\Administrator.Violet\Desktop\usbip-win-0.0.9>usbip.exe list -l
 - busid 1-105 (03f0:1027)
   Hewlett-Packard : Virtual keyboard and mouse (03f0:1027)
 - busid 1-85 (0658:0200)
   Sigma Designs, Inc. : unknown product (0658:0200)
 - busid 1-202 (0951:1689)
   Kingston Technology : DataTraveler SE9 (0951:1689)

C:\Users\Administrator.Violet\Desktop\usbip-win-0.0.9>usbip.exe bind -b 1-85
usbip: info: bind_device: bind device on busid 1-85: complete

C:\Users\Administrator.Violet\Desktop\usbip-win-0.0.9>usbipd.exe -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 192.168.1.41:37418
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: open_stub_dev: invalid devno: 85
usbip: error: export_device: cannot open devno: 85
usbip: error: failed to export device: 1-85, err:-1
usbip: info: recv_pdu: request 0x8003: failed

on the linux end, error is always the same: 

root@Violet-Vm-HA-01:/home/warmachine# usbip attach -r 192.168.1.21 -b 1-85
usbip: error: Attach Request for 1-85 failed - Request Failed

cezanne, 2020-03-31:
> one thing to note is there are no exe in the 5e83ddd version, so I followed the 'readme' to the 009 page for those. should I not be using those? if not, which versions should I be using?

v0.0.9 would be fine.. It's an up-to-date version to resolve this issue.
I often suffered from your problem. In my case, usbipd detected a usbip device after several times repetition of bind and unbind. Can you try ?

And can you share the result of lsusb command in a linux machine?

nebakanezzer, 2020-04-01:
it fails to unbind. I bound it again after the failure anyway to see if that would work, it did not:

`C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbip.exe bind -b 1-85
usbip: info: bind_device: bind device on busid 1-85: complete

C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbip.exe unbind -b 1-85
usbip: error: failed to rollback driver: 103
usbip: error: failed to unbind device

C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbip.exe unbind -b 1-85
usbip: error: failed to rollback driver: 103
usbip: error: failed to unbind device

C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbip.exe bind -b 1-85
usbip: info: bind_device: bind device on busid 1-85: complete

C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbipd.exe -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 192.168.1.41:37188
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: open_stub_dev: invalid devno: 85
usbip: error: export_device: cannot open devno: 85
usbip: error: failed to export device: 1-85, err:-1
usbip: info: recv_pdu: request 0x8003: failed
`

lsusb on linux:

`root@Violet-Vm-HA-01:/home/warmachine# lsusb
Bus 017 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 016 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 015 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 014 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 013 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 012 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 011 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 010 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

cezanne, 2020-04-01:
@nebakanezzer : lsusb listing is not required. I was confused and asked you a useless request. Please excuse me.
How about checking that a stub device is successfully created after binding completes? A stub device might not be created for unknown reasons such as a usbip bug or other drivers may remove the stub device.

Your logs give some clue.
> `usbip: error: failed to rollback driver: 103`

0x103 unbinding error indicates that there's no driver to  rollback. This means that a stub device may be removed at unbinding.

nebakanezzer, 2020-04-02:
there are currently 2 devices showing in device manager labeled 'usb stub'

these remain whether I bind the device or not

cezanne, 2020-04-02:
> there are currently 2 devices showing in device manager labeled 'usb stub'
> 
> these remain whether I bind the device or not

Please remove all stub devices including hidden stub devices  before you're going to bind. Device manager can display hidden stub devices if "show hidden devices" is checked in view menu.

nebakanezzer, 2020-04-02:
only way to remove is uninstall. once complete, same results:

C:\Users\Administrator.Violet\Desktop\usbip-win-5e83ddd5224ee343c25eac5c94af571c7981b348>usbipd.exe -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 192.168.1.41:33686
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: open_stub_dev: invalid devno: 85
usbip: error: export_device: cannot open devno: 85
usbip: error: failed to export device: 1-85, err:-1
usbip: info: recv_pdu: request 0x8003: failed

cezanne, 2020-04-02:
@nebakanezzer : 
> usb device also no longer shows in device manager, is this normal?

Do you mean that a stub device does not exist in device manager when attaching ?  If yes, it's not normal.

nebakanezzer, 2020-04-24:
the stub device exists but the original device does not show.

if I uninstall usb stub, physically unplug and plug the device back in, the original driver shows.

I ended up just using this instead, which worked without issue: https://www.virtualhere.com/windows_server_software

xiaolin0199, 2020-04-26:
Hello, I also encountered the same problem

> usbip: error: failed to rollback driver: 103
> usbip: error: failed to unbind device

So, do you think this is a bug? When will it be possible to fix it?

xiaolin0199, 2020-04-26:
After recompiling with the latest code, I can solve this problem, thank you！

closed 2020-06-06 by cezanne