cezanne/usbip-win #131: Feature request: register for sleep request in usbip.exe

bartlomiejcieszkowski, 2020-03-31:
Backstory: I'm using laptop for debug, it has 30 minute sleep time. So it goes into sleep, i wake it up. From the Windbg i can see that the device was there when we go to sleep. But once it goes out of sleep, it will terminate - due to a read error in the usbip.exe, yeah, no wonder, sockets and going to sleep, what can go wrong? ;)

```
usbip: error: read_devbuf: failed to read: err: 0x40 
usbip: error: read_devbuf: could the client have dropped the connection?

```

This wouldn't be so troublesome, if the device didn't leave in some weird state on the server -_-" so had to do unbind/bind device to get it back working normally.

So my proposal is to register for power event in attach, just as we do signal interrupt register for ctrl-c and gracefuly close the connection, and not leave server without goodbye :)

Reference documentation:
https://docs.microsoft.com/en-us/windows/win32/power/registering-for-power-events

cezanne, 2020-04-07:
@bartlomiejcieszkowski : It's an interesting issue. But network connection would  be lost even though vhci manages to establish connection state. Thus, a solution might be a matter of reconnection.
I don't know how LINUX thinks about reconnection.. 😸

bartlomiejcieszkowski, 2020-04-07:
yes, it's lost, but *after* the sleep we get the unplug..
i'm doing this with linux server - and the observed weird acting was on linux server - might need to check the usbip server code for that

nevertheless 'nice' behavior would be appreciated :)