cezanne/usbip-win #133: build fail - cat - Multiple certificates were found that meet all the given

bartlomiejcieszkowski, 2020-04-02:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error		Multiple certificates were found that meet all the given	usbip_vhci	Z:\Development\github\usbip-win\driver\vhci\EXEC	1	
Error	MSB3073	The command "gencat.bat Z:\Development\github\usbip-win\Debug\x64\ 10_x64
:VCEnd" exited with code -1.	usbip_vhci	Z:\Microsoft\VisualStudio2019\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets	149	

```

I'm running into this error with the latest bat.
My guess thats because certificate is selected in visual studio and installed on the os, so signtool doesnt know what to pick...

I recall that i tried before making package also a try with method from here https://social.msdn.microsoft.com/Forums/lync/en-US/9e65eb3f-4ced-430d-a607-ad6c28ec412b/making-cab-of-driver-package?forum=wdk

pre-build - remove logs, and 
AfterTargets="Inf2Cat" with xcopy of cat only should be sufficient

bartlomiejcieszkowski, 2020-04-02:
ah, nope, as there are a lot of artifacts, looks like a package one is the sane approach

mohammad362, 2020-04-05:
i have same problem.

thruhtz, 2020-04-05:
I had that multiple certificate problem as well

The following description solved the problem by deleting the multiple test certificates

https://infosys.beckhoff.com/content/1033/tc3_c/63050394893880971.html

koudis, 2020-04-11:
Hello, i create PR for this
https://github.com/cezanne/usbip-win/pull/138

koudis, 2020-04-19:
Merged. @bartlomiejcieszkowski can you test the master branch pls? So we can close the issue.

bartlomiejcieszkowski, 2020-04-19:
sure

bartlomiejcieszkowski, 2020-04-19:
@koudis no longer seeing the issue, closing :)

closed 2020-04-19 by bartlomiejcieszkowski