cezanne/usbip-win #134: Stress test - attach detach - PNP_DETECTED_FATAL_ERROR (ca)

bartlomiejcieszkowski, 2020-04-02:
with added #129 

when i'm doing fast attach detach of 3 devices (it's a stress test) i can get into fail on IoInvalidateRelation

```
usbip_vhci:(EE) why we are not present
usbip_vhci:(EE) why we are not present
usbip_vhci:(EE) why we are not present
KDTARGET: Refreshing KD connection

*** Fatal System Error: 0x000000ca
                       (0x0000000000000002,0xFFFFB88659E1D080,0x0000000000000000,0x0000000000000000)

Break instruction exception - code 80000003 (first chance)
nt!DbgBreakPointWithStatus:
fffff806`1d3ca210 cc              int     3
2: kd> !analyze -v
Connected to Windows 10 18362 x64 target at (Thu Apr  2 20:57:58.024 2020 (UTC + 2:00)), ptr64 TRUE
Loading Kernel Symbols
...............................................................
................................................................
................................................................
.....
Loading User Symbols
............................
Loading unloaded module list
...................
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

PNP_DETECTED_FATAL_ERROR (ca)
PnP encountered a severe error, either as a result of a problem in a driver or
a problem in PnP itself.  The first argument describes the nature of the
problem, the second argument is the address of the PDO.  The other arguments
vary depending on argument 1.
Arguments:
Arg1: 0000000000000002, Invalid PDO
	An API which requires a PDO has been called with either an FDO,
	a PDO which hasn't been initialized yet (returned to PnP in a
	QueryDeviceRelation/BusRelations), or some random piece of
	memory.
Arg2: ffffb88659e1d080, Purported PDO.
Arg3: 0000000000000000, Driver object.
Arg4: 0000000000000000

Debugging Details:
------------------

*** WARNING: Unable to verify checksum for usbip.exe

KEY_VALUES_STRING: 1

PROCESSES_ANALYSIS: 1

SERVICE_ANALYSIS: 1

STACKHASH_ANALYSIS: 1

TIMELINE_ANALYSIS: 1

DUMP_CLASS: 1

DUMP_QUALIFIER: 0

BUILD_VERSION_STRING:  18362.1.amd64fre.19h1_release.190318-1202

DUMP_TYPE:  0

BUGCHECK_P1: 2

BUGCHECK_P2: ffffb88659e1d080

BUGCHECK_P3: 0

BUGCHECK_P4: 0

BUGCHECK_STR:  0xCA_2

DEVICE_OBJECT: ffffb88659e1d080

DRIVER_OBJECT: ffffb886591902e0

IMAGE_NAME:  usbip_vhci.sys

DEBUG_FLR_IMAGE_TIMESTAMP:  5e862b7f

MODULE_NAME: usbip_vhci

FAULTING_MODULE: fffff80618ac0000 usbip_vhci

CPU_COUNT: 4

CPU_MHZ: 95b

CPU_VENDOR:  GenuineIntel

CPU_FAMILY: 6

CPU_MODEL: 3a

CPU_STEPPING: 9

CPU_MICROCODE: 6,3a,9,0 (F,M,S,R)  SIG: 20'00000000 (cache) 20'00000000 (init)

DEFAULT_BUCKET_ID:  WIN8_DRIVER_FAULT

PROCESS_NAME:  usbip.exe

CURRENT_IRQL:  1

ANALYSIS_SESSION_HOST:  BART-PC

ANALYSIS_SESSION_TIME:  04-02-2020 20:57:59.0354

ANALYSIS_VERSION: 10.0.18362.1 amd64fre

LAST_CONTROL_TRANSFER:  from fffff8061d4a9942 to fffff8061d3ca210

STACK_TEXT:  
ffffac82`93547f48 fffff806`1d4a9942 : 00000000`00000002 00000000`00000003 ffffac82`935480b0 fffff806`1d25f9f0 : nt!DbgBreakPointWithStatus
ffffac82`93547f50 fffff806`1d4a9032 : 00000000`00000003 ffffac82`935480b0 fffff806`1d3d6a60 00000000`000000ca : nt!KiBugCheckDebugBreak+0x12
ffffac82`93547fb0 fffff806`1d3c2487 : ffffb886`52ead150 00000000`00000024 ffffb886`59e1d080 ffffb886`54faca00 : nt!KeBugCheck2+0x952
ffffac82`935486b0 fffff806`1d422bb3 : 00000000`000000ca 00000000`00000002 ffffb886`59e1d080 00000000`00000000 : nt!KeBugCheckEx+0x107
ffffac82`935486f0 fffff806`18ad3ce6 : ffffb886`59164780 00000000`00000002 00000000`00000000 00000000`00000002 : nt!IoInvalidateDeviceState+0xa3f73
ffffac82`93548740 fffff806`18ad0929 : ffffb886`00000000 ffffb886`54fdbb40 fffff806`18ad6360 fffff806`18aca438 : usbip_vhci!vhci_unplug_dev+0x2a6 [Z:\Development\github\usbip-win\driver\vhci\vhci_pnp.c @ 811]
ffffac82`935487b0 fffff806`1d30a939 : ffffb886`54fdb9f0 ffffb886`59164780 00000000`00000001 ffffb886`591648e0 : usbip_vhci!vhci_ioctl+0x219 [Z:\Development\github\usbip-win\driver\vhci\vhci_ioctl.c @ 269]
ffffac82`93548820 fffff806`1d8b2bd5 : ffffb886`59164780 00000000`00000000 00000000`00000000 ffffb886`5ae8fb70 : nt!IofCallDriver+0x59
ffffac82`93548860 fffff806`1d8b29e0 : 00000000`00000000 ffffac82`93548b80 ffffb886`59164780 ffffac82`93548b80 : nt!IopSynchronousServiceTail+0x1a5
ffffac82`93548900 fffff806`1d8b1db6 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : nt!IopXxxControlFile+0xc10
ffffac82`93548a20 fffff806`1d3d3c18 : 00000000`00000000 00000000`00000000 00000000`00000000 fffff806`1d7f4afd : nt!NtDeviceIoControlFile+0x56
ffffac82`93548a90 00007ffb`c6a7c1a4 : 00007ffb`c3e457d7 0000000b`68b3f1f8 00007ff6`e5667b07 00000000`00000000 : nt!KiSystemServiceCopyEnd+0x28
0000000b`68b3efe8 00007ffb`c3e457d7 : 0000000b`68b3f1f8 00007ff6`e5667b07 00000000`00000000 00007ffb`c6a09c37 : ntdll!NtDeviceIoControlFile+0x14
0000000b`68b3eff0 00007ffb`c4cc63e0 : 00000000`002a4004 00007ff6`e562c1c8 000001b4`af3e5e90 00000000`00000001 : KERNELBASE!DeviceIoControl+0x67
0000000b`68b3f060 00007ff6`e559e333 : 00000000`00000000 cccccccc`cccccccc 0000000b`68b3f0f0 00000000`00000000 : KERNEL32!DeviceIoControlImplementation+0x80
0000000b`68b3f0b0 00000000`00000000 : cccccccc`cccccccc 0000000b`68b3f0f0 00000000`00000000 00000000`00000000 : usbip!usbip_vhci_detach_device+0x83 [Z:\Development\github\usbip-win\userspace\src\usbip\usbip_vhci.c @ 145]

THREAD_SHA1_HASH_MOD_FUNC:  a654659aacf78e2e52098df13f18e63f0cb4bf4f

THREAD_SHA1_HASH_MOD_FUNC_OFFSET:  6cadfdc2d4b8aceaf41879b9da5881c0d6a4eb06

THREAD_SHA1_HASH_MOD:  5fac2d917b1163b8aa1ac411ddff47f1c96aecce

FOLLOWUP_IP: 
usbip_vhci!vhci_unplug_dev+2a6 [Z:\Development\github\usbip-win\driver\vhci\vhci_pnp.c @ 811]
fffff806`18ad3ce6 488b442420      mov     rax,qword ptr [rsp+20h]

FAULT_INSTR_CODE:  24448b48

FAULTING_SOURCE_LINE:  Z:\Development\github\usbip-win\driver\vhci\vhci_pnp.c

FAULTING_SOURCE_FILE:  Z:\Development\github\usbip-win\driver\vhci\vhci_pnp.c

FAULTING_SOURCE_LINE_NUMBER:  811

FAULTING_SOURCE_CODE:  
   807: 				complete_pending_irp(vpdo);
   808: 				SET_NEW_PNP_STATE(vpdo, PNP_DEVICE_REMOVED);
   809: 				IoInvalidateDeviceState(vpdo->common.Self);
   810: 			}
>  811: 		}
   812: 		ExReleaseFastMutex(&vhub->Mutex);
   813: 
   814: 		DBGI(DBG_PNP, "Device %u plug out finished\n", port);
   815: 		return  STATUS_SUCCESS;
   816: 	}

SYMBOL_STACK_INDEX:  5

SYMBOL_NAME:  usbip_vhci!vhci_unplug_dev+2a6

FOLLOWUP_NAME:  MachineOwner

STACK_COMMAND:  .thread ; .cxr ; kb

BUCKET_ID_FUNC_OFFSET:  2a6

FAILURE_BUCKET_ID:  0xCA_2_usbip_vhci!vhci_unplug_dev

BUCKET_ID:  0xCA_2_usbip_vhci!vhci_unplug_dev

PRIMARY_PROBLEM_CLASS:  0xCA_2_usbip_vhci!vhci_unplug_dev

TARGET_TIME:  2020-04-02T18:24:04.000Z

OSBUILD:  18362

OSSERVICEPACK:  0

SERVICEPACK_NUMBER: 0

OS_REVISION: 0

SUITE_MASK:  784

PRODUCT_TYPE:  1

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

OSEDITION:  Windows 10 WinNt TerminalServer SingleUserTS Personal

OS_LOCALE:  

USER_LCID:  0

OSBUILD_TIMESTAMP:  unknown_date

BUILDDATESTAMP_STR:  190318-1202

BUILDLAB_STR:  19h1_release

BUILDOSVER_STR:  10.0.18362.1.amd64fre.19h1_release.190318-1202

ANALYSIS_SESSION_ELAPSED_TIME:  2434

ANALYSIS_SOURCE:  KM

FAILURE_ID_HASH_STRING:  km:0xca_2_usbip_vhci!vhci_unplug_dev

FAILURE_ID_HASH:  {dff59a3e-dbac-01e2-da68-c1bc3321ada4}

Followup:     MachineOwner
---------
```

The vpdo used here is in following state:
```
  | Name | Value | Type
-- | -- | -- | --
◢ | vpdo | 0xffffb886`59e1d1d0 struct usbip_vpdo_dev_t * | struct usbip_vpdo_dev_t *
  | ◢ common | struct dev_common_t | struct dev_common_t
  | ▶ Self | 0xffffb886`59e1d080 struct _DEVICE_OBJECT * | struct _DEVICE_OBJECT *
  | is_vhub | 0x00 '' | unsigned char
  | DevicePnPState | 0n8 (No matching enumerant) | _DEVICE_PNP_STATE
  | PreviousPnPState | NotStarted (0n0) | _DEVICE_PNP_STATE
  | SystemPowerState | PowerSystemWorking (0n1) | _SYSTEM_POWER_STATE
  | DevicePowerState | PowerDeviceD3 (0n4) | _DEVICE_POWER_STATE

```

bartlomiejcieszkowski, 2020-04-02:
also that was detach of 3 divivces , so we are excercising path with `usbip.exe detach -p 0` -  `all`

powershell commands:
```
Write-Host "Press any key to attach"
Read-Host
start powershell -WorkingDirectory $pwd { .\usbip.exe attach -r 192.168.1.15 -b 1-1.2.1.2 }
start powershell -WorkingDirectory $pwd { .\usbip.exe attach -r 192.168.1.15 -b 1-1.2.1.4 }
start powershell -WorkingDirectory $pwd { .\usbip.exe attach -r 192.168.1.15 -b 2-1 }
Write-Host "Press any key to detach"
Read-Host
.\usbip.exe detach -p 0
```

cezanne, 2020-04-08:
@bartlomiejcieszkowski : I like "detaching all" feature. But how about another option instead of "-p 0" for linux compatibility? usbip port number of linux starts from 0.

cezanne, 2021-01-14:
@bartlomiejcieszkowski : I'll close this issue. Please reopen if you need.

Newly committed 0232d02f582381e516b5a995347c4ef5495ec7ed assigns port number starting from 0. usbip.exe with `*` as port will detach all devices.

closed 2021-01-14 by cezanne