cezanne/usbip-win #135: And how about doing it local ?

Inth, 2020-04-06:
Hi

I have been using your version of USBIP project to stream USB device over IP.
It has been working great, thank you for so much work.

If possible could you help me understand one thing ?

since we install a device on the client computer
is there any way to send data localy to this device? 
so instead of reciving keystrokes or mouse clicks from remote host 
it will recive these instructions from lets sayan application which is running on the client machine ?

if yes cold you meaybe point me to the right place where
I can understand this communication better and do some tests ?

best regards
Mat

cezanne, 2020-04-07:
@Inth : Do you mean that your application wants to receive a usbip packet as a  client ? If so, look into `userspace/lib/usbip_forward.c`. You'll easily hook usbip packet at user-level.

Inth, 2020-04-07:
Thank you for your response.
No I want my application to send a USBIP packet to the Virtual device (the one I install when instaling USBIP) within 1 computer (Localy)

Alternative explanation:
Oryginal scenario using IP

Computer1             Computer2
Host(sender) ----->Clinet(Reciver)

My scenario:

Computer2
Application----->(Reciver)

is it possible? and if yes where should I look for the communication

best regards
Mat

cezanne, 2020-04-07:
@Inth : 
> is it possible? and if yes where should I look for the communication

Maybe. Please have a look at [virtual stub project](https://github.com/cezanne/usbip-vstub).

cezanne, 2020-06-06:
@Inth : Close this issue. Please reopen if you want.

closed 2020-06-06 by cezanne