cezanne/usbip-win #137: usbipd cannot list usb devices

KNpTrue, 2020-04-09:
- Information
Version: 0.0.9
Server: Windows 10
Client: Ubuntu18.04 in WSL2
- Step
1. server     `usbip bind --busid=1-200`
2. server     `usbipd -d`
3. client      `sudo usbip list --remote=172.25.0.1`
- Log
1. Server
```
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening :::3240
usbip: info: listening on :::3240
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 172.25.4.211:32990
usbip: info: recv_pdu: received request: 0x8005 - list devices
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 86. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 129. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 63. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 15. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 167. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 101. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 46. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 48. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 208. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 205. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 182. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 124. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 68. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 243. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 113. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 115. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 195. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 116. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 91. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 25. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 163. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 233. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: info: exportable devices: 0
usbip: info: recv_pdu: request 0x8005: complete
```
2. Client
```
usbip: info: no exportable devices found on 172.25.0.1
```
- Note
This question may be similar to question #128

cezanne, 2020-04-25:
@KNpTrue : Could you see a stub device for 1-200 after bind like as the below figure? 

[image]
 
Maybe, many error logs of usbipd.exe are not severe errors. They should be fixed.

cezanne, 2020-04-25:
@KNpTrue :
> Maybe, many error logs of usbipd.exe are not severe errors. They should be fixed.

To remove wrong error logs, a new [usbipd_listr branch](https://github.com/cezanne/usbip-win/tree/usbipd_listr) is created. Please check.

KNpTrue, 2020-04-30:
> @KNpTrue : Could you see a stub device for 1-200 after bind like as the below figure?
> 
> [image]
> 
> Maybe, many error logs of usbipd.exe are not severe errors. They should be fixed.

I can see the name of the USB device is changed to "USB / IP STUB", usbip successfully binds 1-200.
But it does not work.

KNpTrue, 2020-04-30:
> @KNpTrue :
> 
> > Maybe, many error logs of usbipd.exe are not severe errors. They should be fixed.
> 
> To remove wrong error logs, a new [usbipd_listr branch](https://github.com/cezanne/usbip-win/tree/usbipd_listr) is created. Please check.

Verification that these logs have been removed.

cezanne, 2020-04-30:
@KNpTrue : Please retry with v0.1.0 and share the outputs of usbip.exe and usbipd.exe.

Marietto2008, 2020-04-30:
@KNpTrue : which version of kernel are u using ? which version of usbip ?

Marietto2008, 2020-04-30:
@KNpTrue : can u send me the userspace binaries that you are using ?

cezanne, 2020-06-06:
@KNpTrue : Close the issue. Please reopen if you  want.

closed 2020-06-06 by cezanne