cezanne/usbip-win #139: USB adb not working

malekaltanani, 2020-04-14:
USB adb connected but not working
Also the connection  to flash disk  is made and appears but no data

cezanne, 2020-04-21:
@malekaltanani :  Could  you clarify your testing environments?

malekaltanani, 2020-04-25:
windows10 x64

cezanne, 2020-04-26:
@malekaltanani : adb means android debug bridge? Do you want to connect an android smartphone remotely via usbip? What error logs were generated?

xuefy, 2020-05-07:
same problem:
server:Win 10 pro x86_64 19041.208
client:WSL2 Ubuntu 18.04, kernel 4.19.84

C:\Users\Administrator\Downloads\usbip-win>adb devices
List of devices attached
756b0e4e0a7e    device

C:\Users\Administrator\Downloads\usbip-win>usbip.exe list -l
 - busid 1-99 (18d1:4ee2)
   Google Inc. : Nexus 4 (debug) (18d1:4ee2)

C:\Users\Administrator\Downloads\usbip-win>usbip.exe bind -b 1-99
usbip: info: bind_device: bind device on busid 1-99: complete

C:\Users\Administrator\Downloads\usbip-win>usbipd.exe -d
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening :::3240
usbip: info: listening on :::3240
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 172.28.21.26:60696
usbip: info: recv_pdu: received request: 0x8005 - list devices
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: build_udev: invalid devno: 3. devpath returned (null)
usbip: error: walker_edev_list: cannot build usbip dev
usbip: info: exportable devices: 1
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:90:[dump_usb_device] path                 = \\?\usb#vid_18d1&pid_4ee2#756b0e4e0a7e#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:91:[dump_usb_device] busid                = 1-99
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:97:[dump_usb_device] Device(C/SC/P)       = (Defined at Interface level) (00/00/00)
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:99:[dump_usb_device] bcdDevice            = 0
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:104:[dump_usb_device] Vendor/Product       = unknown vendor : unknown product (18d1:4ee2)
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:106:[dump_usb_device] bNumConfigurations   = 0
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:107:[dump_usb_device] bNumInterfaces       = 0
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:109:[dump_usb_device] speed                = Unknown Speed
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:111:[dump_usb_device] busnum               = 1
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_common.c:112:[dump_usb_device] devnum               = 63
usbip: info: recv_pdu: request 0x8005: complete
usbip: info: connection from 172.28.21.26:60698
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_import.c:18:[forwarder_stub] stub forwarding started
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_import.c:108:[recv_request_import] import request busid 1-99: complete
usbip: info: recv_pdu: request 0x8003: complete
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_import.c:28:[forwarder_stub] stub forwarding stopped

root@desktop:/mnt/c/Users/Administrator/Downloads/usbip-win# usbip list -r 172.28.21.26
Exportable USB devices
======================
 - 172.28.21.26
       1-99: Google Inc. : Nexus 4 (debug) (18d1:4ee2)
           : \\?\usb#vid_18d1&pid_4ee2#756b0e4e0a7e#{fb265267-c609-41e6-8eca-a20d92a833e6}
           : (Defined at Interface level) (00/00/00)

root@desktop:/mnt/c/Users/Administrator/Downloads/usbip-win# usbip attach -r 172.28.21.26 -b 1-99
usbip: error: import device

dmesg:
[  487.299544] vhci_hcd: Failed attach request for unsupported USB speed: UNKNOWN

cezanne, 2020-12-28:
@malekaltanani , @xuefy : I have checked adb over usbip-win with [v0.3.1](https://github.com/cezanne/usbip-win/releases/tag/v0.3.1). `adb devices -l` seems to work well for following cases.   
- win10(vhci), linux server(ubuntu 20.04)
- win10(stub), linux client(ubuntu 18.04) 
- win10(vhci), win10(stub)

I did not try WSL.
 
I'll close this issue. Please check and reopen if you have any problem.

closed 2020-12-28 by cezanne