cezanne/usbip-win #140: mismatch version between linux and usbip.exe [usbip (usbip 1.0.0)]

nakhla, 2020-04-16:
hello
Thank you for this great piece  of software, I am happy that there are some attempts to port usbip to windows, I am interested to make a windows as a server and use ordinary usbip client on Ubuntu bionic minimal version,

I downloaded usbip 0.9 release and signed the certificate in both mentioned locations.
I got this error.
and after some investigations I found that this error due to a usbip-utils version mismatch between the client and server. 
After executing `usbip version` in the Client  Ubuntu bionic I got `usbip (usbip-utils 2.0)`
but in `usbip.exe` in windows  server I got`[ usbip (usbip 1.0.0)]`

[image]

cezanne, 2020-04-27:
@nakhla : try to attach 1-96 not 1-59.

cezanne, 2020-06-06:
@nakhla : Close the old issue. Please reopen if you need.

closed 2020-06-06 by cezanne