cezanne/usbip-win #141: How to Use Usbip Stubs

dwernli, 2020-04-17:
What is the usbip_stub project for? Is it intended to simulate a device? Or the usbip_vhci driver? Or both? Is there documentation on how to use it?

Also, regarding files annotated with "stub" that aren't in the usbip_stub project, what are they for and how are they used? Specifically:
-- include\usbip_stub_api.h
-- userspace\lib\usbip_stub.c & .h (in the usbip_common project)
--userspace\src\usbipd\usbipd_stub.c & .h (in the usbipd project)

Do these stubs allow development and/or integration testing without the uncertainty introduced by using actual devices? That would be an awesome convenience and time saver if that is the case. Thank you very much.

cezanne, 2020-04-21:
> What is the usbip_stub project for? Is it intended to simulate a device? Or the usbip_vhci driver? Or both? Is there documentation on how to use it?

The usbip_stub VC project is for a usbip server to simulate a USB device. [README.md](https://github.com/cezanne/usbip-win/blob/master/README.md) is the only document for a stub driver.  😢

dwernli, 2020-04-21:
Thank you for your answer, that helps. I really appreciate it.

closed 2020-06-06 by cezanne