cezanne/usbip-win #142: USBIP Client on Windows 10 Cannot Read Device Information From Server

muhammadhaqyaunoora, 2020-04-20:
So, I am currently trying to connect to my Arduino Uno remotely using USBIP via Raspberry Pi 3 B. The problem that I currently have is that all of the information about the Arduino (which can be seen on the Raspberry) cannot be seen on the Windows 10 client. Besides that, whenever I tried to 
bind to the Arduino my Windows 10 client cannot find the correct driver and only shows USB Device over IP (I will attach some screenshots below). I am pretty sure I have followed the steps correctly and already tried solutions from older issues but nothing seems to worked out yet.
[image]
[image]
[image]

cezanne, 2020-04-27:
@muhammadhaqyaunoora : Which version did you test ? `root\usbipenum` in your screenshot is a very old vhci device name. A recent one is `root\usbip_vhci`. Check your `usbip_vhci.inf`.

muhammadhaqyaunoora, 2020-04-30:
@cezanne I think I forgot to delete the old version of the vhci. I just tried redoing everything again with the most recent one from this repository and it still does the same as before.
Here is the screenshot from the usbip_vhci.inf.
[image]
[image]
And here is what I got after redoing everything
[image]
[image]

cezanne, 2020-04-30:
@muhammadhaqyaunoora : What is the device which you want to attach? Code 28 may indicate that there's no function driver. Did you install a driver for a device to be attached?

muhammadhaqyaunoora, 2020-04-30:
I tried to attach to Arduino Uno and the driver is already installed

cezanne, 2020-05-01:
@muhammadhaqyaunoora : 
> I am trying to attach to Arduino Uno and the driver is already installed

I see. Can you share kernel log or USB capture. Please refer https://github.com/cezanne/usbip-win/issues/144#issuecomment-619335255 .

muhammadhaqyaunoora, 2020-05-01:
This is the USB Capture
https://www.upload.ee/files/11602733/arduinocapt.pcap.html

cezanne, 2020-05-01:
@muhammadhaqyaunoora : Thanks for the capture. Is it obtained while the arduino is directly attached without usbip-win? It seems too short. See my wireshark screenshot for your capture.

[image]

muhammadhaqyaunoora, 2020-05-01:
yes, it was attached directly to my laptop, do I need to attach it via the usbip-win?

cezanne, 2020-05-01:
@muhammadhaqyaunoora : 
> do I need to attach it via the usbip-win?

No. A log with no usbip-win is required.
Then your usb capture is too simple. What is your device number among 1.5.0, 1.4.0, 1.3.0 in screenshot ?

muhammadhaqyaunoora, 2020-05-01:
I think it was the 1.4.0, or is it better if I did another capture?

cezanne, 2020-05-01:
@muhammadhaqyaunoora : Yes please..

muhammadhaqyaunoora, 2020-05-01:
https://www.upload.ee/files/11603318/arduinocapt2.pcap.html
In this one I capture it from when I plugged the cable in until I plugged it off again. Sorry for the bother.

cezanne, 2020-05-01:
@muhammadhaqyaunoora : This capture seems good..

muhammadhaqyaunoora, 2020-05-01:
Are there anything else that I could try to help resolve this problem?

cezanne, 2020-05-01:
@muhammadhaqyaunoora : Where's arduino driver SW? Just install arduino IDE?

cezanne, 2020-05-01:
@muhammadhaqyaunoora : I have checked that arduino uno works correctly. Please retry with v0.1.0, an up-to-date release.

muhammadhaqyaunoora, 2020-05-02:
> @muhammadhaqyaunoora : I have checked that arduino uno works correctly. Please retry with v0.1.0, an up-to-date release.

Can you give a screenshot of how it look like on your device? Because it is still showing code 28 on mine, and when I checked the usbip list it is still not showing the device information after I retried using the v0.1.0.
[image]
[image]

cezanne, 2020-05-02:
@muhammadhaqyaunoora : 
A device is successfully attached with vstub.
[image]

This is a [vhci kernel log](https://github.com/cezanne/usbip-win/files/4567251/arduino.log).
Well, I have checked that all URB's were invoked as your USB captured log. But `STATUS_NOT_SUPPORTED` in last part of my log should be more investigated.

Anyway, one suspicious thing in your screenshot, why did you stop attach command with ctrl+c ? Usually, `usbip detach -p 1` would be the best way for detaching the device.

muhammadhaqyaunoora, 2020-05-02:
Do you think another log while using usbip could help investigate the problem? Because I haven't done that yet. And for that ctrl+c situation, that is just me panicking because my laptop started to slow down and I'm afraid if I open another window it would hang. I guess I opened too many applications when I did the trial, nothing to worry about that.

cezanne, 2020-05-02:
@muhammadhaqyaunoora : 
> Do you think another capture while using usbip could help investigate the problem? 

No.. Instead, can you share your kernel log? Kernel log generation method is described in README.md.

muhammadhaqyaunoora, 2020-05-03:
Here it is. I did attach and detach while logging. I hope this is good enough.
[haqysarduino.LOG](https://github.com/cezanne/usbip-win/files/4569625/haqysarduino.LOG)

cezanne, 2020-05-03:
@muhammadhaqyaunoora : Please check again that you *did install* an arduino driver on an attaching machine. I had the same log with yours when there was no arduino driver installed.

muhammadhaqyaunoora, 2020-05-04:
Okay, it seems to clears up the problem a little bit but I got a new problem with it which is a code 10. I have installed the latest driver by the way.
[image]

muhammadhaqyaunoora, 2020-05-04:
Okay, I installed the latest driver and my laptop now recognizes the Arduino but it throws code 10 and I have tried many solutions from the internet but none works yet.
[image]

cezanne, 2020-05-04:
@muhammadhaqyaunoora : How about kernel logs in that case?

muhammadhaqyaunoora, 2020-05-04:
Here it is, I saw slight differences with your log but I cannot really tell what seems to be the problem.
[haqysarduino2.LOG](https://github.com/cezanne/usbip-win/files/4572988/haqysarduino2.LOG)

cezanne, 2020-05-04:
@muhammadhaqyaunoora : In your log, there's a warning to cause your device to work incorrectly.
```
00000206	0.03614570	usbip_vhci:(WW) URB_FUNCTION_SELECT_INTERFACE: wrong status: USBD_STATUS_STALL_PID
```
I have no idea about why stall occurred. Please check your usbip_vhci.sys is up-to-date.

muhammadhaqyaunoora, 2020-05-05:
I redownloaded that file from the v0.1.0 release, but it was still showing code 10 and that stall warning.

cezanne, 2020-05-05:
@muhammadhaqyaunoora : Ok, I'll try to  get a physical arduino and test.

muhammadhaqyaunoora, 2020-05-06:
Do you think it is something to do with my laptop? Or are there other things that might cause this problem?

cezanne, 2020-05-06:
@muhammadhaqyaunoora : 
> Do you think it is something to do with my laptop? Or are there other things that might cause this problem?

I'm not sure but your laptop has no problem if arduino works well without usbip-win. I'm still trying to find a physical arduino. 😞

muhammadhaqyaunoora, 2020-05-07:
Ooh, okay. I will try looking for a way to troubleshoot while waiting updates from you. Thanks for your help, I really appreciate it.

cezanne, 2021-01-14:
@muhammadhaqyaunoora : How about trying with the latest usbip-win version ?

cezanne, 2021-01-16:
@muhammadhaqyaunoora : I'll close it. Please reopen if you need.

closed 2021-01-16 by cezanne