cezanne/usbip-win #143: USB SPDIF card doesn't work properly

hathlife, 2020-04-20:
Local:latest pre-release binary on Win10 1909
Remote:OpenWrt 19.07.2 mips (4.14.171) with server version 2.0-11
Connect:OK
Windows detecting & configuring the device : reporting OK
Problem:
1. Lag when adjusting volume
2. Can't play anything

hathlife, 2020-04-20:
The commandline spam:
```
usbip: error: unknown command in pdu header: 0
```

cezanne, 2020-06-06:
@hathlife : Can you share windows kernel log or USB capture? Please refer to https://github.com/cezanne/usbip-win/issues/144#issuecomment-619335255

hathlife, 2020-06-07:
> windows kernel log

[image]
Should I use this to capture the kernal log?
I've reset machine & router once before, So I need to re-install the driver on both of them again.

cezanne, 2020-06-08:
>  Should I use this to capture the kernal log?

Your DebugView has no vhci log. If a debug print is enabled correctly, DebugView shows logs similar to following logs.
```
00000001        0.00000000      usbip_vhci: DriverEntry: Enter
00000002        0.00000228      usbip_vhci: RegistryPath FFFFD304EE01C000
00000003        0.00000313      usbip_vhci: DriverEntry: Leave
00000004        0.00006997      usbip_vhci: adding root: pdo: 0xFFFFD304F1F38E40
00000005        0.00007936      usbip_vhci: creating child cpdo
00000006        0.00015360      usbip_vhci: root added: vdev: FFFFD304ED81BC60
00000007        0.00019257      usbip_vhci: vhci_pnp(root): Enter: minor:IRP_MN_QUERY_LEGACY_BUS_INFORMATION, irp: FFFFD304EFBC2CF0
00000008        0.00019513      usbip_vhci: vhci_pnp(root): Leave: irp:FFFFD304EFBC2CF0, status:STATUS_NOT_SUPPORTED
00000009        0.00022727      usbip_vhci: vhci_pnp(root): Enter: minor:IRP_MN_FILTER_RESOURCE_REQUIREMENTS, irp: FFFFD304EFBC2CF0
00000010        0.00022869      usbip_vhci: vhci_pnp(root): Leave: irp:FFFFD304EFBC2CF0, status:STATUS_NOT_SUPPORTED
00000011        0.00025315      usbip_vhci: vhci_pnp(root): Enter: minor:IRP_MN_START_DEVICE, irp: FFFFD304EFBC2CF0
00000012        0.00025457      usbip_vhci: device(root) started
```

All log entries related to vhci driver have "usbip_vhci:" Did you set a IHVDRIVER registry entry as in README.md ?

cezanne, 2020-12-28:
@hathlife: Too old issue. I'll close. Please reopen if you need.

closed 2020-12-28 by cezanne