cezanne/usbip-win #144: Silicon Labs CP210x - Error

pasha-zzz, 2020-04-23:
Server: ARM Linux 4.4.219 (DD-WRT)
Client: Windows 10 x64 1909

After attach I can see device (child of USB/IP VHCI) but with exclamation mark. Info is "Cannot start device. Code 10."

When attach device to Windows machine directly - it works fine.
USB\VID_10C4&PID_EA60&REV_0100

cezanne, 2020-04-25:
@pasha-zzz : Windows kernel log is required to resolve your issue. Log generation methods are described in README.md. Or USB packet logs of directly attached your device would be helpful. One of such a tool is [USBPcap](https://desowin.org/usbpcap/).

pasha-zzz, 2020-04-25:
USBPCap file:
https://www.upload.ee/files/11559558/CP210x.pcap.html
https://sendit.cloud/4g2pwg22r9fm

cezanne, 2020-04-25:
@pasha-zzz : Thanks for the USBPCap file. I'm trying to make a virtual CP210x device via [usbip-vstub](https://github.com/cezanne/usbip-vstub). Maybe the device needs a function driver. Can you share a CP210x driver ?

pasha-zzz, 2020-04-25:
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

cezanne, 2020-04-27:
A vstub for cp210x is implemented and here is client-side(vhci) [kernel log].(https://github.com/cezanne/usbip-win/files/4538360/DESKTOP-3V3FHQU.LOG)

I had tested it with [control_transfer branch](https://github.com/cezanne/usbip-win/tree/control_transfer). As shown in the log file, the device was removed after a fifth URB_FUNCTION_CONTROL_TRANSFER(seq num: 16) was executed.

@pasha-zzz : Could you share a kernel log with a real cp210x device?

cezanne, 2020-04-27:
@pasha-zzz : Here is a builtin binary [vhci driver](https://github.com/cezanne/usbip-win/files/4538727/usbip_vhci.zip).

pasha-zzz, 2020-04-29:
> Could you share a kernel log with a real cp210x device?

Instructions for kernel logging are not full.

- Add registry key and reboot
- DebugView is a good tool to view
- Thats all

But I not need view. I want collect logs. And what I must do with device? Eject, plug, or maybe real data exchange through virtula COM is needed?

cezanne, 2020-04-29:
@pasha-zzz : Did you install DebugView successfully ? Some options of debugview should be marked like as a below figure.
[image]

> But I not need view. I want collect logs. And what I must do with device? Eject, plug, or maybe real data exchange through virtula COM is needed?

I need kernel logs of usbip-win with a real cp210x. For now, when getting logs, ejection or any other action is not required.

pasha-zzz, 2020-04-29:
Kernel logs of attach command. Scenario like in 1st message:
https://www.upload.ee/files/11594031/usbip-win.zip.html
https://sendit.cloud/oubcdsdm77xq

cezanne, 2020-04-30:
@pasha-zzz : You don't need to attach a file using an external service. Github supports file attachmenent in comment dialog. 
Anyway,  a kernel log of your https://sendit.cloud/oubcdsdm77xq does not seem to be obtained from an up-to-date vhci driver. Because a following log cannot be generated.
```
00000475	1.77483726	usbip_vhci:(WW) process_irp_urb_req: unhandled function: URB_FUNCTION_CONTROL_TRANSFER: len: 136
```

[This](https://github.com/cezanne/usbip-win/files/4556991/cp210.LOG)  is my kernel log via usbip-vstub. The recent vhci driver seems to be able to handle a cp210x properly.

pasha-zzz, 2020-04-30:
Sorry, I didnt think that only driver is sufficient for correction! Reistalled driver and all work fine. Thanks a lot!

cezanne, 2020-04-30:
@pasha-zzz : Good to hear.. Please close this issue later if your issue will be resolved.

closed 2020-04-30 by pasha-zzz