cezanne/usbip-win #147: some help to understand how it works.

Marietto2008, 2020-04-27:
Hello.

Actually I'm enjoying myself by playing with WSL. I would like to use my XBOX / KINECT inside the WSL. To do that,I need to enable the USB ports inside it,because as u probably know,they are disabled. I think that this project :

https://github.com/cezanne/usbip-win

can help me. To make it works I need to identify what's the device ID of my XBOX,because the usbip tool does not recognize it,as it seems :

usbip.exe list -l - busid 1-129 (05e3:0608) Genesys Logic, Inc. : Hub (05e3:0608)
busid 1-167 (05e3:0608) Genesys Logic, Inc. : Hub (05e3:0608)
busid 1-149 (0480:a007) Toshiba America Inc : External Disk USB 3.0 (0480:a007)
busid 1-175 (2109:0813) VIA Labs, Inc. : unknown product (2109:0813)
busid 1-89 (2109:2813) VIA Labs, Inc. : unknown product (2109:2813)
busid 1-220 (25a7:fa23) unknown vendor : unknown product (25a7:fa23)
busid 1-177 (048d:8297) Integrated Technology Express, Inc. : unknown product (048d:8297)
busid 1-122 (1058:0704) Western Digital Technologies, Inc. : My Passport Essential (WDME) (1058:0704)
busid 1-43 (2109:0813) VIA Labs, Inc. : unknown product (2109:0813)
busid 1-144 (05ac:0250) Apple, Inc. : Aluminium Keyboard (ISO) (05ac:0250)
busid 1-184 (1058:25a3) Western Digital Technologies, Inc. : unknown product (1058:25a3)
busid 1-218 (2109:2813) VIA Labs, Inc. : unknown product (2109:2813)
busid 1-209 (045e:02c4) Microsoft Corp. : unknown product (045e:02c4)
busid 1-181 (0480:a207) Toshiba America Inc : unknown product (0480:a207)
busid 1-29 (093a:2510) Pixart Imaging, Inc. : Optical Mouse (093a:2510)
busid 1-134 (0bc2:61b5) Seagate RSS LLC : unknown product (0bc2:61b5)
busid 1-158 (05ac:1006) Apple, Inc. : Hub in Aluminum Keyboard (05ac:1006)

I'm not able to understand to which kind of interval it belongs. I went on a real linux installation and this is what I saw :

[ 2.392735] usb 2-8: new SuperSpeed Gen 1 USB device number 4 using xhci_hcd
[ 2.413596] usb 2-8: New USB device found, idVendor=045e, idProduct=02c4, bcdDevice= 1.00
[ 2.413596] usb 2-8: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 2.413597] usb 2-8: Product: Xbox NUI Sensor
[ 2.413597] usb 2-8: Manufacturer: Microsoft

This is the example explained in the tutorial :

usbip.exe list -l
busid 1-59 (045e:00cb) Microsoft Corp. : Basic Optical Mouse v2.0 (045e:00cb)
busid 1-30 (80ee:0021) VirtualBox : USB Tablet (80ee:0021)

As u can see,the memory interval where is the Optical mouse 2.0 starts with "045e" and also my xbox NUI sensor starts with "045e",so the right interval could be 1-59 ? I did :

λ usbip.exe bind -b 1-59
but this error came out :
usbip: error: bind_device: error binding device on busid 1-59: err: 0

instead this command worked :

λ usbipd.exe -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240

The true is that I don't know what I'm doing :P Can someone help me ? 10x

cezanne, 2020-04-27:
@Marietto2008 : You can browse your USB device with vendor and product ID such as 045e:02c4. usbip-win has relatively old [usb id repository](https://github.com/cezanne/usbip-win/blob/master/userspace/usb.ids) and an up-to-date version may be [here?](http://www.linux-usb.org/usb.ids)

However, there seems to be no matching device entry of 045e:02c4 even in up-to-date USB id repository. 

> so the right interval could be 1-59 ?

1-59 means `<bus id>-<device id>`,  not interval. Currently, usbip-win server(stub) always uses bus id as 1.

BTW, what's the result of binding 1-209?

Marietto2008, 2020-04-27:
λ usbip.exe bind -b 1-209
usbip: error: cannot load certificate: subject: USBIP Test
usbip: error: bind_device: error binding device on busid 1-209: err: 0

Marietto2008, 2020-04-27:
you can watch here what I did :

https://drive.google.com/open?id=1fZjObcLl9KiKkW4PUMnYABaJYJhryKpS

cezanne, 2020-04-27:
@Marietto2008 : In Readme.md
```
Certificate should be installed into
"Trusted Root Certification Authority" in "Local Computer" (not current user) and
"Trusted Publishers" in "Local Computer" (not current user)
```
You may select the location of the certificate, not automatically.

[image]

Marietto2008, 2020-04-27:
https://drive.google.com/open?id=18hAGRj2OwR1cqlyP57_V0aHw098octLg

cezanne, 2020-04-27:
@Marietto2008 : Did you install the certificate into "trusted publishers" section too ? If yes, it will be a difficult issue. :cry: In that case, please try another device..

Marietto2008, 2020-04-27:
sorry I'd forgot. done. now this is the situation :

C:\Users\marietto2020\Desktop\WSL\Kinect\usbip-win (master -> origin)
λ usbip.exe bind -b 1-209
usbip: info: bind_device: bind device on busid 1-209: complete

cezanne, 2020-04-27:
@Marietto2008 : Maybe attach in WSL if WSL supports usbip ?

Marietto2008, 2020-04-27:
root@DESKTOP-N9UN2H3:/home# usbip attach -r 192.168.1.6 -b 1-209
usbip: invalid option -- 'r'
usbip: invalid option -- 'b'
Usage: usbip [options]
        -a, --attach [host] [bus_id]
                Attach a remote USB device.

        -x, --attachall [host]
                Attach all remote USB devices on the specific host.

        -d, --detach [ports]
                Detach an imported USB device.

        -l, --list [hosts]
                List exported USB devices.

        -p, --port
                List virtual USB port status.

        -D, --debug
                Print debugging information.

        -v, --version
                Show version.

        -h, --help

Marietto2008, 2020-04-27:
I tried also different combinations,like : 

usbip attach -a 192.168.1.6 -p 1-209
or
usbip attach -x 192.168.1.6 -p 1-209

don't work

cezanne, 2020-04-27:
@Marietto2008 : Command help says you should  invoke as follows:
`usbip -a 192.168.1.6 1-209`

Marietto2008, 2020-04-27:
root@DESKTOP-N9UN2H3:/home# usbip -a 192.168.1.6 1-209
usbip err: usbip_network.c: 119 (usbip_recv_op_common) recv op_common, -1
usbip err: vhci_attach.c: 324 (query_import_device) recv op_common
usbip err: vhci_attach.c: 362 (attach_device) query

Marietto2008, 2020-04-27:
λ usbipd -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 192.168.1.6:3299
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_network.c:156:[usbip_net_recv_op_common] version mismatch: 262 273
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_accept.c:18:[recv_pdu] recv_pdu: could not receive opcode: 0

cezanne, 2020-04-27:
> usbip: debug: C:\work\usbip-win\userspace\lib\usbip_network.c:156:[usbip_net_recv_op_common] version mismatch: 262 273

The version(262, 0x106) of usbip in WSL is not up-to-date (273, 0x111) .

Marietto2008, 2020-04-27:
how can I upgrade it ?

cezanne, 2020-04-28:
@Marietto2008 : I'm not assure but upgrade would be possible only if WSL support a newer usbip.

Marietto2008, 2020-04-28:
root@DESKTOP-N9UN2H3:/etc/apt# usbip

Command 'usbip' not found, but can be installed with:

apt install linux-oem-5.6-tools-common  # version 5.6.0-1007.7, or
apt install linux-tools-common          # version 5.4.0-26.30

Marietto2008, 2020-04-28:
UPDATE :

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip# ./configure

checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
[... 71 lines cut ...]
config.status: executing depfiles commands
config.status: executing libtool commands

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip# make install
Making install in libsrc
make[1]: Entering directory '/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip/libsrc'
  CC       libusbip_la-names.lo
  CC       libusbip_la-usbip_host_driver.lo
  CC       libusbip_la-usbip_device_driver.lo
In file included from /usr/include/string.h:495,
                 from usbip_device_driver.c:25:
In function ‘strncpy’,
    inlined from ‘read_usb_vudc_device’ at usbip_device_driver.c:108:2:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:10: error: ‘__builtin_strncpy’ specified bound 256 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘strncpy’,
    inlined from ‘read_usb_vudc_device’ at usbip_device_driver.c:127:2:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:10: error: ‘__builtin_strncpy’ specified bound 32 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:472: libusbip_la-usbip_device_driver.lo] Error 1
make[1]: Leaving directory '/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip/libsrc'
make: *** [Makefile:498: install-recursive] Error 1

Marietto2008, 2020-04-28:
maybe this is the patch :

https://patchwork.kernel.org/patch/10538575/

Marietto2008, 2020-04-28:
patched,update 2 :

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip# make install

Making install in libsrc
make[1]: Entering directory '/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip/libsrc'
  CC       libusbip_la-usbip_device_driver.lo
  CC       libusbip_la-usbip_common.lo
  CC       libusbip_la-usbip_host_common.lo
  CC       libusbip_la-vhci_driver.lo
  CC       libusbip_la-sysfs_utils.lo
  CCLD     libusbip.la
make[2]: Entering directory '/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip/libsrc'
 /usr/bin/mkdir -p '/usr/local/lib'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libusbip.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libusbip.so.0.0.1 /usr/local/lib/libusbip.so.0.0.1
libtool: install: (cd /usr/local/lib && { ln -s -f libusbip.so.0.0.1 libusbip.so.0 || { rm -f libusbip.so.0 && ln -s libusbip.so.0.0.1 libusbip.so.0; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libusbip.so.0.0.1 libusbip.so || { rm -f libusbip.so && ln -s libusbip.so.0.0.1 libusbip.so; }; })
libtool: install: /usr/bin/install -c .libs/libusbip.lai /usr/local/lib/libusbip.la
libtool: install: /usr/bin/install -c .libs/libusbip.a /usr/local/lib/libusbip.a
libtool: install: chmod 644 /usr/local/lib/libusbip.a
libtool: install: ranlib /usr/local/lib/libusbip.a
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip/libsrc'
make[1]: Leaving directory '/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip/libsrc'
Making install in src
make[1]: Entering directory '/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip/src'
  CC       usbip.o
  CC       utils.o
  CC       usbip_network.o
usbip_network.c: In function ‘usbip_net_pack_usb_device’:
usbip_network.c:91:32: error: taking address of packed member of ‘struct usbip_usb_device’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
   91 |  usbip_net_pack_uint32_t(pack, &udev->busnum);
      |                                ^~~~~~~~~~~~~
usbip_network.c:92:32: error: taking address of packed member of ‘struct usbip_usb_device’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
   92 |  usbip_net_pack_uint32_t(pack, &udev->devnum);
      |                                ^~~~~~~~~~~~~
usbip_network.c:93:32: error: taking address of packed member of ‘struct usbip_usb_device’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
   93 |  usbip_net_pack_uint32_t(pack, &udev->speed);
      |                                ^~~~~~~~~~~~
usbip_network.c:95:32: error: taking address of packed member of ‘struct usbip_usb_device’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
   95 |  usbip_net_pack_uint16_t(pack, &udev->idVendor);
      |                                ^~~~~~~~~~~~~~~
usbip_network.c:96:32: error: taking address of packed member of ‘struct usbip_usb_device’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
   96 |  usbip_net_pack_uint16_t(pack, &udev->idProduct);
      |                                ^~~~~~~~~~~~~~~~
usbip_network.c:97:32: error: taking address of packed member of ‘struct usbip_usb_device’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
   97 |  usbip_net_pack_uint16_t(pack, &udev->bcdDevice);
      |                                ^~~~~~~~~~~~~~~~
In file included from usbip_network.c:33:
usbip_network.c: In function ‘usbip_net_send_op_common’:
[... 14 lines cut ...]
      |                                ^~~~~~~~~~~~~~~~~~~~
usbip_network.c:155:2: note: in expansion of macro ‘PACK_OP_COMMON’
  155 |  PACK_OP_COMMON(1, &op_common);
      |  ^~~~~~~~~~~~~~
usbip_network.c: In function ‘usbip_net_recv_op_common’:
usbip_network.h:36:32: error: taking address of packed member of ‘struct op_common’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
   36 |  usbip_net_pack_uint16_t(pack, &(op_common)->version);\
      |                                ^~~~~~~~~~~~~~~~~~~~~
usbip_network.c:179:2: note: in expansion of macro ‘PACK_OP_COMMON’
  179 |  PACK_OP_COMMON(0, &op_common);
      |  ^~~~~~~~~~~~~~
[... cut ...]
usbip_network.h:38:32: error: taking address of packed member of ‘struct op_common’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
   38 |  usbip_net_pack_uint32_t(pack, &(op_common)->status);\
      |                                ^~~~~~~~~~~~~~~~~~~~
usbip_network.c:179:2: note: in expansion of macro ‘PACK_OP_COMMON’
  179 |  PACK_OP_COMMON(0, &op_common);
      |  ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:431: usbip_network.o] Error 1

Marietto2008, 2020-04-28:
Update 3 :

https://sources.debian.org/src/linux/5.4.19-1/debian/patches/bugfix/all/usbip-network-fix-unaligned-member-access.patch/

Marietto2008, 2020-04-28:
so. now from windows 10,I do :

C:\Users\marietto2020\Desktop\WSL\Kinect\usbip-win (master -> origin)
λ usbip.exe bind -b 1-209
usbip: info: bind_device: bind device on busid 1-209: complete

C:\Users\marietto2020\Desktop\WSL\Kinect\usbip-win (master -> origin)
λ usbipd -d -4
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240

but on linux WSL,what should I do ? the syntax of the command is changed :

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip# usbip version
usbip (usbip-utils 2.0)
root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/WSL2-Linux-Kernel/tools/usb/usbip# usbip -a 192.168.1.6 1-209
usbip: invalid option
usage: usbip [--debug] [--log] [--tcp-port PORT] [version]
             [help] <command> <args>

cezanne, 2020-04-28:
@Marietto2008 :
> ```usbip: debug: C:\work\usbip-win\userspace\lib\usbip_network.c:156:[usbip_net_recv_op_common] version mismatch: 4353 273```

usbipd denies if protocol versions are mismatched. 

https://github.com/cezanne/usbip-win/blob/b8d86d07a920519f5cf0fcf1bd7fe40be6bd3988/userspace/lib/usbip_network.c#L155-L158

Network byte order problem? 4353 (0x1101) vs 273 (0x0111) 
Later one is correct.

Marietto2008, 2020-04-28:
I have fixed all the bugs that I found. Now it works,BUT it does not work because the Windows version and the linux versions use different protocol. So,what to do next ? Or I use another version in Linux or in Windows,right ? At this point maybe it's easier to use the correct version in Windows. what do u think ?

cezanne, 2020-04-29:
@Marietto2008 : How about checking #149 ? @thedozo has a similar environment to yours.

Marietto2008, 2020-04-29:
在到達我所在的位置之前，他還有很長的路要走

Marietto2008, 2020-04-29:
@cezanne : I need that u compile the same exact version that we run on WSL2,I suppose this version  

root@DESKTOP-N9UN2H3:/mnt/i/macos-haxm# usbip version
usbip (usbip-utils 2.0)

this goes beyond my abilities and beyond the abilities of a lot of us.

thedozo, 2020-04-29:
@Marietto2008  this is my current scenario: https://github.com/cezanne/usbip-win/issues/149#issuecomment-621042309

Marietto2008, 2020-04-29:
@thedozo : you should give the following commands one after another :

1) on Windows : 

a) usbip.exe bind -b 1-209

if you have installed the certificates correctly,u should see this message :
usbip: info: bind_device: bind device on busid 1-209: complete

b) usbipd -d -4
you should see :

usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 192.168.1.6:2174

2) on WSL2 / Ubuntu 20 :

root@DESKTOP-N9UN2H3:/mnt/i/macos-haxm# usbip attach -r 192.168.1.6 -b 1-209
usbip: error: Attach Request for 1-209 failed - Request Completed Successfully

as soon as you gave that command,on the windows terminal you will see :

usbip: debug: C:\work\usbip-win\userspace\lib\usbip_network.c:156:[usbip_net_recv_op_common] version mismatch: 4353 273
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_accept.c:18:[recv_pdu] recv_pdu: could not receive opcode: 0

if you have a different result than this,you get an error somewhere.

thedozo, 2020-04-29:
@Marietto2008 
I gave all commands correctly and certs are installed ok as well. But when I give:

`usbip.exe bind -b 1-209`

It's throwing me the error:
**usbip: error: failed to update driver: 103
usbip: error: bind_device: error binding device on busid 1-209: err: 0**

which I already mentioned above. That's my issue...

Marietto2008, 2020-04-29:
u can give a look here :

https://github.com/cezanne/usbip-win/issues/108

thedozo, 2020-04-29:
> u can give a look here :
> 
> #108

@Marietto2008  this doesn't solve my issue. However I find that there is no USB/IP STUB device in my PC under System devices!! Why is that? I have usbip_stub.sys under C:/windows/system32/drivers

Marietto2008, 2020-04-29:
make me a screenshot,I want to see where should be and if I have it.

Marietto2008, 2020-04-29:
> @Marietto2008 <https://github.com/Marietto2008> screenshot of what?
> System devices? Or the system32/drivers folder?

system device where should be the usb stub of the usb ip stack

thedozo, 2020-04-29:
> system device where should be the usb stub of the usb ip stack Il mer 29 apr 2020, 21:31 Abhi <notifications@github.com> ha scritto:
> […](#)

Ok, it’s there now suddenly but with yellow error mark 

[image]

Marietto2008, 2020-04-29:
you have cut the picture,so I don't understand where the stub should be. Anyway,it seems that I haven't it.

[image]

thedozo, 2020-04-29:
> you have cut the picture,so I don't understand where the stub should be. Anyway,it seems that I haven't it.
> 
> [image]

@Marietto2008 it supposed to be under system drivers as per #108, so my location seems ok.

Anyway is it happening due to Ubuntu 20.04?? I can bind not a single USB device with usbip. Error every time.

Marietto2008, 2020-04-29:
I don't think. I'm using ubuntu 20. Do u want to start a teamviewer session ? I'm sure that you are not installing the certificates correctly. I would try.

Marietto2008, 2020-04-29:
did u give this command ? 

bcdedit /set testsigning on

Marietto2008, 2020-04-29:
@cezanne : is this the missing component that I need ?

https://github.com/barbalion/usbip-win-client

cezanne, 2020-04-29:
@Marietto2008 :
> @cezanne : We need that u compile the same version of usbip for Windows that we run on WSL2,I suppose this version :
>
>root@DESKTOP-N9UN2H3:/mnt/i/macos-haxm# usbip version
usbip (usbip-utils 2.0)

The version of usbip.exe does not matter to your  problem. It just a simple string which does not relate to kernel. The most important version is a USBIP_VERSION macro in source. I have confirmed that an up-to-date linux kernel(such as ubuntu 20.04) still uses 0x111 as USBIP_VERSION, which is same with the version of usbip-win.

cezanne, 2020-04-29:
@Marietto2008 :
> @cezanne : is this the missing component that I need ?
>
> https://github.com/barbalion/usbip-win-client

Maybe no.. It's a helper utility that support automatic start-up via windows service.

Marietto2008, 2020-04-29:
so ? what should I do to make it work ?

thedozo, 2020-04-30:
> did u give this command ?
> 
> bcdedit /set testsigning on

Well, i gave the command and after rebooting my PC the yellow error icon is gone. But this doesn't solve my problem. Still I can't bind 1-209 which is my iPhone. Giving error 103 as stated above.

thedozo, 2020-04-30:
> @Marietto2008 :
>
>
> The version of usbip.exe does not matter to your problem. It just a simple string which does not relate to kernel. The most important version is a USBIP_VERSION macro in source. I [...]

@Marietto2008 I tried to recompile another kernel, and now I'm using kernel **4.19.104** (cloned and recompiled from WSL2-Kernel github). Now here's my debug information:

[image]

**You can see that my iphone is showing in the list but it's not exportable. But now I can attach USB pendrive with WSL. so anything to do with iPhone driver?**

Marietto2008, 2020-04-30:
@thedozo : can u confirm that in your situation you haven't the problem that I have ? I mean,the protocol mismatch.

Marietto2008, 2020-05-01:
it has been hard,a full day of work to make work again the kinect,but I've been able to do that. So,I want to continue to work on the usb-ip configuration. this is what I did :

ziomarietto@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard$ dir

COPYING Kbuild MAINTAINERS Module.symvers System.map built-in.a drivers include kernel modules.builtin samples sound virt
CREDITS Kconfig Makefile README arch certs firmware init lib modules.order scripts tools vmlinux
Documentation LICENSES Microsoft README-Microsoft.WSL2 block crypto fs ipc mm net security usr vmlinux.o

ziomarietto@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard$ sudo make modules_install -j 12

[sudo] password for ziomarietto:

INSTALL drivers/hid/hid-generic.ko
INSTALL drivers/hid/hid.ko
INSTALL drivers/hid/usbhid/usbhid.ko
INSTALL drivers/net/mii.ko
INSTALL drivers/net/phy/libphy.ko
INSTALL drivers/net/usb/asix.ko
INSTALL drivers/net/usb/ax88179_178a.ko
INSTALL drivers/net/usb/cdc_ether.ko
INSTALL drivers/net/usb/cdc_ncm.ko
INSTALL drivers/net/usb/cdc_subset.ko
INSTALL drivers/net/usb/net1080.ko
INSTALL drivers/net/usb/rndis_host.ko
INSTALL drivers/net/usb/usbnet.ko
INSTALL drivers/net/usb/zaurus.ko
INSTALL drivers/usb/class/cdc-acm.ko
INSTALL drivers/usb/common/usb-common.ko
INSTALL drivers/usb/core/usbcore.ko
INSTALL drivers/usb/phy/phy-generic.ko
INSTALL drivers/usb/serial/ftdi_sio.ko
INSTALL drivers/usb/serial/usbserial.ko
INSTALL drivers/usb/storage/usb-storage.ko
INSTALL drivers/usb/usbip/usbip-core.ko
INSTALL drivers/usb/usbip/vhci-hcd.ko
DEPMOD 4.19.104-microsoft-standard

ziomarietto@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard$ sudo make install

sh ./arch/x86/boot/install.sh 4.19.104-microsoft-standard arch/x86/boot/bzImage
System.map "/boot"
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.19.104-microsoft-standard /boot/vmlinuz-4.19.104-microsoft-standard
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.19.104-microsoft-standard /boot/vmlinuz-4.19.104-microsoft-standard
update-initramfs: Generating /boot/initrd.img-4.19.104-microsoft-standard

W: mkconf: MD subsystem is not loaded, thus I cannot scan for arrays.
W: mdadm: failed to auto-generate temporary mdadm.conf file.

run-parts: executing /etc/kernel/postinst.d/pm-utils 4.19.104-microsoft-standard /boot/vmlinuz-4.19.104-microsoft-standard
run-parts: executing /etc/kernel/postinst.d/unattended-upgrades 4.19.104-microsoft-standard /boot/vmlinuz-4.19.104-microsoft-standard
run-parts: executing /etc/kernel/postinst.d/update-notifier 4.19.104-microsoft-standard /boot/vmlinuz-4.19.104-microsoft-standard

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe usbcore
modprobe: ERROR: could not insert 'usbcore': Unknown symbol in module, or unknown parameter (see dmesg)

[   49.155158] hv_balloon: Max. dynamic memory size: 26100 MB
[  907.557942] WSL2: Performing memory compaction.
[ 1809.551238] usbcore: Unknown symbol usb_hcd_amd_remote_wakeup_quirk (err -2)

@cezanne  : what happens now ?

Marietto2008, 2020-05-01:
root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# sudo modprobe usb-common

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# sudo modprobe hid-generic

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# sudo modprobe hid

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# sudo modprobe usbnet
modprobe: ERROR: could not insert 'usbnet': Unknown symbol in module, or unknown parameter (see dmesg)

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe cdc_ether
modprobe: ERROR: could not insert 'cdc_ether': Unknown symbol in module, or unknown parameter (see dmesg)

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe rndis_host
modprobe: ERROR: could not insert 'rndis_host': Unknown symbol in module, or unknown parameter (see dmesg)

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe usbserial
modprobe: ERROR: could not insert 'usbserial': Unknown symbol in module, or unknown parameter (see dmesg)

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe usb-storage
modprobe: ERROR: could not insert 'usb_storage': Unknown symbol in module, or unknown parameter (see dmesg)

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe cdc-acm
modprobe: ERROR: could not insert 'cdc_acm': Unknown symbol in module, or unknown parameter (see dmesg)

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe ftdi_sio
modprobe: ERROR: could not insert 'ftdi_sio': Unknown symbol in module, or unknown parameter (see dmesg)

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe usbip-core

root@DESKTOP-N9UN2H3:/mnt/c/Users/marietto2020/Desktop/WSL/WSL/Ubuntu-KVM/4.19.104/WSL2-Linux-Kernel-4.19.104-microsoft-standard# modprobe vhci-hcd
modprobe: ERROR: could not insert 'vhci_hcd': Unknown symbol in module, or unknown parameter (see dmesg)

thedozo, 2020-05-03:
@Marietto2008 
I have found another working method for USB devices in WSL 2: https://github.com/microsoft/WSL/issues/5165

it's working for me. You can try it too. it's little complex then USB/IP, but not give too much errors.

cezanne, 2020-06-06:
@Marietto2008 : Close this issue. Please reopen if you want.

closed 2020-06-06 by cezanne