cezanne/usbip-win #149: usbip Error in Windows 10 (WSL 2)

thedozo, 2020-04-28:
I am running usbip release **usbip-win 0.1.0** under Windows 10 (64-bit, build 2004) and WSL 2. My WSL distro is Ubuntu 20.04.

I followed the instruction of installing certificate into both trusted stores in local machine and everything. Also I have test signing on in my machine. still I am getting following error:

> **usbip: error: load_cert_context: failed to open certificate store: 5
> usbip: error: cannot load certificate: subject: USBIP Test
> usbip: error: bind_device: error binding device on busid 1-209: err: 0**

I can list my devices well though:

>  -  busid 1-191 (5986:210f)
>    Acer, Inc : unknown product (5986:210f)
>  - busid 1-209 (05ac:12a8)
>    Apple, Inc. : iPhone5/5C/5S/6 (05ac:12a8)
>  - busid 1-30 (8087:0a2a)
>    Intel Corp. : unknown product (8087:0a2a)
> 

What to do now? How can I solve it?

cezanne, 2020-04-29:
@thedozo : 
Please, check again the certificate installation. Refer https://github.com/cezanne/usbip-win/issues/147#issuecomment-619945893
If the installation has no problem, did you execute the bind command with administrator privilege?

thedozo, 2020-04-29:
> @thedozo :
> Please, check again the certificate installation. Refer [#147 (comment)](https://github.com/cezanne/usbip-win/issues/147#issuecomment-619945893)
> If the installation has no problem, did you execute the bind command with administrator privilege?

ok, after running as administration the error is gone, but new error is giving:

**usbip: error: failed to update driver: 103
usbip: error: bind_device: error binding device on busid 1-209: err: 0**

what to do now? I am using Windows 10 (x64) build 2004

thedozo, 2020-04-29:
@cezanne 

Here's my debug output:

C:\usbip>usbipd -d -4

usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240

usbip attach -r 172.16.160.1 -b 1-209

usbip: info: connection from 172.26.168.72:43778
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: open_stub_dev: invalid devno: 209
usbip: error: export_device: cannot open devno: 209
usbip: error: failed to export device: 1-209, err:-1
usbip: info: recv_pdu: request 0x8003: failed

Marietto2008, 2020-04-29:
which version of ubuntu are u running ?

cezanne, 2020-04-30:
@thedozo :

> usbip attach -r 172.16.160.1 -b 1-209 (from WSL)
>
>usbip: info: connection from 172.26.168.72:43778
usbip: info: recv_pdu: received request: 0x8003 - attach device
usbip: error: get_device_path: traverse_intfdevs failed, returned: 0
usbip: error: open_stub_dev: invalid devno: 209
usbip: error: export_device: cannot open devno: 209
usbip: error: failed to export device: 1-209, err:-1
usbip: info: recv_pdu: request 0x8003: failed

Did you succeed to bind ? If not, please update a driver of a iphone device manually as shown below.

[image]

Here is a [zip file](https://github.com/cezanne/usbip-win/files/4555601/stub_for_update.zip) for stub driver update.

thedozo, 2020-04-30:
@cezanne its saying iPhone driver is already updated and installing your given updated stub update couldn’t solve the issue. Still having same problem.

thedozo, 2020-04-30:
@cezanne  can you pls take a look at this: https://github.com/cezanne/usbip-win/issues/147#issuecomment-621645971

cezanne, 2020-04-30:
@thedozo : 
> @cezanne can you pls take a look at this: #147 (comment)

I examined your screenshot throughly. 😄 You might not use v0.1.0 usipd.exe. Logs in the following screenshot cannot be generated from v0.1.0. Please update usbipd.exe to v0.1.0.
[image]

> @cezanne its saying iPhone driver is already updated and installing your given updated stub update couldn’t solve the issue. Still having same problem.

Does your device manager look like mine? Then update manually, not automatically. 
[image]

Marietto2008, 2020-04-30:
ehy boys,what about the protocol mismatch ? I didn't understand if there is a solution or not.

cezanne, 2020-04-30:
@Marietto2008 
> ehy boys,what about the protocol mismatch ? I didn't understand if there is a solution or not.

The possibility of protocol mismatch is very low. @KNpTrue in #137  has not suffered from your case. How about trying to get the userspace binaries from them?

Marietto2008, 2020-04-30:
@cezanne : I see that he used ubuntu 18.04. So it's should be enough to install this version of ubuntu. Right now I tried with 16.04 (as suggested by your tutorial) and 20,but not 18.04.

cezanne, 2020-04-30:
@Marietto2008 : The usbip of linux has been using the same version(0x111) very long time. Thus, 16.04, 18.04 and 20.04 may have the same results. By the way, is there any reason not to use a usbip  distribution  package?

Marietto2008, 2020-04-30:
what do u mean with "usbip distribution package" ?

Marietto2008, 2020-04-30:
If 16.04, 18.04 and 20.04 may have the same results,it does not make sense di get the userspace binaries from one of these distros. Am I right ?

cezanne, 2020-04-30:
@Marietto2008 : 
> Plus,usbip is integrated with linux,it does not exists in a form of standalone package.

No. Ubuntu usually packages tools of a kernel named as linux-tool-xxx. Please try install that package.

thedozo, 2020-04-30:
> @thedozo :
>
>
> I examined your screenshot throughly. 😄 You might not use v0.1.0 usipd.exe. Logs in the following screenshot cannot be generated from v0.1.0. Please update usbipd.exe to v0.1.0. [...]

@cezanne 
bro I've double checked, it's version **0.1.0.** Downloaded from your Github repo. Check the version below:
```
C:\usbip>usbip version
usbip (usbip 1.0.0)
```
When I tried to update the stub driver manually as you said, it's showing me following:
[image]

Also if I try to attach the iphone into WSL, it's giving same error, debug log is given:
[image]

Marietto2008, 2020-04-30:
@cezanne : no

root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# usbip
WARNING: usbip not found for kernel 4.19.84-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-4.19.84-microsoft-standard
    linux-cloud-tools-4.19.84-microsoft-standard

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard
    linux-cloud-tools-standard

root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# apt install  linux-tools-standard
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-tools-standard

root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# apt install linux-cloud-tools-standard
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-cloud-tools-standard

root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# apt install linux-cloud-tools-4.19.84-microsoft-standard
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-cloud-tools-4.19.84-microsoft-standard
E: Couldn't find any package by glob 'linux-cloud-tools-4.19.84-microsoft-standard'
E: Couldn't find any package by regex 'linux-cloud-tools-4.19.84-microsoft-standard'

Marietto2008, 2020-04-30:
root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# apt install  linux-tools
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package linux-tools is a virtual package provided by:
  linux-tools-oem-osp1 5.0.0.1050.53
  linux-tools-gke-5.3 5.3.0.1018.8
  linux-tools-gke-5.0 5.0.0.1035.23
  linux-tools-virtual-hwe-18.04-edge 5.3.0.51.104
  linux-tools-virtual-hwe-18.04 5.3.0.51.104
  linux-tools-virtual-gkeop-5.0 5.0.0.47.104
  linux-tools-virtual 4.15.0.99.89
  linux-tools-oracle-lts-18.04 4.15.0.1038.47
  linux-tools-oracle-edge 5.3.0.1016.17
  linux-tools-oracle 5.3.0.1016.17
  linux-tools-oem 4.15.0.1080.84
  linux-tools-lowlatency-hwe-18.04-edge 5.3.0.51.104
  linux-tools-lowlatency-hwe-18.04 5.3.0.51.104
  linux-tools-lowlatency 4.15.0.99.89
  linux-tools-gke-4.15 4.15.0.1058.62
  linux-tools-generic-hwe-18.04-edge 5.3.0.51.104
  linux-tools-generic-hwe-18.04 5.3.0.51.104
  linux-tools-generic 4.15.0.99.89
  linux-tools-gcp-edge 5.3.0.1018.17
  linux-tools-gcp 5.3.0.1018.17
  linux-tools-azure-edge 5.3.0.1020.20
  linux-tools-azure 5.3.0.1020.20
  linux-tools-aws-lts-18.04 4.15.0.1066.69
  linux-tools-aws-edge 5.3.0.1017.18
  linux-tools-aws 5.3.0.1017.18
You should explicitly select one to install.

E: Package 'linux-tools' has no installation candidate
root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# apt install linux-tools-generic 4.15.0.99.89
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package 4.15.0.99.89
E: Couldn't find any package by glob '4.15.0.99.89'
E: Couldn't find any package by regex '4.15.0.99.89'
root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# apt install linux-tools-generic
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  linux-tools-4.15.0-99 linux-tools-4.15.0-99-generic
The following NEW packages will be installed:
  linux-tools-4.15.0-99 linux-tools-4.15.0-99-generic linux-tools-generic
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 4750 kB of archives.
After this operation, 22.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 linux-tools-4.15.0-99 amd64 4.15.0-99.100 [4746 kB]
Get:2 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 linux-tools-4.15.0-99-generic amd64 4.15.0-99.100 [1996 B]
Get:3 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 linux-tools-generic amd64 4.15.0.99.89 [2384 B]
Fetched 4750 kB in 1s (6015 kB/s)
Selecting previously unselected package linux-tools-4.15.0-99.
(Reading database ... 208699 files and directories currently installed.)
Preparing to unpack .../linux-tools-4.15.0-99_4.15.0-99.100_amd64.deb ...
Unpacking linux-tools-4.15.0-99 (4.15.0-99.100) ..............................................................................................................]
Selecting previously unselected package linux-tools-4.15.0-99-generic.........................................................................................]
Preparing to unpack .../linux-tools-4.15.0-99-generic_4.15.0-99.100_amd64.deb ...
Unpacking linux-tools-4.15.0-99-generic (4.15.0-99.100) ......................................................................................................]
Selecting previously unselected package linux-tools-generic.##########........................................................................................]
Preparing to unpack .../linux-tools-generic_4.15.0.99.89_amd64.deb ...
Unpacking linux-tools-generic (4.15.0.99.89) ...###############################...............................................................................]
Setting up linux-tools-4.15.0-99 (4.15.0-99.100) ...############################################..............................................................]
Setting up linux-tools-4.15.0-99-generic (4.15.0-99.100) ...######################################################............................................]
Setting up linux-tools-generic (4.15.0.99.89) ...##################################################################################...........................]
root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# usbip
WARNING: usbip not found for kernel 4.19.84-microsoft

thedozo, 2020-04-30:
> @cezanne : no
>
> root@DESKTOP-N9UN2H3:/mnt/j/Downloads/New6/Emulators/cmder# usbip
> WARNING: usbip not found for kernel 4.19.84-microsoft
>
> You may need to install the following packages for this [...]
you will not find that package into Ubuntu repostitory, because it's a kernel by microsoft. 
However, I have compiled usbip for that kernel version (4.19.84-microsoft-standard), you can use my pre-compiled usbip library if you want. You can get it from here: 
[libusbip.zip](https://github.com/cezanne/usbip-win/files/4557834/libusbip.zip)
extract and copy the library file into /lib directory, hopefully you can run usbip command from now on :)

Marietto2008, 2020-04-30:
it's not enough to have the file "libusbip.so.0" to run usbip. I need also the executable and some other libraries...

cezanne, 2020-04-30:
@thedozo : Couldn't you see a below dialog while updating a driver manually ?

[image]

Marietto2008, 2020-04-30:
anyway I don't see a reason to do that,since you are using 2.0 like me. and as @cezanne said : 16,18 and 20 may have the same result. I think that the usbip that I'm using now on ubuntu 20 come from the source code of the kernel 4.19.84 (recompiled for make work kvm). do u think that the fact that I have recompiled it,makes usbip stop to be the right version ?

thedozo, 2020-04-30:
> @thedozo : Couldn't you see a below dialog while updating a driver manually ?
> 
> [image]

@cezanne Ok, which option to choose? 
[image]
MTP USB or MTP IP? which drive I need to update manually?

thedozo, 2020-04-30:
@cezanne and If I give manual location of stub driver, it's showing updated driver is already installed.

cezanne, 2020-04-30:
@cezanne : Choose "Have Disk" and specify the location of a specially built stub driver, catalog and inf.

thedozo, 2020-04-30:
> @cezanne : Choose "Have Disk" and specify the location of a specially built stub driver, catalog and inf.

@cezanne so for MTP USB or MTP IP? which device driver I need to set the location for?

cezanne, 2020-04-30:
@thedozo : Then, can you uninstall a MTP driver ? And let me know the location of MTP driver. I try to install the driver.

thedozo, 2020-04-30:
> @thedozo : Then, can you uninstall a MTP driver ? And let me know the location of MTP driver. I try to install the driver.

Ok, I unistalled the MTP driver for iPhone and here's the driver location:
[image]

cezanne, 2020-04-30:
@thedozo : 
> I can't see any stub driver here...

Right.. Because you failed to bind. Your details screenshot is the same with mine before binding.

thedozo, 2020-04-30:
> @thedozo :
> 
> > I can't see any stub driver here...
> 
> Right.. Because you failed to bind. Your details screenshot is the same with mine before binding.

@cezanne so how to solve the binding problem? what to do now?

Marietto2008, 2020-04-30:
so,if I'm right,if you fix this problem,you will face the problem that I'm facing ? since you are using the same version of my kernel and my userland files,on ubuntu and windows ?

Marietto2008, 2020-04-30:
ohh no...i forgot : now you are using a new kernel : 4.19.104.

thedozo, 2020-04-30:
> ohh no...i forgot : now you are using a new kernel : 4.19.104.

yes, I recompiled another kernel and using it now, I couldn't even bind an USB pendrive with previous kernel. this kernel solved that.

cezanne, 2020-04-30:
@thedozo : 
> Can you bind your iphone? is there anything to do with driver version or so?

Yes, I can. I think that a MTP device driver may cause the problem.. Please answer the following question.

> @thedozo : Then, can you uninstall a MTP driver ? And let me know the location of MTP driver. I try to install the driver.

thedozo, 2020-04-30:
> @thedozo :
>
>
> Yes, I can. I think that a MTP device driver may cause the problem.. Please answer the following question.

@cezanne  how to uninstall MTP driver for iPhone, can you give me a screenshot?

thedozo, 2020-04-30:
@cezanne or can you take a look using Teamviewer? I can give you access in Teamviewer

Marietto2008, 2020-04-30:
so,I suppose that the new kernel solves the old problem,but creates a new one.

thedozo, 2020-04-30:
> so,I suppose that the new kernel solves the old problem,but creates a new one.

@Marietto2008 I had the issue with iphone binding since previous one, it's an old issue, not new. This kernel solved pendrive issue but can't solve iPhone issue.

Marietto2008, 2020-04-30:
@thedozo : can u give me the new kernel ? it should be the file vmlinux. thanks...

thedozo, 2020-04-30:
> @thedozo : can u give me the new kernel ? it should be the file vmlinux. thanks...

@Marietto2008 yes, I can. it's the bzImage file, not vmlinux. You can boot from it with .wslconfig file. Keep the image in your C:\Users\<username> folder and create .wslconfig file in the same folder.

[bzImage.zip](https://github.com/cezanne/usbip-win/files/4558027/bzImage.zip)

Here's my .wslconfig file content:

```
[wsl2]
kernel=C:\\Users\\abhiw\\bzImage
```
replace abhiw with your username

Marietto2008, 2020-04-30:
it's not bzImage. Right now I'm using the file vmlinux.

thedozo, 2020-04-30:
> it's not bzImage. Right now I'm using the file vmlinux.

oh no idea then. I'm using bzImage. my system is x86_64 and there's no vmlinux file in mine.

thedozo, 2020-04-30:
@Marietto2008 can you try boot from the image i have given by .wslconfig file? if you are under x64 system as mine then I think it will work.

cezanne, 2020-04-30:
@thedozo :
> @cezanne how to uninstall MTP driver for iPhone, can you give me a screenshot?

Maybe you did not install a MTP driver S/W. I don't know the uninstall because I have heard about MTP from you first time. But I'm trying to install a MTP.

Marietto2008, 2020-04-30:
I have tried. it is not recognized. I need the vmlinux file. I don't know why.

thedozo, 2020-04-30:
> @thedozo :
> 
> > @cezanne how to uninstall MTP driver for iPhone, can you give me a screenshot?
> 
> Maybe you did not install a MTP driver S/W. I don't know the uninstall because I have heard about MTP from you first time. But I'm trying to install a MTP.

how to install MTP driver??? I simply connect my iphone with my PC and this driver/iPhone appears automatically under **portable devices** section. I have iTunes installed though.

thedozo, 2020-04-30:
> I have tried. it is not recognized. I need the vmlinux file. I don't know why.

then you may need to compile the kernel by yourself!
Still I'm confused with all these workaround of WSL 2, seems bit strange. Can't even edit any file using VScode due to permission issue. It wasn't like that in old WSL1.

thedozo, 2020-04-30:
> but you should have it already. can u check ? u can save me some time
Ok, give me some time. I'm looking into it...

cezanne, 2020-04-30:
@thedozo :
> @cezanne or can you take a look using Teamviewer? I can give you access in Teamviewer

Please email me accesss information.

thedozo, 2020-04-30:
> @thedozo :
> 
> > @cezanne or can you take a look using Teamviewer? I can give you access in Teamviewer
> 
> Please email me accesss information.

@cezanne pls check email. Given. Take access.

cezanne, 2020-04-30:
@Marietto2008 

> I have tried. it is not recognized. I need the vmlinux file. I don't know why.

just unzip the bzImage,  then rename it to vmlinux.

thedozo, 2020-04-30:
@cezanne pls send me a mail when you will be ready for teamviewer session. I will give details again.

thedozo, 2020-04-30:
> but you should have it already. can u check ? u can save me some time

@Marietto2008 

Ok, here's vmlinux file from my machine, kernel **4.19.104** you can try to use it - 
[vmlinux.zip](https://github.com/cezanne/usbip-win/files/4558530/vmlinux.zip)

Marietto2008, 2020-04-30:
it's not the right file. Anyway I compiled mine. Have u been able to fix your problem ?

Marietto2008, 2020-04-30:
Nothing to do for me,man :

λ bind
usbip: info: bind_device: bind device on busid 1-209: complete
usbipd: info: starting usbipd (usbip 1.0.0)
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_sock.c:38:[build_sockfd] opening 0.0.0.0:3240
usbip: info: listening on 0.0.0.0:3240
usbip: info: connection from 192.168.1.6:50674
usbip: debug: C:\work\usbip-win\userspace\lib\usbip_network.c:156:[usbip_net_recv_op_common] version mismatch: 4353 273
usbip: debug: C:\work\usbip-win\userspace\src\usbipd\usbipd_accept.c:18:[recv_pdu] recv_pdu: could not receive opcode: 0

Marietto2008, 2020-04-30:
@thedozo : can u give me your userspace files ?

thedozo, 2020-04-30:
> it's not the right file. Anyway I compiled mine. Have u been able to fix your problem ?

@Marietto2008  not yet, @cezanne took a look into my issue via TV and hopefully he will investigate into it

thedozo, 2020-04-30:
> @thedozo : can u give me your userspace files ?

@Marietto2008 those files located in which folder? Tell me

Marietto2008, 2020-04-30:
I've created an ubuntu 18.04 vm and I'm using the kernel 4.19.104. I went on the usbip source files and I compiled it. When I run it,I see this error :

/usr/local/sbin/./usbip: error while loading shared libraries: libusbip.so.0: cannot open shared object file: No such file or directory

do you have a copy of this file ? u sent me a file like this,but it was compiled for the kernel 4.19.84. I need the same file compiled for the kernel 4.19.104

Marietto2008, 2020-04-30:
I have this situation :

[image]

[image]

it seems that that file is simlinked. but I don't know how to fix the error

[86 more replies]

closed 2020-05-01 by thedozo