cezanne/usbip-win #153: Other build configs

pasha-zzz, 2020-05-06:
Wait for other than x64/debug configs:

- release (x64 + x86)
-  maybe debug x86

pasha-zzz, 2020-06-01:
There are debug/release, x86/x64 builds - https://ci.appveyor.com/project/pasha-zzz/usbip-win

cezanne, 2020-12-28:
@pasha-zzz : Thanks. I created my own [appveyor CI](https://ci.appveyor.com/project/cezanne/usbip-win).

pasha-zzz, 2021-01-16:
Yes but there are no artifacts at this CI... nothing to download

cezanne, 2021-01-17:
@pasha-zzz : I didn't know that. I'll try.

cezanne, 2021-01-18:
@pasha-zzz : Some artifacts were roughly generated but making a zip for release is not easy. Can you help or guide?

pasha-zzz, 2021-01-18:
[image]

And after build script (Cmd):

```
@echo off

curl -sSL -o userspace\usb.ids http://www.linux-usb.org/usb.ids

for /d %%a in (Debug Release) do if exist %%a for /d %%b in (%%a\*) do for /d %%c in (%%b\*) do rmdir /s /q %%c

for /d %%a in (Debug Release) do if exist %%a for /d %%b in (%%a\*) do 7z a -ms -mqs -mx -myx -mf=BCJ2 -m0=LZMA:22:fb273:mc1000000000 -xr!*.pdb -xr!*.lib -xr!*.ilk -xr!*.iobj -xr!*.ipdb -i!%APPVEYOR_BUILD_FOLDER%\%%b\* -i!%APPVEYOR_BUILD_FOLDER%\README.md -i!%APPVEYOR_BUILD_FOLDER%\driver\usbip_test.pfx -i!%APPVEYOR_BUILD_FOLDER%\driver\stub\usbip_stub.inx -i!%APPVEYOR_BUILD_FOLDER%\userspace\usb.ids usbip-win_%%a_%%~nxb.7z

for /d %%a in (Debug Release) do if exist %%a for /d %%b in (%%a\*) do appveyor PushArtifact usbip-win_%%a_%%~nxb.7z

```

cezanne, 2021-01-25:
@pasha-zzz : Thanks. I have completed appveyor artifact setup using the following batch script.

```
if %PLATFORM% == x86 set PLATFORM=Win32
set output=%APPVEYOR_BUILD_FOLDER%\%CONFIGURATION%\%PLATFORM%
set outzip=usbip-win-%APPVEYOR_BUILD_VERSION%.zip

set files=usbip.exe usbipd.exe attacher.exe usbip_root.inf usbip_vhci.inf usbip_vhci.sys usbip_vhci.cat usbip_stub.inx usbip_stub.sys usbip_vhci_ude.inf usbip_vhci_ude.sys usbip_vhci_ude.cat usbip_vhci_ude.pdb usb.ids usbip_test.pfx

cd %output%
copy /y ..\..\userspace\usb.ids
copy /y ..\..\driver\stub\usbip_stub.inx
copy /y ..\..\driver\usbip_test.pfx

del /q /f %outzip%
7z a %outzip% %files%
```

closed 2021-01-25 by cezanne