cezanne/usbip-win #154: Request for win7 client build

betzuka, 2020-05-07:
Please would it be possible to do a release build for windows7?

Also does this client work with the default linux usbip server?

Thanks

betzuka, 2020-05-07:
Just an update, I have tried with release 0.10 using the vhci driver for windows7 which was built on 31/07/2019 with the inf file from 0.10. Driver installed fine, I can list the remote devices but the attach fails with 0x57:

`D:\tools\usbip>usbip list -r 192.168.1.224
Exportable USB devices
======================
 - 192.168.1.224
      1-1.2: unknown vendor : unknown product (f266:9a0a)
           : /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2
           : (Defined at Interface level) (00/00/00)`
`D:\tools\usbip>usbip attach -r 192.168.1.224 -b 1-1.2
usbip: error: usbip_vhci_attach_device: DeviceIoControl failed: err: 0x57
usbip: error: import device
usbip: error: read_devbuf: failed to read: err: 0x6`

szolnokit, 2020-05-19:
Same problem.
Server: Win10_x64/usbpid 1.0.0, works very well.

Client: Win7_x64.
VHCI diver from: https://github.com/cezanne/usbip-win/issues/57#issuecomment-516861019
VHCI driver installed susessfully.

Listing devices from client, working.
But client attach not working:
usbip: info: using port 10100 ("10100")
usbip: debug: C:\work\usbip-win\userspace\src\usbip\usbip.c:147:[run_command] running command: `attach'
usbip: debug: C:\work\usbip-win\userspace\src\usbip\usbip_vhci.c:60:[usbip_vhci_driver_open] device path: \\?\root#unknown#0000#{d35f7840-6a0c-11d2-b841-00c04fad5171}
usbip: debug: C:\work\usbip-win\userspace\src\usbip\usbip_attach.c:58:[import_device] got free port 
usbip: error: usbip_vhci_attach_device: DeviceIoControl failed: err: 0x57
usbip: error: import device
usbip: error: read_devbuf: failed to read: err: 0x6

Working clinet and drivers for Windows 7 (x86/x64) would be a better :)

cezanne, 2020-06-08:
@szolnokit @betzuka : A driver for win7 can be built with minor change as mentioned in README.md. Here is  [v0.1.0 win7(x64) driver](https://github.com/cezanne/usbip-win/files/4743857/usbip_vhci_v0.1.0_win7_x64.zip).

technoscavenger, 2020-07-30:
It took me a while to figure out how to use it on Windows Server 2008 R2, I have document some steps here:
[](https://8thstring.blogspot.com/2020/07/how-to-install-usbip-for-windows-on.html)

cezanne, 2020-08-29:
@technoscavenger : Thanks for your report.

@betzuka : This issue will be closed. Please reopen if you want.

closed 2020-08-29 by cezanne