cezanne/usbip-win #158: USB 3 SuperSpeed supported?

yenchee1970, 2020-05-13:
USBIP in Linux support SuperSpeed. Does this project support it as well? The best performance I tested between Linux USB server and Windows client is not more than 40MB/s on storage sharing. Is it an issue because of no SuperSpeed support? Thanks.

Oxalin, 2020-05-13:
Last time I was working on the code (which is about a year ago, sorry @cezanne for not having been able to continue the work I was doing at the time), it was not supported yet, even though it seemed relativelly simple to add.

cezanne, 2021-05-05:
@yenchee1970 : Please try with https://github.com/cezanne/usbip-win/releases/tag/v0.3.5-dev

windowsair, 2021-05-07:
@cezanne 
Hi, My USB 3.0 device is already working properly.😁

However the commands like detach, list, port  do not work properly in the dev version of the release page. The latest dev branch doesn't seem to have this problem.

cezanne, 2021-05-12:
@windowsair : 
> Hi, My USB 3.0 device is already working properly.😁
Oh, really? What is your device?

windowsair, 2021-05-12:
> @windowsair :
> 
> > Hi, My USB 3.0 device is already working properly.😁
> 
> Oh, really? What is your device?

A simple WinUSB device to test whether usbip supports superspeed endpoint. The case `bMaxBurst > 0` has not been tested.

cezanne, 2021-07-26:
@yenchee1970 : I have tested USB 3 stick with v0.3.5 version(works OK for both ude and wdm) even though no bandwidth measurement was done. I'll close this issue. Please reopen if you want.

closed 2021-07-26 by cezanne

windowsair, 2021-07-28:
Hi, @cezanne : I have a question about usb2.0 and usb3.0.

usb3.0 supports full-duplex communication, while earlier usbip seems to be built on the half-duplex communication of usb2.0.

I have not read the source code in detail, do you have any good ideas? Thanks!

cezanne, 2021-07-28:
@windowsair : I have no idea and there's no clue in usbip-win source code.