cezanne/usbip-win #159: Release/x64 cannot build

yenchee1970, 2020-05-15:
Just found Release/x64 build failed due to some header file is not included. I took a look at it and try to fix it by the following patch. Not sure if it is a correct way.

diff --git a/driver/vhci/vhci_dbg.h b/driver/vhci/vhci_dbg.h
index d0e8030..f6fc7aa 100644
--- a/driver/vhci/vhci_dbg.h
+++ b/driver/vhci/vhci_dbg.h
@@ -2,10 +2,10 @@

 #define DRVPREFIX      "usbip_vhci"
 #include "dbgcommon.h"
+#include "usbreq.h"

 #ifdef DBG

-#include "usbreq.h"
 #include "dbgcode.h"

 /* NOTE: LSB cannot be used, which is system-wide mask. Thus, DBG_XXX start from 0x0002 */
diff --git a/driver/vhci/vhci_pnp_vpdo.c b/driver/vhci/vhci_pnp_vpdo.c
index 8ecb18b..f91660a 100644
--- a/driver/vhci/vhci_pnp_vpdo.c
+++ b/driver/vhci/vhci_pnp_vpdo.c
@@ -3,6 +3,7 @@
 #include <wdmguid.h>
 #include <usbbusif.h>

+#include "usbip_proto.h"
 #include "vhci_pnp.h"
 #include "usbip_vhci_api.h"

cezanne, 2020-05-24:
@yenchee1970 : Thanks for your patch. I had not checked the release build. The master branch has been  patched based on your comment.

pasha-zzz, 2020-06-03:
Try to make autocompiling for  all 4 builds - https://ci.appveyor.com/project/pasha-zzz/usbip-win

cezanne, 2020-06-06:
@pasha-zzz : great!! thanks.. I would try..

closed 2020-06-10 by cezanne