cezanne/usbip-win #160: Detect disconnection of devices

fabiorush, 2020-05-18:
I'm using WSL 2 to build Android images and it was working well until the coming of Android 10. On this version they changed the fastboot flashing to use two stages: the first stage does the flash and reboot the device into the new fastboot, and in the new fastboot mode (second stage), flash the rest of the system.  The problem is that between the first and second stage, there is a moment when the device is rebooting and not connected to the host USB bus. The fastboot client waits it show up to beging the flashing of the second stage.

The problem is that using USBIP this moment where the device is diconnected doesn't shows to the WSL 2 environment, and the fastboot client thinks that the device is immediatly available after reboot. Here is some log showing this behavior:

`flus@DESKTOP-2MV215D:~/AOSP/PLATFORM$ fastboot flashall
--------------------------------------------
Bootloader Version...: b4s4-0.2-5736883
Baseband Version.....: g670-00030-191028-B-5972069
Serial Number........: 938AY07X9G
--------------------------------------------
Checking 'product'                                 OKAY [  0.059s]
Setting current slot to 'a'                        OKAY [  0.382s]
Sending 'boot_a' (65536 KB)                        OKAY [  2.810s]
Writing 'boot_a'                                   OKAY [  0.740s]
Sending 'dtbo_a' (8192 KB)                         OKAY [  0.450s]
Writing 'dtbo_a'                                   OKAY [  0.124s]
Sending 'vbmeta_a' (8 KB)                          OKAY [  0.120s]
Writing 'vbmeta_a'                                 OKAY [  0.065s]
Rebooting into fastboot                            OKAY [  0.059s]
fastboot: error: Failed to boot into userspace fastboot; one or more components might be unbootable.`

Instead of the error, it should be showing the message "< waiting for any device>".

I don't know if this is necessarily an issue or simply a limitation of the protocol, but it would be nice if we have a way to make the disconnections of USB devices on the USBIP server machine be reflected in the USBIP client. There is a easy way to make this work?

k-str50, 2020-07-19:
Hi,
this is a great tool.
Yes fabiorush I think this would be a good feature, because it's a little tricky to bring everything up.
If you work at night (being an astronmer) then it's not easy to handle it when it's dark.

cezanne, 2020-12-29:
@fabiorush : A stub server running on windows?

cezanne, 2021-01-14:
@fabiorush : I'll close this issue. Please reopen if you want.

closed 2021-01-14 by cezanne