cezanne/usbip-win #162: Unable to attach Device

Merith-TK, 2020-05-26:
```
C:\Tools\usbip>usbip.exe attach -r 192.168.0.113 -b 1-1.2
usbip: error: open vhci driver
usbip: error: read_devbuf: failed to read: err: 0x6
```

is the full error i get when trying to attach any device 
[image]

kmvi-dev, 2020-05-26:
@Merith-TK 
There is not enough data that you provided to detect problems starting usbip.exe. But I think that first you need to run the command line as administrator. If this does not help, then check if you forgot to install the usbip_vhci.sys driver

Merith-TK, 2020-05-26:
ran as admin, didnt work

and it tells me there is a probelm when i try to use the install.inf file,

so how would i install it properly? if you included a batch script that
would work

Merith-TK, 2020-05-26:
okay it let me install the driver, but it still reuses to connect with the same error

kmvi-dev, 2020-05-27:
@Merith-TK
Мake sure your driver version is install and running. To do this, use the device manager. See screenshot 
http://prntscr.com/soidnu
Otherwise, try the item with README.md -> Install-> Windows USB / IP client-> install USB / IP VHCI driver-> Manual install.
And of course, the Reporting Bug item has not been canceled. By doing it, you get a more specific answer to your problem.

Merith-TK, 2020-05-27:
okay yeah i made sure to follow that (forgot the `usbip.exe install` part)

 When i ran the attach command, i heard a USB connect/disconnect sound, but the controller isnt there

Merith-TK, 2020-06-07:
NEW PC, i can install the vhci driver, but when i run `usbip install` it bluescreens my computer

kmvi-dev, 2020-06-07:
@Merith-TK 

recommended it to me  from cezanne :
befor install driver, you should first remove an old usbip_vhci.inf.
CMD> pnputil /enum-drivers
You can list up all drivers installed. If you find an old usbip_vhci.inf as (on example) oem2.inf , you must delete this entry
 CMD> pnputil /delete-driver oem2.inf

Merith-TK, 2020-06-07:
Fresh brand new computer
There are no old drivers aside from the latest release @kmvi-dev

ruebenwurz, 2020-06-09:
Same here...
I tried 0.1.0 too... same problem, can someone help me?

usbip.exe list -r 172.30.48.2
Exportable USB devices
======================
 - 172.30.48.2
      1-1.1: Seiko Epson Corp. : Printer (Composite Device) (04b8:0803)
           : /sys/devices/platform/ahb/1b000000.usb/usb1/1-1/1-1.1
           : (Defined at Interface level) (00/00/00)
           :  0 - Mass Storage / SCSI / Bulk-Only (08/06/50)
           :  1 - Printer / Printer / Bidirectional (07/01/02)

--------------------

usbip.exe attach -r 172.30.48.2 -b 1-1.1
usbip: error: recv op_common
usbip: error: read_devbuf: failed to read: err: 0x6

New installed PC Win 10 Pro 190318-1202 Testmode activated
Server: OpenWRT - usbip 2.0

Connecting with linux no problems

Merith-TK, 2020-06-09:
RIght now its about the fact that when i try to use `usbip.exe install`, my computer bluescreens

rexx0520, 2020-06-11:
having the 
```
usbip: error: recv op_common
usbip: error: read_devbuf: failed to read: err: 0x6
``` 
issue too.

ruebenwurz, 2020-06-13:
I've figured it out...
With the new Version (0.2) i can't establish a connection.
I solved it by following steps (downgrade to 0.1.0)

1. Uninstall the devices in the hardware in the device manager (or remove the device, and list all drivers with "pnputil /enum-drivers", then uninstall the usbip_vhci.inf with "pnputil /delete-driver oem?.inf"
If you find an usbip_root try the same procedure for this driver

2. delete any file of usbip, or use an new folder

3. run regedit an delete any usbip entries
3.1 if Windows "Testmode" not is not active, enable Windows "Testmode" by opening a PowerShell / CMD in **Administrator-Mode**
Insert **bcdedit.exe** /set TESTSIGNING ON
an press Return :)

4. reboot the computer
5. Download the "old" version 0.1.0 (to be safe... all files)
https://github.com/cezanne/usbip-win/releases/tag/v0.1.0
6. install the PFX by right-clicking in the explorer
usbip_test.pfx (you need the password **usbip** to open/install it)

Certificate should be installed into
"Trusted Root Certification Authority" in "Local Computer" (not current user) and
"Trusted Publishers" in "Local Computer" (not current user)

7. Run PowerShell or CMD as an Administrator
go to the usbip folder
pnputil /add-driver usbip_vhci.inf

8. Start Device manager
Choose "Add Legacy Hardware" from the "Action" menu.
Select "Install the hardware that I manually select from the list".
Click "Next".
Click "Have Disk", click "Browse", choose the copied folder, and click "OK".
Click on the "USB/IP VHCI Root", and then click "Next".
Click Finish at "Completing the Add/Remove Hardware Wizard".

And there you go :)

Cheers!

ruebenwurz, 2020-06-13:
Windows could not install the VHCI_Driver correctly.
usbip_vhci
failed IRP_MN_QUERY_ID-BusQueryDeviceID

cezanne, 2020-12-29:
Please, use [a recent release](https://github.com/cezanne/usbip-win/releases/tag/v0.3.1) and report any problem.

cezanne, 2021-01-11:
@Merith-TK : Close this issue. Please reopen If you need.

closed 2021-01-11 by cezanne