cezanne/usbip-win #164: USBPcap capture

Shamshala, 2020-05-27:
Hi,

sorry for little offtopic but i think it's more reasonable here. Does anybody know whether is possible to capture communication between USB device and USBIP server? (let say "before ethernet") I thought it should be possible via USBPcap but when i tested it (Windows 10 v1909, cezanne's USBIP server, Ubuntu as a client) my idea failed big time :slightly_smiling_face: .

Any advice would be appreciated

cezanne, 2020-06-06:
@Shamshala : Yeah, it's not an easy task to run a usbip-win stub driver with a USBPcap driver. Unbind error would be the same sort of the problem. BTW, what's your intention of capturing on a usbip server, not locally?

Shamshala, 2020-06-06:
It turns out the STUB driver was "stuck" after power-loss. After i uninstalled the driver, un/bind is working again without issues. (it is repeatable)

I have better access to the server device and if anything, i don't care about the server device what is complete opposite for the more sensitive client side. :slightly_smiling_face: (sure, backup is the way but still it'll be more hassle)

I'm not sure if i understand the new release's changelog correctly but wouldn't be possible to USBPcap that now?

Thank you for your great work btw! USB/IP is really nice piece of software so i'm glad there is somebody who is taking care about it! :clap:

cezanne, 2020-06-06:
> I'm not sure if i understand the new release's changelog correctly but wouldn't be possible to USBPcap that now?

Sadly, there's no code change for stub device in the new release(v0.2.0). USBPcap may not work with stub. I'll check..

Wqrld, 2020-09-09:
With 3.0 you can use usbpcap, tested with wireshark on win10 to a raspberry pi and ubuntu server.

Shamshala, 2020-09-09:
Oh, thank you for letting me know! :+1: I'll definitely have to try that.

Shamshala, 2020-09-13:
> With 3.0 you can use usbpcap, tested with wireshark on win10 to a raspberry pi and ubuntu server.

Have you compiled debug or release version? I tried it myself but unfortunately USBPcap still gets nothing.

Wqrld, 2020-09-13:
> > With 3.0 you can use usbpcap, tested with wireshark on win10 to a raspberry pi and ubuntu server.
> 
> Have you compiled debug or release version? I tried it myself but unfortunately USBPcap still gets nothing.

I am using the release UDE version, check if your attaching to the correct USB interface. Strangely i did have usbpcap send a good amount of empty packets (just headers and no data) so it might not be 100% ready for use.

Shamshala, 2020-09-16:
@Wqrld Ok, now i know what is the problem, my UDE driver never got compiled because of missing /Qspectre (even though it was auto-set to enabled).
And on top of that when it was in a process of installing it, Win10 decided it's the perfect time to restart for update. Since then VisualStudio throws at me bunch of errors while compilation (even after reinstall) :rofl: :cry: Love you Gates...

Please, could you upload it somewhere for me? :slightly_smiling_face:

Wqrld, 2020-09-16:
@Shamshala You can download it from the releases page, no need to compile anything (make sure you grab the latest 0.3 UDE and all files)

Shamshala, 2020-09-16:
Oh now i see the actual problem. I haven't noticed you are talking about RaspberryPi/Ubuntu server. I need to use USBPcap on Win10 server side.

cezanne, 2020-12-29:
@Shamshala : I'll close this issue. Please reopen if you need.

closed 2020-12-29 by cezanne