cezanne/usbip-win #165: Issues with binding Silicon Labs CP210x

kumekay, 2020-05-30:
I have a number of development boards that use cp120x  usb-serial chips.

My intent is use Windows 10 host (2004) and WSL 2 (ubuntu 18.04).

I compiled WSL2 kernel with usbip support and required modules, downloaded build 0.10.0 of the tool and installed certificates as described in the readme. And most of my devices work just fine, for example ST-Link and Profilic based USB-UART boards, I mean they are binded successfully on windows and then attached in WSL 2.

However I experience difficulties with all mine CP120x based boards. I've tried to use another linux machine as a host and attach them in WSL2 - it also works fine, but using Windows as host I cannot bind them.

Example output:

```
PS C:\Users\serge\Tools\uip> .\usbip.exe list -l
...
 - busid 1-185 (10c4:ea60)
   Cygnal Integrated Products, Inc. : CP210x UART Bridge / myAVR mySmartUSB light (10c4:ea60)
```

```
PS C:\Users\serge\Tools\uip> .\usbip.exe --debug bind -b 1-185
usbip: debug: C:\work\usbip-win\userspace\src\usbip\usbip.c:147:[run_command] running command: `bind'

usbip: error: failed to update driver: 0
usbip: error: bind_device: error binding device on busid 1-185: err: 0
```

I've tried to compile latest code from the git (commit 6d82d0a4f3bf28d2700e257ad0cd2aacc5b2fc89 ), but got a slightly different error:

```
PS C:\Users\serge\Projects\usbip-win\Debug\x64> .\usbip.exe --debug bind -b 1-185
usbip: debug: C:\Users\serge\Projects\usbip-win\userspace\src\usbip\usbip.c:147:[run_command] running command: `bind'

usbip: error: calc_hash: path not found: C:\Users\serge\AppData\Local\Temp\stuD670.tmp\usbip_stub.inf
usbip: error: failed to update driver: 2
usbip: error: bind_device: error binding device on busid 1-185: err: 0
```

@cezanne Do you know something special about these USB-UART bridges?
And thank you so much for this wonderful tool!

kumekay, 2020-06-01:
I finally managed to make cp210x work by manually selecting USP/IP STUB driver for the device,
though it's behavior is not always correct when I try to attach a different cp210x device, bind command reports success, but then attach in linux fails ( in device manager device is listed as serial adapter, but not usd-ip stub)

cezanne, 2020-06-06:
@kumekay : 

This error means that you did not copy usbip_stub.inx into a working directory. 
Well, usbip.exe error message should be more friendly.

```
usbip: error: calc_hash: path not found: C:\Users\serge\AppData\Local\Temp\stuD670.tmp\usbip_stub.inf
usbip: error: failed to update driver: 2
usbip: error: bind_device: error binding device on busid 1-185: err: 0
```

I remembered that there's no case of cp210x on stub. cp210x on vhci seems to work. (refer #144)
I have no cp210x device. So, your windows kernel log is required to resolve your issue.

kumekay, 2020-07-28:
Ok, it doesn't seem to be a usbip-win issue, so close for now

closed 2020-07-28 by kumekay