cezanne/usbip-win #166: Distribute Release as a single zip PLEASE

Merith-TK, 2020-06-07:
As the title says, could you PLEASE distribute related files together? so i dont have to download every single file individually, find them in my downloads folder, and move them to the new location?

Merith-TK, 2020-06-07:
[usbip_win.zip](https://github.com/cezanne/usbip-win/files/4741203/usbip_win.zip)
I packaged this by just downloading and zipping

koudis, 2020-06-25:
Look at
https://github.com/cezanne/usbip-win/issues/162

You must uninstall previous usbip_vhci driver and the Windows test mode must be enabled.

septatrix, 2020-10-18:
Why are these files distributed separate anyway? It feels like they could simply be included in the exe (which could unpack them if necessary)

Merith-TK, 2020-10-20:
They are inherently seperate because of how windows loads crap. also easier to distribute.
However. having seperate files is DIFFERENT, than having a release, for every single file

cezanne, 2020-12-29:
@Merith-TK : I agree that a single zip would be better. I'll try.

closed 2020-12-29 by cezanne

Merith-TK, 2020-12-29:
If you compile to a folder, you can simply zip that folder and release it.