cezanne/usbip-win #167: Binding to new device

mousumih, 2020-06-08:
Hello - I have recently started using USBIP with WSL2 and it seems to work. But I have one issue I need some help with. My device shows up at first as 1-130. I can bind to it from WSL2 and flash new firmware into the USB device. But the way my device works, on flashing this new firmware, it changes to a new device 1-101. I need to now talk to 1-101 from inside WSL2. But if I try to bind 1-101 before it is present, it fails from either side (windows and wsl2). Is there any way to prematurely bind to this device before it shows up in the device list so my device can continue once it changes from 1-130 to 1-101?

koudis, 2020-06-25:
This is problem of WSL2. You must compile drivers manually  - for WSL2 Linux kernel.

cezanne, 2020-12-29:
@mousumih : What's the purpose of your suggested premature bind ? Just remove annoyance of typing bind command? 
Otherwise client disconnection or whatever? 

Current busid scheme of usbip-win depends on device instance id. I think a more elaborated scheme would resolve your issue.

cezanne, 2021-01-14:
@mousumih : I'll close this issue. Please reopen if you need.

closed 2021-01-14 by cezanne