cezanne/usbip-win #169: Does usbip-win support winusb and webusb?

versaloon, 2020-06-21:
I tested usbip with a winusb+webusb device, but failed. And I recoreded all the transactions on the server side(server side is a PC software enumlating a USB device, client side is usbip-win):
[image]
[image]

In the image, usbip_server: send indicates the data sent to the usbip-win32.
While usbip_server: recv indicates the data received from the usbip-win32.

I found that after IMPORT, the first URB is quite strange, normally, it should be a ReadDeviceDescriptor, but in the image, it's ReadConfigurationDescriptor.

cezanne, 2020-06-30:
@versaloon : Which version did you test ?

versaloon, 2020-07-03:
> @versaloon : Which version did you test ?

0.2.0, I will later test the latest version

windowsair, 2020-07-16:
It seems that the order of requests has always been the same. In addition, the winusb request will not be initiated...

cezanne, 2021-01-20:
@versaloon : Could you describe a your problem in more detail? I have confirmed that an attached device is queried by a simple WinUSB application using both vhci versions(ude or wdm).

versaloon, 2021-02-07:
@cezanne OK, I will try the latest version.

closed 2021-03-07 by versaloon