cezanne/usbip-win #170: seems the roothub in new vhci(0.2.0) does not start

versaloon, 2020-06-21:
I installed the 0.2.0 usbip-win and updated the vhci. But after vhci is not usable, and in the device manager, there is a USB Root Hub problem:

Device status:
This device is not working properly because Windows cannot load the drivers required for this device. (Code 31)

The request is not supported.

swamiller, 2020-06-25:
I have the same issue.

koudis, 2020-06-25:
Look at
#162

You must uninstall previous usbip_vhci driver and the Windows test mode must be enabled.

versaloon, 2020-06-27:
> Look at
> #162
> 
> You must uninstall previous usbip_vhci driver and the Windows test mode must be enabled (mentonied in README).

Yes, I know that the old driver should be removed, but the roothub does not work in device manager.
My pc is already in test mode.

swamiller, 2020-06-29:
I was in test mode and I did a clean install using the latest version.  The root hub would not work with the driver.

I rolled back to v0.1.0 and it now works.

cezanne, 2020-06-30:
@versaloon : v0.2.0 had been tested just in my environment and a root hub did start well. Indeed, this version was very experimental and failed to get an expected result. If you need a virtual root hub, please use newly uploaded v0.3.0.

versaloon, 2020-07-03:
> @versaloon : v0.2.0 had been tested just in my environment and a root hub did start well. Indeed, this version was very experimental and failed to get an expected result. If you need a virtual root hub, please use newly uploaded v0.3.0.

OK, I will test the latest version. Actually, if the rooh hub did not start, usbip will fail to connect to vhci

cezanne, 2021-01-11:
@versaloon : This issue is too old and related to obsolete v0.2.0. So 'll close it.

closed 2021-01-11 by cezanne