cezanne/usbip-win #171: USBIP Win - Sign by EV CodeSign certificate

koudis, 2020-06-25:
Hello all,

it will be really nice to have usbip win driver signed properly. The driver must be signed by EV CodeSign cert which is not cheap.

Once the community agrees that the driver is stable I can discuss with our CEO about signing the driver by our EV Cert.

@cezanne let me known when the EV CodeSign will be needed.

cezanne, 2020-06-26:
@koudis: It's great to hear the news from you. EV certificate will be required for using usbip-win at production level.

But usbip-win kernel drivers are still unstable and lacks some functionality such as application compatibility. Current WDM-based vhci does not solve several issues such as #111 even though a commercial virtual USB product has no problem.
Maybe, most application and USB libraries depend on microsoft provided pure root hub(usbhub.sys or usbhub3.sys).

Thus, I’m currently developing a kmdf-based vhci via UDE(USB device emulation).

hnwentao, 2020-07-12:
Come on

Ale268, 2020-08-08:
@cezanne 
When do you expect the Project to be this far? 

I am really interested in using this code without breaking my computers security. 

Thanks anyway for the work yet done!

cezanne, 2020-08-29:
@Ale268 : 
> When do you expect the Project to be this far?

My hope answers that this project requires 4~6 months until vhci(UDE) deserves an EV certificate. 😕

saulrh, 2020-10-18:
Most anti-cheat software complains when driver test signing is on, meaning I can't use `usbip` until the client drivers are signed for release.

Is there a specific list of issues that are blocking the certification process or a concrete roadmap for a production version? I'd like to see if there's anything I can do, even if I only ever did device drivers on linux and that was years ago.

septatrix, 2020-10-18:
> My hope answers that this project requires 4~6 months until vhci(UDE) deserves an EV certificate.

Would this also eliminate the need for `bcdedit.exe /set TESTSIGNING ON`? If so I am very eager to get this as it allows using USBIP on a windows PC with secure boot enabled.

jenglilu, 2020-10-21:
usbip_win project target on Win10(x64) OS, build usbip driver(with EV Code Signing). Install usbip driver on security boot with trusted platform module. The driver cannot be installed due to an exception(error:52).
Does the driver fail to install must go through WHQL inspection and review?

DocMAX, 2021-01-08:
You can use EfiGuard!

mahdibx, 2021-01-16:
Any news on this topic? could we organize a crowdfunding if the certification is expensive?

versaloon, 2021-03-06:
Any progress about the certification?
We want to use usbip binaries in our commercial software, and can help to get the driver signed.
Is there any dedicated time for the code sign certification?

versaloon, 2021-03-06:
BTW: we plan to use usbip as PC driver for our wireless USB hardwares

dpvdberg, 2021-04-14:
This would be great :)

koudis, 2021-04-15:
Hi all :),

we are still waiting for stable version. How it looks, @cezanne ?

MinHyukPark121, 2021-05-12:
Hi everyone, do we have any updates on this? Even an update on the estimate would be nice :)

joaoabreufilho, 2021-07-09:
Hi everyone, do we have any updates on this? Even an update on the estimate would be nice (:

TheMohawkNinja, 2021-08-30:
@cezanne any updates? Once the cross-signed cert gets implemented, I can remove the need for a hardware USB switch for my project.

kadrim, 2021-09-08:
very intersted on this topic :-)

sensiki, 2021-09-25:
Any news on this topic?

alexmi256, 2021-11-17:
This would be nice to have. AFAICT it's about $700USD per year for a signature https://www.digicert.com/order/order-1.php
@cezanne I'd be willing to donate something and I'm sure others would as well

maxdd, 2022-01-11:
up

cezanne, 2022-06-18:
@forlayo: I would appreciate your EV certificate. However, usbip-win vhci drivers should get attestation sign at MS partner portal after the EV certificate is registered in my partner portal. But I'm not sure that an EV certificate can be registered into multiple accounts. If it's not possible, you may be asked to provide your partner account or create my account on your partner portal.
Or you can contribute to sign vhci drivers yourself.

A newly released [0.3.6-dev package](https://github.com/cezanne/usbip-win/releases/tag/v0.3.6-dev)  has MS signed vhci drivers. I managed to acquire an EV certificate but its validation period will expire soon. Thus, another EV might be needed.

lebtron, 2022-06-19:
> A newly released [0.3.6-dev package](https://github.com/cezanne/usbip-win/releases/tag/v0.3.6-dev) has MS signed vhci drivers. I managed to acquire an EV certificate but its validation period will expire soon. Thus, another EV might be needed.

I confirm this works as expected.

MinHyukPark121, 2022-06-20:
> A newly released [0.3.6-dev package](https://github.com/cezanne/usbip-win/releases/tag/v0.3.6-dev) has MS signed vhci drivers. I managed to acquire an EV certificate but its validation period will expire soon. Thus, another EV might be needed.

Could we know when the current EV Certificate will expire?

maxdd, 2022-06-20:
I guess you should wrap

      Enable test signing
      > bcdedit.exe /set TESTSIGNING ON
      reboot the system to apply

in something like "if not a signed release"

cezanne, 2022-06-20:
@MinHyukPark121 :

> Could we know when the current EV Certificate will expire?

Maybe after 3 months. However, once signed package can be safely installed with no test mode even though the certificate expires. Expiration matters only for package signing.

cezanne, 2022-06-20:
@maxdd:

> in something like "if not a signed release"

Good comment. thanks.

paulpv, 2022-09-21:
Happy to contribute to signing cert!

As an alternative, would something like EfiGuard be of any use to workaround the signing requirement in trusted environments?
https://muffsec.com/blog/how-to-use-efiguard-to-disable-patchguard/

CpServiceSpb, 2022-09-27:
If somebody send me OV sign for signing I will try to sign drivers by the sign.
I did it with my OV sign successfully and drivers were installed under Win10 LTSC19 well.

forlayo, 2022-11-25:
@cezanne if you can guide me to set up my certificate on partner portal to sign the driver I'll be happy on helping. I've just received my EV certificate today.

CpServiceSpb, 2022-11-26:
@forlayo if you can you can try it to your own to sign the drivers using utilities from Visual Studioand DDK and got the EV by you.
Uf you doon' t have the utilities I can send you it which I signed my driver by OV on previous year.

forlayo, 2022-11-28:
@CpServiceSpb It is first time I try to sign kernel model drivers so I am a bit lost. 

If I set the "Sign Mode" of libdrv project of this repo to "Production Sign" it ask to put a "Cross-Signing Certificate". And as far as I am aware the drivers are now signed without cross-signing certificates, as [Microsoft deprecated that way](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates) in favor of [Windows HLK](https://learn.microsoft.com/en-us/windows-hardware/test/hlk/getstarted/windows-hlk-getting-started).

Then I understand the process as:

1. Having a partners account with Microsoft ( done )
2. Having an EV and add it to partners account ( done ).
3. Download Microsoft VHLK -> https://learn.microsoft.com/en-us/windows-hardware/test/hlk/
4. Mount VHLK using Hype-V virtual machine -> https://learn.microsoft.com/en-us/windows-server/virtualization/hyper-v/get-started/create-a-virtual-machine-in-hyper-v
5. Set the environment with our driver and run the Windows Hardware Compatibility playlist, it's neede to have the tests passing.-> https://learn.microsoft.com/en-us/windows-hardware/test/hlk/
6. Create a submision package (.hlkx)
7. Submit the package on partners portal under hardware, as explained here -> https://www.ssl.com/how-to/signing-kernel-mode-drivers-for-windows-using-evcs-or-ovcs-certificates/

I have no problems on following this path, however I would like to have confirmation from @cezanne about this is the correct path and if possible to know how the driver should be compiled/signed; as apparently I can't set "Production Sign" there.

forlayo, 2022-11-28:
I am checking also "[Windows 10 attestation signed drivers"](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-reqs#windows-10-attestation-signed-drivers) possible path, [which looks easier](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-attestation?source=recommendations).

But not sure if it would suffice, would be great if someone with experience with this sutff confirms :)

forlayo, 2022-11-28:
Ok Windows 10 attestation signed works, let me know when you need me to sign usbip-win drivers ( as I saw that 0.3.6 is already signed and published. )

CpServiceSpb, 2022-11-29:
@forlayo I signed my drivers by old sign issued on 2021 year worked fin on some W10.
On freshest one I didn' t test.
I did it with cross-signed certificate, not on "new way" .
I can either try to sign drivers by your EV sign if you send it or send you utilities I used.
But some additional lifehacks may be required.

Btw, is there USP-IP freshest driver for Win7 ?
I am still with it also.

TheMohawkNinja, 2022-11-30:
> @forlayo I signed my drivers by old sign issued on 2021 year worked fin on some W10. On freshest one I didn' t test. I did it with cross-signed certificate, not on "new way" . I can either try to [...]

You mention a cross-signed cert. Is that now in the repo? Or did you purchase your own cross-signed cert?

CpServiceSpb, 2022-11-30:
I used at those moment curremt cross sign certificate added to certificate
storage of localachine.

Regards, CpServiceSPb

maxdd, 2023-02-22:
News here?

forlayo, 2023-02-23:
> You mention a cross-signed cert. Is that now in the repo? Or did you purchase your own cross-signed cert?

The cross-signing certificate is a way of sign drivers that was deprecated by Microsoft, then it's needed to use an EV and being enrolled on Microsoft Partner portal. Then you've the option of getting Microsoft attestation which is enough for installing the driver on regular devices, or passing the full validation that needs VHLK and so on.

I am signing other drivers with EV and partner portal, and it's fine for a general usage. I can contribute with this project to sign the driver when it's needed; as right now it has a version already signed that works, then it's not needed to sign it again.

MKPang, 2023-05-03:
Hello, I've been using [https://woshub.com/how-to-sign-an-unsigned-driver-for-windows-7-x64/](https://woshub.com/how-to-sign-an-unsigned-driver-for-windows-7-x64/) to successfully self sign some old drivers I had for a USB TV Tuner (in Windows 11). Would this help for self signing usbip drivers? I had a go and couldn't get it to work server side (I don't have enough knowledge of Windows drivers). However, if somebody believes this option may work, I can provide a script to carry out the steps (of which, some are erroneous, by the way) mentioned in the provided website.

maxdd, 2023-05-05:
@MKPang doesn't it work because we need kernel mode and not user mode? Where are you stuck at?