cezanne/usbip-win #172: BSOD

cbuliga-ab, 2020-06-25:
Client: W10
Server: RPi

Here is the dump:
SYSTEM_SERVICE_EXCEPTION (3b)
An exception happened while executing a system service routine.
Arguments:
Arg1: 00000000c0000420, Exception code that caused the bugcheck
Arg2: fffff80464541037, Address of the instruction which caused the bugcheck
Arg3: ffffb68c624a9cb0, Address of the context record for the exception that caused the bugcheck
Arg4: 0000000000000000, zero.

Debugging Details:
------------------

*** WARNING: Unable to verify timestamp for usbip_vhci.sys

KEY_VALUES_STRING: 1

PROCESSES_ANALYSIS: 1

SERVICE_ANALYSIS: 1

STACKHASH_ANALYSIS: 1

TIMELINE_ANALYSIS: 1

DUMP_CLASS: 1

DUMP_QUALIFIER: 400

BUILD_VERSION_STRING:  18362.1.amd64fre.19h1_release.190318-1202

SYSTEM_MANUFACTURER:  Dell Inc.

SYSTEM_PRODUCT_NAME:  Latitude 7480

SYSTEM_SKU:  07A0

BIOS_VENDOR:  Dell Inc.

BIOS_VERSION:  1.14.1

BIOS_DATE:  04/03/2019

BASEBOARD_MANUFACTURER:  Dell Inc.

BASEBOARD_PRODUCT:  00F6D3

BASEBOARD_VERSION:  A00

DUMP_TYPE:  2

BUGCHECK_P1: c0000420

BUGCHECK_P2: fffff80464541037

BUGCHECK_P3: ffffb68c624a9cb0

BUGCHECK_P4: 0

EXCEPTION_CODE: (NTSTATUS) 0xc0000420 - An assertion failure has occurred.

FAULTING_IP: 
usbip_vhci+1037
fffff804`64541037 cd2c            int     2Ch

CONTEXT:  ffffb68c624a9cb0 -- (.cxr 0xffffb68c624a9cb0)
rax=0000000000000004 rbx=ffffa408c3cc0dd0 rcx=0000000000000003
rdx=0000000000000000 rsi=0000000000000000 rdi=ffffa408bcfb5550
rip=fffff80464541037 rsp=ffffb68c624aa6a0 rbp=0000000000000000
 r8=0000000004000006  r9=0000000000000000 r10=fffff80464556640
r11=ffffb68c624aa6f8 r12=0000000000000001 r13=0000000000000000
r14=ffffa408bcfb5550 r15=ffffa408bc9d42f0
iopl=0         nv up ei pl nz na pe nc
cs=0010  ss=0018  ds=002b  es=002b  fs=0053  gs=002b             efl=00040202
usbip_vhci+0x1037:
fffff804`64541037 cd2c            int     2Ch
Resetting default scope

BUGCHECK_STR:  0x3B_c0000420

CPU_COUNT: 4

CPU_MHZ: b58

CPU_VENDOR:  GenuineIntel

CPU_FAMILY: 6

CPU_MODEL: 8e

CPU_STEPPING: 9

CPU_MICROCODE: 6,8e,9,0 (F,M,S,R)  SIG: B4'00000000 (cache) B4'00000000 (init)

BLACKBOXBSD: 1 (!blackboxbsd)

BLACKBOXNTFS: 1 (!blackboxntfs)

BLACKBOXPNP: 1 (!blackboxpnp)

BLACKBOXWINLOGON: 1

CUSTOMER_CRASH_COUNT:  1

DEFAULT_BUCKET_ID:  WIN8_DRIVER_FAULT

PROCESS_NAME:  usbip.exe

CURRENT_IRQL:  0

ANALYSIS_SESSION_HOST:  US05MW15597

ANALYSIS_SESSION_TIME:  06-25-2020 14:32:22.0184

ANALYSIS_VERSION: 10.0.18362.1 amd64fre

LAST_CONTROL_TRANSFER:  from 0000000000000001 to fffff80464541037

STACK_TEXT:  
ffffb68c`624aa6a0 00000000`00000001 : fffff804`536a0d99 00000000`00000010 fffff804`645436e3 ffffa408`c3cc0dd0 : usbip_vhci+0x1037
ffffb68c`624aa6a8 fffff804`536a0d99 : 00000000`00000010 fffff804`645436e3 ffffa408`c3cc0dd0 00000000`00000018 : 0x1
ffffb68c`624aa6b0 00000000`00000000 : 00000000`00000000 00000000`00000000 ffffa408`c3cc0dd0 fffff804`64543d23 : nt!KeReleaseSpinLock+0x49

THREAD_SHA1_HASH_MOD_FUNC:  1c027c82d9c4709b4c656109cc81f55b70cbeee3

THREAD_SHA1_HASH_MOD_FUNC_OFFSET:  e69310c9c85322da42012dca2aa2a26a47707d05

THREAD_SHA1_HASH_MOD:  c85c4f69122c7000039b2d3635f7979b39a8c94e

FOLLOWUP_IP: 
usbip_vhci+1037
fffff804`64541037 cd2c            int     2Ch

FAULT_INSTR_CODE:  44c72ccd

SYMBOL_STACK_INDEX:  0

SYMBOL_NAME:  usbip_vhci+1037

FOLLOWUP_NAME:  MachineOwner

MODULE_NAME: usbip_vhci

IMAGE_NAME:  usbip_vhci.sys

DEBUG_FLR_IMAGE_TIMESTAMP:  5ef4244c

STACK_COMMAND:  .cxr 0xffffb68c624a9cb0 ; kb

BUCKET_ID_FUNC_OFFSET:  1037

FAILURE_BUCKET_ID:  0x3B_c0000420_usbip_vhci!unknown_function

BUCKET_ID:  0x3B_c0000420_usbip_vhci!unknown_function

PRIMARY_PROBLEM_CLASS:  0x3B_c0000420_usbip_vhci!unknown_function

TARGET_TIME:  2020-06-25T20:53:25.000Z

OSBUILD:  18362

OSSERVICEPACK:  836

SERVICEPACK_NUMBER: 0

OS_REVISION: 0

SUITE_MASK:  272

PRODUCT_TYPE:  1

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

OSEDITION:  Windows 10 WinNt TerminalServer SingleUserTS

OS_LOCALE:  

USER_LCID:  0

OSBUILD_TIMESTAMP:  unknown_date

BUILDDATESTAMP_STR:  190318-1202

BUILDLAB_STR:  19h1_release

BUILDOSVER_STR:  10.0.18362.1.amd64fre.19h1_release.190318-1202

ANALYSIS_SESSION_ELAPSED_TIME:  ac6d

ANALYSIS_SOURCE:  KM

FAILURE_ID_HASH_STRING:  km:0x3b_c0000420_usbip_vhci!unknown_function

FAILURE_ID_HASH:  {7c2fe553-7197-1f78-013b-43a947aa6dd1}

Followup:     MachineOwner

cezanne, 2021-01-11:
@cbuliga-ab : Sadly, this crash dump does not help to guess a problem. Please explain more about your environment. 
: VHCI version, the device you tested and etc. VHCI kernel logs would be helpful.

cezanne, 2021-01-16:
@cbuliga-ab : I'll close this issu. Please reopen if you need.

closed 2021-01-16 by cezanne