cezanne/usbip-win #177: Connection drops after a small period of time with 0.3.0 UDE

peter-vanpoucke, 2020-07-01:
I tried to connect a device on a RPI (with Buster) using (the latest version)[https://github.com/cezanne/usbip-win/tree/v0.3.0], but after two or three minutes to be no longer "connected". I found nothing in the debug logs on either end. The Wireshark didn't show any abrupt disconnections either.

Then tried it out with (v0.1.0)[https://github.com/cezanne/usbip-win/tree/v0.1.0] and the VHCI driver which then seemed to work fine. Only tried that for five minutes though but that seemed to keep on working. After that I tried 0.3.0 again but had the same result.

Can I help with device details, a Wireshark capture, debug logging (from both ends)?

kr

Peter

cezanne, 2020-07-02:
@peter-vanpoucke : Thanks for your response. UDE vhci is a very preliminary implementation and quite  a few framework routines are not implemented yet. Your issue can be  plausible to occur. The reason why there was no debug log when disconnection occurred seems to be that there's unimplemented framework routines. I will try to add all framework routines even if they just show debug messages.

cezanne, 2020-07-03:
@peter-vanpoucke : Did you try to get debug logs using TraceView.exe or something similar ? A new UDE vhci(v0.3.0) implementation uses WPP tracing to provide debug messages. DebugView.exe cannot catch the debug message. I'll update "How to get windows kernel log" section in README.md.

cezanne, 2021-01-11:
@peter-vanpoucke : Close this issue. Please reopen if you want.

closed 2021-01-11 by cezanne