cezanne/usbip-win #178: error: recv op_common

yuripace, 2020-07-02:
using usbipd on debian 10 as server. works corrrectly using the standard linux client. In windows è get this error with the last published release

\usbip.exe attach_ude -r ipaddress -b 1-1.3
usbip: error: recv op_common
usbip: error: read_devbuf: failed to read: err: 0x6

on server i get
 usbip-host 1-1.3: unlinked by a call to usb_unlink_urb()

what's (and where's) the problem?

cezanne, 2020-07-05:
@yuripace : 
>  In windows è get this error with the last published release

I have updated v0.3.0. Please try again.

yuripace, 2020-07-08:
different error now on windows client (windows 10 x64 virtualbox VM)
usbip: error: usbip_vhci_attach_device: DeviceIoControl failed: err: 0x57
usbip: error: import device
usbip: error: read_devbuf: failed to read: err: 0x6

on server (debian 10 VM) i get

[ 2989.744501] usbip-host 1-1.3: device reset
[ 3220.122862] usbip-host 1-1.3: stub up
[ 3220.206890] usbip-host 1-1.3: recv a header, 0
[ 3220.305515] usbip-host 1-1.3: reset full-speed USB device number 5 using dwc_otg

cezanne, 2020-07-18:
@yuripace : You seemed to run with "attach" command instead of "attach_ude".

TheMohawkNinja, 2020-10-27:
Hello,

I am also receiving the original error (0x6) while attempting to attach a device to a Windows machine that is physically located on a Linux machine. I have attempted to use both the master branch release (which does not contain the "attach_ude" command, and also v0.3.0 per [this page](url) which does contain the "attach_ude" command, which I tried using, but received the same error.

Daemon is running on Linux machine, and the VHCI root driver is installed.

I do see mention in Device Manager of a "usbip-win CPDO" which got installed, but is showing yellow. Google didn't seem to hint at what this is needed for, so that could be the cause of the problem, however I don't see documentation on how to install that specific driver.

cezanne, 2021-01-04:
@TheMohawkNinja : 
> EDIT: Issue resolved. I had been following the older documentation that specified starting the daemon with "usbipd -D", however starting it with "systemctl start usbipd" resolved the issue.

Thanks for sharing your troubleshooting. Your issue seems to be related with permission.

closed 2021-01-04 by cezanne