cezanne/usbip-win #179: Uninstalling stub drivers

TheProgramableTurtle, 2020-07-04:
Every time I connect a device by USB it is captured by the stub and rendered useless until the driver is manually updated to the correct default. I've removed the certificates from the stores, deleted the driver from the system drivers folder, and deleted the drivers from the Device Manager when that didn't work, but my PC still insists on using the stub drivers for everything. Every device is unbound using usbip, and I've disabled test codesigning. How can I fully uninstall the drivers? (It's driving me up the wall having to use the keyboard to change the driver for my mouse each time I boot the computer).

cezanne, 2020-07-05:
@TheProgramableTurtle : 
> but my PC still insists on using the stub drivers for everything.

It's weird. When usbip.exe binds a USB device, a stub driver is updated to the only bound device.
Can you share the output of `pnputil.exe -e` in command pronmpt ?

TheProgramableTurtle, 2020-07-05:
I've removed the registry entries, but I'll reinstall and check.

TheProgramableTurtle, 2020-07-05:
I have uninstalled the Windows provided driver and installed the Apple driver as suggested [here](https://github.com/cezanne/usbip-win/issues/149#issuecomment-622367314), and my HWID matches, but I still get the failed to export device error.

cezanne, 2021-01-11:
@TheProgramableTurtle : How about your issue with an up-to-date version?

TheProgramableTurtle, 2021-01-11:
@cezanne Unfortunately I don't have the time to test right at the moment, and I already have my system set up for other use. I will be able to test on a new system in a couple of months though. Since it doesn't seem like anyone else has the same issue I'll close it. Thanks for following up though, most other people just ignore issues and close them after a few months.

TheProgramableTurtle, 2021-01-11:
I thought I closed this huh

closed 2021-01-11 by TheProgramableTurtle