cezanne/usbip-win #180: VHCI UDE driver does not work with USB 3.0 device

koudis, 2020-07-05:
I test USB 3.0 Flash drive and it does not work.

I checkout clean usbip-win repository, checkout to vhci_ude (77f9c6fab0), build, install by `usbip.exe install_ude` and try to attach by `ysbip.exe attach_ude -r ... -p ...` and i've got

```
usbip: error: usbip_vhci_attach_device: DeviceIoControl failed: err: 0x57
usbip: error: import device
usbip: error: read_devbuf: failed to read: err: 0x6
```

Screenshot from Device manager and log from TraceView for vhci_ude driver
 
- [vhci_ude.zip](https://github.com/cezanne/usbip-win/files/4875062/vhci_ude.zip)
- [image]

gtxent, 2020-08-19:
I am experiencing a similar issue with a USB 3.0 flash drive, usb 2.0 works fine.  any solution yet?

koudis, 2020-08-21:
@cezanne is this due to not upported USB 3.0 so we need add signatures etc for USB 3.0 to ude driver?

cezanne, 2020-08-27:
@koudis : UDE does not seem to support 3.0. And I remembered that it did not work with a simple workaround.

koudis, 2020-08-28:
@cezanne are you sure? The documentation for UDE ha an example with USB 3.0
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/writing-a-ude-client-driver

cezanne, 2020-08-29:
@koudis : 
> are you sure? The documentation for UDE has an example with USB 3.0

So\~orrry\~ I'm not sure.. 😢 and you're almost right.
Currently, I'm trying to resolve another bug of vhci(UDE).

windowsair, 2021-02-13:
@koudis :

Hello, I'm having the same issue.  Can you provide the original USB device description, thank you! 😇

cdeadspine, 2021-02-26:
Not sure how/if this is related (I think in one of the other issues the developer says he is working on using a "native usb root hub" for some reason)

But here is how this is currently failing for me:
linux host usbipd on Alpine 3.13
usbip-win VHCI(ude)

The attach works fine, i have software that sees and communicates with it, windows makes many "attach and detach sounds" for a few seconds and eventually disables itself.
```
Device Manager -> Universal Serial Bus Controllers -> USB Root Hub (USB 3.0)
Windows has stopped this device because it has reported problems. (Code 43)
The USB hub failed to reset.

Events
    Device USB\ROOT_HUB30\1&2b53a856&1&0 was started.

Driver Name: usbhub3.inf
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Service: USBHUB3
Lower Filters: 
Upper Filters: 

    Device USB\ROOT_HUB30\1&2b53a856&1&0 was configured.

Driver Name: usbhub3.inf
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Driver Date: 05/09/2020
Driver Version: 10.0.19041.264
Driver Provider: Microsoft
Driver Section: Generic.Install.NT
Driver Rank: 0x800000
Matching Device Id: USB\ROOT_HUB30
Outranked Drivers: 
Device Updated: false
Parent Device: ROOT\USB\0000

    Device USB\ROOT_HUB30\1&2b53a856&1&0 was not migrated due to partial or ambiguous match.

Last Device Instance Id: USB\VID_0930&PID_6544\001CC0C60D83C030342B06FF
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Location Path: 
Migration Rank: 0xF000FFFFFFFFF132
Present: false
Status: 0xC0000719
```

So things are working for a bit, but then something automatic with this associated "usb root hub" trying to "migrate" or "reset" or something i dont understand goes wrong

cezanne, 2021-05-05:
Please try with https://github.com/cezanne/usbip-win/releases/tag/v0.3.5-dev

cezanne, 2021-07-26:
@koudis: The recent ude(v0.3.5) can support USB3. I'll close it. Please reopen if you need.

closed 2021-07-26 by cezanne