cezanne/usbip-win #184: unable to connect to 192.168.180.22 port 3240

k-str50, 2020-07-13:
Hi,
I'm not sure whats wrong.
When I execute teh command
 `.\usbip.exe list --remote 192.168.180.22`
I get the error 
`unable to connect to 192.168.180.22 port 3240`
I have no idea whats wrong.
Please help!

closed 2020-07-13 by k-str50

Chris2011, 2020-11-11:
If you got it working, it would be good for everyone to know your solution :). It worked for couple of hours ago after whatever reason, I got the exact same result: usbip: error: unable to connect to 192.168.178.42 port 3240:

So please tell us, how did you solve it? Raspi as my USB Server is working just fine, I'm using Windows and just `usbip.exe list -r <ip>` is not working and shows me this error.

Chris2011, 2020-11-11:
If you got it working, it would be good for everyone to know your solution :). It worked for couple of hours ago after whatever reason, I got the exact same result: usbip: error: unable to connect to 192.168.111.11 port 3240:

So please tell us, how did you solve it? Raspi as my USB Server is working just fine, I'm using Windows and just usbip.exe list -r <ip> is not working and shows me this error.

Chris2011, 2020-11-11:
Ok for me and for everyone else, the usbipd daemon wasn't running anymore. Not it is working again. Just use `usbipd -D`