cezanne/usbip-win #185: usbip v 0.3.0: Unable to connect from win10 client to devices on raspberry server failed to read: err: 0x6

k-str50, 2020-07-13:
Hi,
I'm not sure what's wrong.
I try to use usbip win 0.3.0.
Making and installing the usbip v0.3.0 is fine.

PS C:\Users\Kurt\usbipV0.3.0> .\usbip.exe install_ude
usbip: info: vhci(ude) driver installed successfully

Even in the device manager there is a new device called usbip-win VHCI(ude)
But.. when I execute the command to attach the remote device:

PS C:\Users\Kurt\usbipV0.3.0> .\usbip.exe attach_ude -r 192.168.180.22 -b 1-1
usbip: error: recv op_common
usbip: error: read_devbuf: failed to read: err: 0x6

On the raspberry I find this:

pi@raspberrypi:~ $ usbip list -l
 - busid 1-1 (05e3:0702)
   Genesys Logic, Inc. : USB 2.0 IDE Adapter [GL811E] (05e3:0702)

The version of usbip is

**On raspberry:**
pi@raspberrypi:~ $ **_usbip version_**
usbip (usbip-utils 2.0)
and 
**on windows client:**
PS C:\Users\Kurt\usbipV0.3.0> **_.\usbip.exe version_**
usbip (usbip 1.0.0)

I have no idea whats wrong.
Please help!
Thanks a lot in advance!

k-str50, 2020-07-19:
Hi,
I could solve this problem.
It's a little tricky but finally it works.
Thanks to cezanne!

alcane2011, 2020-08-31:
hi k-str50,
I have same problem in same setup. Any help ? :-)
Thank you for your assistance.
Alcane

k-str50, 2020-09-05:
Hi  alcane2011,
yes I get it working,, but I do not find the description for this.
I use on the 
- raspberry the standard usbip 
- on windows 10 x64 I got something using **_usbip.exe attach_ude -r <serverIP> -b 1-1_**
Maybe somebody knows the link for this description.

Wqrld, 2020-09-06:
> Hi alcane2011,
> yes I got it working!
> I use on the
> 
> * raspberry the standard usbip
> * on windows 10 x64 I got something using **_usbip.exe attach_ude -r  -b 1-1_**
> 
> You can find more here : ([v0.3.0...vhci_ude](https://github.com/cezanne/usbip-win/compare/v0.3.0...vhci_ude))

I am using the vhci ude branch and am having the same issue connecting to a ubuntu server

Client: 
```
usbip.exe attach_ude -r remoteip.local -b 1-1
usbip: error: recv op_common
usbip: error: read_devbuf: failed to read: err: 0x6
```

server:
```
root@node:/usr/share# usbipd
usbipd: info: starting usbipd (usbip-utils 2.0)
usbipd: info: listening on 0.0.0.0:3240
usbipd: info: listening on :::3240
usbipd: info: connection from myip
usbipd: info: received request: 0x8003(5)
usbipd: info: requested device not found: 1
usbipd: info: request 0x8003(5): failed
```

What did you change to get it working?

k-str50, 2020-09-06:
Hi Wqrld,
for better understanding type usbip -h either on the **client or server**
it worlks for me in the following way:
First make this steps on the **server:**

1. start the usbip deamon : **_sudo usbip -D_** ---> now it runs on background
2. execute **_sudo usbip list -l_** ---> now you should see the device connected to the usb port on the server. It's local
3. naxt is **_sudo usbip bind -b 1-1_** ---> this prvides the usb device here e.g : here  bus id = 1-1; I use a raspberry zero w it has only one USB port

Now continue on the **client** using **power shell** **as administrator** if it's a windows10 computer

1. **_./usbip.exe attach_ude r {ip_of_your_server} -b 1-1_** ---> bus id as shown on the server

Now you should be able to access the remote USB device.
Please see also : []https://wiki.archlinux.org/index.php/USB/IP but on the client use **attach_ude**
I did it using my Canon EOS camera and an formatted USB stick
good luck

Wqrld, 2020-09-07:
strange, im stil getting the same error even with your, strangely enough it seems like the server itself has no issues and completes the request just fine. @cezanne do the latest commits change anything related to this? if so are you able to compile a updated .exe/driver as my development environment doesnt really work.

k-str50, 2020-09-07:
Hi Wqrld,
I get the same error when I do it in the way you described.
On server :
```
pi@TeleskopEOS:~ $ usbipd
usbipd: error: not running as root?
usbipd: info: starting usbipd (usbip-utils 2.0)
usbipd: info: listening on 0.0.0.0:3240
usbipd: info: listening on :::3240
usbipd: info: connection from 192.168.180.38:61971
usbipd: info: received request: 0x8003(5)
usbipd: info: requested device not found: 1-1
usbipd: info: request 0x8003(5): failed
```
And on the client : 
```
\usbip.exe attach_ude -r myrpi -b 1-1
usbip: error: recv op_common
usbip: error: read_devbuf: failed to read: err: 0x6
```
I think your problem is that you first have to run the usbip deamon on the server. It's importend to execute this commands in this sequence:
On the server (**here called myrpi**): 
```
pi@myrpi:~ $ sudo usbipd -D
pi@myrpi:~ $ sudo usbip list -l
 - busid 1-1 (048d:1176)
   Integrated Technology Express, Inc. : unknown product (048d:1176)

pi@myrpi:~ $ sudo usbip bind -b 1-1
usbip: info: bind device on busid 1-1: complete
```
then on the client : 
```
.\usbip.exe list -r myrpi
usbip: error: failed to open usb id database
Exportable USB devices
======================
 - myrpi
        1-1: unknown vendor : unknown product (048d:1176)
           : /sys/devices/platform/soc/20980000.usb/usb1/1-1
           : (Defined at Interface level) (00/00/00)
.\usbip.exe attach_ude -r myrpi -b 1-1
```
Now you should have access to your USB device.
Good luck!
I write this so detailed because usbip is a very usefull tool made by @cezanne.

Wqrld, 2020-09-07:
@k-str50 Thanks alot, i got it to connect to my ubuntu server now but that usb device kept making disconnect sounds (nothing to worry about as it was a virtual usb device in a vm and probably not meant to be connected to)

I will try it soon on my raspberry pi where i initially wanted it. I still find it strange that running as a daemon doesnt work where running usbipd normally in another tmux screen doesnt. Let's hope that it works first try on the PI 

Also: maybe open a PR to add some docs to the project if you have the time, looking around i wasnt the only one with this issue.

cezanne, 2021-01-11:
@k-str50 @Wqrld : It looks like that your issues have been almost resolved. I'll close this issue. Please reopen if required.

closed 2021-01-11 by cezanne