cezanne/usbip-win #186: Automatically connect/disconnect devices as they are bound on the server

RoganDawes, 2020-07-14:
Hi. Thanks for a great tool!

I have a need to automatically bind and unbind devices on the Windows client as they are bound/unbound on the server. This seems related to this other issue: https://github.com/cezanne/usbip-win/issues/160

In my case, the intention is to make a Linux usbip server's USB port "available" to a Windows client. On Linux, it is fairly straightforward to automate binding and unbinding the device as it is plugged in using appropriate `udev` rules that match the bus/port in question. However, it is still necessary to manually execute the `bind`/`unbind` commands on the Windows client. Ideally, the Windows client would poll periodically for any new/removed devices, and execute the `bind`/`unbind` operations automatically.

This could perhaps be enabled by adding an `auto -i seconds -r target` option that polls the specified server for any changes. This could perhaps be implemented using a Powershell script, but unfortunately, the output of `usbip list` is not the easiest to parse. It seems to me that it would make more sense to implement this within the `usbip` program itself?

cezanne, 2021-01-11:
@RoganDawes : Yeah, it's wonderful if usbip.exe supports a persistent connection as you proposed.

RoganDawes, 2021-01-11:
Not sure if it would require significant architectural changes to implement this? Without having looked at the sources, I imagine that each "bind" operation results in a backgrounded process that manages the socket connected to the server, and each "unbind" operation (or server unexport) essentially kills that process? (Replace "process" with "thread", perhaps?)
If that is the case, adding one extra process or thread that periodically polls the server to check the current status doesn't sound terribly hard.

cezanne, 2021-01-11:
@RoganDawes : 
> If that is the case, adding one extra process or thread that periodically polls the server to check the current status doesn't sound terribly hard.

 I agree. 😄   However, an external process approach seems to be better in architectural viewpoint. Of course, remote listing should support parsable output. Local listing already supports.

RoganDawes, 2021-01-12:
Making the output parsable would probably be the least effort (and probably a nett improvement overall anyway).

cezanne, 2021-01-24:
@RoganDawes : Can you resolve this issue with features on dev branch?
- attach command displays a result port number.
- port command can check the sanity of an attached port. 

Please check [dev branch](https://github.com/cezanne/usbip-win/tree/dev) and [wiki](https://github.com/cezanne/usbip-win/wiki/External-Program-Integration).

RoganDawes, 2021-01-24:
I imagine the process would look something like (bash-style pseudocode):
```
while true; do
  # try to connect to device
  PORT=$(KEEPALIVE_TIMEOUT=3 usbip.exe attach -r <host> -t -b 1-1)
  if [ %error_level% -gt 0 ] ; then
    while usbip.exe port -p $PORT; do
      # check periodically for network timeout or device disconnection
      sleep 5
    done
  fi
  # wait a bit before trying again
  sleep 5
done
```
Looks reasonable to me, once I translate that into decent CMD.exe or PS!

FL0WL0W, 2021-02-13:
Hey Everybody,

Needed the same thing so I whipped up a quick windows service in C#
https://github.com/FL0WL0W/usbipdaemon

hope this helps!

RoganDawes, 2021-02-13:
404?

FL0WL0W, 2021-02-13:
> 404?
> […](#)

Fixed. Sorry it was set to private

cezanne, 2021-07-26:
@RoganDawes : Please reopen if you need.

closed 2021-07-26 by cezanne