cezanne/usbip-win #188: various errors using 0.3.0 ude release

andrewzli, 2020-07-15:
I'm attempting to connect my Windows 10 PC (as a client) to a USB device exposed on a Raspberry Pi 4 (4.19.118-v7l+) host. I believe I've had a successful install and certificate import (I followed the instructions in the [vhci_ude branch](https://github.com/cezanne/usbip-win/tree/vhci_ude#windows-usbip-client) readme), and running `install_ude` runs without errors

```console 
C:\Users\user\Downloads\usbip-win>usbip install_ude
usbip: info: vhci(ude) driver installed successfully
```

To double check if it was a problem on the server side, I've tested connecting to the Pi server via another client Pi and it works without any problems. When I list the devices available on the Pi server (192.168.1.69) from my Windows PC, I get a `usb id database` error but it does recognize there is a device available, albeit without recognizing the vendor + product for some reason. #128 has the same database error but that appears to be from a different release.

```console
C:\Users\user\Downloads\usbip-win>usbip list -r 192.168.1.69
usbip: error: failed to open usb id database
Exportable USB devices
======================
 - 192.168.1.69
      1-1.1: unknown vendor : unknown product (3923:709b)
           : /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1
           : (Defined at Interface level) (00/00/00)
```

When I run `attach_ude`, I get a `tcp connect` error. But I don't think its the same issue as in #3 . Both the raspberry pi source tree [4.19.y](https://github.com/raspberrypi/linux/blob/fe915debc1925397f78db8bbfc9f488b0ce4e009/tools/usb/usbip/configure.ac#L5) and the [vhci_ude branch](https://github.com/cezanne/usbip-win/blob/77f9c6fab0ad36ad8856f99d68d5b3523c13614b/userspace/config.h#L27) contain the same USBIP_VERSION.

```console
C:\Users\user\Downloads\usbip-win-master>usbip.exe attach_ude -r 192.169.1.69 -b 1-1.1
usbip: error: tcp connect
```
I'm not sure what to do at this point but any help would be appreciated!

k-str50, 2020-07-16:
Hi andrewzli 
I have the same problem!
Is there something wrong with the usbip server? What do I need for an raspberry#Zero and a raspberry 3?
Please for any help
Thanks for any assistence!
Kurt

andrewzli, 2020-07-16:
@k-str50 I think your errors are slightly different than mine but they still could be related. As for the `no exportable devices found on 192.168.180.22`, did you bind the device on your pi or connect your pi devices to another client? To bind the device use `sudo usbip bind -b 1-1` on your pi. As for the error 6 you have there, did you run `bcdedit.exe /set TESTSIGNING ON` and reboot your machine? Try those out and see if it works, if not, I would revert to [usbip version 0.1.0](https://github.com/cezanne/usbip-win/releases/tag/v0.1.0). Reverting to the 0.1.0 release worked for me. As mentioned on the [usbip version 0.3.0](https://github.com/cezanne/usbip-win/releases/tag/v0.3.0) release page:
> These binaries were built against vhci_ude branch, which is developed from scratch via KMDF(Kernel-Mode Driver Framework) UDE(USB Device Emulation).

I presume the errors I'm getting are due to the version 0.3.0 being quite new, being based off a different framework.

k-str50, 2020-07-19:
Hi andrewzli,
I got version 0.3.0 working very well.
Good tool!

wanglao677, 2020-07-24:
> Hi andrewzli,
> I got version 0.3.0 working very well.
> Good tool!
Hello!I use win10PC as sever and win10PC as client,but it failed.I want to know your test environment.How could u work well?I want to know lol .Thank you very much.

algarih, 2020-08-18:
tried usbip-win 0.3.0 and usbip-win 0.0.9

but i get **usbip: error: tcp connect** when trying to attach

gammeflamme, 2020-10-19:
I've tried using v1.0 v2.0 and v3.0 I've also tried the og from 2011and all get "usbip: error: tcp connect" I'm trying to connect windows 10 to raspberry pi running usbip-utils 2.0 and spent 8h searching for some solution so at last, I come here to seek knowledge.

designworxz, 2020-10-28:
> Hi andrewzli,
> I got version 0.3.0 working very well.
> Good tool!

Can you share how as am getting same errors
usbip: error: recv op_common
usbip: error: read_devbuff: failed to read: err: 0x6

cezanne, 2021-01-18:
I'll close this issue. Please, retry with [the recent release](https://github.com/cezanne/usbip-win/releases/tag/v0.3.3-dev).

closed 2021-01-18 by cezanne