cezanne/usbip-win #191: attach_ude multiple clients?

k9bbzJavftZaqAqSW3dY, 2020-08-01:
Hi,
I have tried to attach multiple devices, hosted on a raspberry pi, to a windows client. From another thread in this issue list, I've seen that the attach_ude command is blocking, but when I open another terminal to attach another device, I get error messages.
Is there a way to use multiple devices with a Windows client? Maybe the device driver can be written with a different serial number and launched from another copy of usbip.exe in a separate folder, so that I can fully parallelize it? I'd be willing to tinker, but have no experience with device drivers... Let me know if I can make it work some way.

Kind regards

cezanne, 2020-08-05:
@k9bbzJavftZaqAqSW3dY : Multiple clients via each attach command should be supported. But I had not checked that  multiple attach_ude commands are working. I'll try. Can you share your error message?

cezanne, 2020-08-07:
@k9bbzJavftZaqAqSW3dY : I have checked that attach_ude is not working with multiple clients. The log below shows the case.

```
C:\work>usbip --debug attach_ude -r 192.168.12.172 -b 1-1
usbip: debug: C:\work\usbip-win\userspace\src\usbip\usbip.c:171:[run_command] running command: `attach_ude'

usbip: debug: C:\work\usbip-win\userspace\src\usbip\usbip_vhci.c:60:[usbip_vhci_driver_open] device path: \\?\root#usb#0000#{d35f7840-6a0c-11d2-b841-00c04fad5171}
usbip: debug: C:\work\usbip-win\userspace\src\usbip\usbip_attach_ude.c:58:[import_device] got free port 1
usbip: error: usbip_vhci_attach_device: DeviceIoControl failed: err: 0xb7
usbip: error: import device
usbip: error: read_devbuf: failed to read: err: 0x6
```

This is due to the bug that vhci(ude) does not track attached devices. :cry:

k9bbzJavftZaqAqSW3dY, 2020-08-07:
Thanks for checking. I'm having other computer problems at the moment, which is why I couldn't constructively follow up on this issue!

cezanne, 2020-08-08:
> This is due to the bug that vhci(ude) does not track attached devices

This comment is not correct. Please check 97964cb0e79690cc5243d3769e46b6174b0dc357, which can resolve this issue.

cezanne, 2020-09-04:
@koudis : I tried several tests similar to your environment. Two devices(USB stick and USB mouse) were working without  disconnection or restart during 5 minutes. Can you explain more about your test?

koudis, 2020-10-08:
@cezanne sorry for late answer. I was ill and we had lot of work so i had no time for myself.

I will try. I will try it again with newest UDE driver and if problem persist I will provide more detailed description.

sq2jul, 2020-10-12:
@cezanne I'm currently testing this great piece of software and fall into the same problem - attaching multiple remote device in the same time. I'm not familiar with Microsoft build stack. Would you be so nice to provide binary version with this patch included for testing purposes?

cezanne, 2020-10-13:
@sq2jul : 
> Would you be so nice to provide binary version with this patch included for testing purposes?

Here it is;  [binaries](https://github.com/cezanne/usbip-win/files/5369729/vhci_ude_97964cb.zip) for 97964cb commit. (x64 debug build)

sq2jul, 2020-10-14:
I did some tests. For simple devices like usb serial port or keyboard I was able to attach multiple devices in the same time. Works like a charm! Thanks. I tried with USB sound card too but it works partially. Sound card device attached in windows system, but i was not able to force windows to play any sound through it.

I have two questions:
- Is it normal behavior that 'usbip attach_ude ...' occupies terminal all the time and the only way to 'detach' and return to the prompt is by pressing ctrl+c?
- Is there any chance for 'multiple devices attached in the same time' feature in main release any time soon?

Thank you for doing great job!
Juliusz

cezanne, 2020-10-16:
@sq2jul : 
> - Is it normal behavior that 'usbip attach_ude ...' occupies terminal all the time and the only way to 'detach' and return to the prompt is by pressing ctrl+c?

Yes. usbip.exe should run while a device is attached. Its limitation is different from linux version.  

> - Is there any chance for 'multiple devices attached in the same time' feature in main release any time soon?

I hope so. But a UDE vhci still has problems for other than a simple USB device.

sq2jul, 2020-10-17:
I don't have experience in development under MS os, but I'm eager to test. I'm in possession of few devices:
- RTL-SDR receiver
- cheap Chinese USB sound card (audio codec)
- RTL232 USB to serial converter
- some wireless keyboard with integrated trackball by Dell. Communication in use of usb dongle.

If you need any test in regards of those devices do not hesitate to ping me.
I'm able to test Windows as a client and raspberry pi at Raspbian as a server for usb devices.

commit 97964cb0 (2020-08-07): Fix vhci(ude) port status

The first byte of vhci port status IOCTL output union is a max port
number. The legacy code seems to be misleading but for the time being
vhci(ude) follows that rule. This commit will resolve #191.
commit 1912060a (2020-12-25): Fix vhci(ude) port status

The first byte of vhci port status IOCTL output union is a max port
number. The legacy code seems to be misleading but for the time being
vhci(ude) follows that rule. This commit will resolve #191.
closed 2020-12-25 by cezanne