cezanne/usbip-win #192: Missing Serial Number and Device Name

jakjablonski, 2020-08-05:
Hi,

I'm trying to share usb devices using your software and I found 2 issues.

1. Missing Device Name and Serial Number
Server:
[image]
Client:
[image]

Device Name has changed to HPDO and Serial Number is missing

2. Max length of the serial number is too small
After setting this line to 18 - #define MAX_VHCI_INSTANCE_ID  16 -> 18
Then I can assign a longer serial number when attaching USB on Windows.
usbip.exe attach -r 192.168.0.1 -b 5-1.1 -i  4600000D0D221600

olehkon, 2020-08-05:
[image]
The serial number is present in the log when connected to the client.

cezanne, 2021-01-12:
@jakjablonski : 
1. Missing Device Name and Serial Number
    I gave up USBdeview working for vhci(WDM). It was too complicated to emulate a virtual hub. That's why I was starting to develop vhci(ude). See #111
2. Max length of the serial number is too small
   I got it. Sadly, vhci(ude) does not support a customized instance id. It can be TODO.

cezanne, 2021-01-14:
@jakjablonski @olehkon : Please check [this wiki](https://github.com/cezanne/usbip-win/wiki/Deployment-Tips#usb-serial-overwriting)

cezanne, 2021-01-23:
@jakjablonski : I'm going to close this isuse. Please reopen if you want.

closed 2021-01-23 by cezanne